noble/noble #360: Characteristic null on Raspberry Pi

jonflanders, 2016-03-24:
When I run my code on osx - everything works fine.  When I run it on Raspberri Pi (latest debian) - the discoverCharactaristics callback return nil (and no error).

Any idea why and what I could do to debug something like this?

jeberhardt, 2016-03-24:
I have seen this behaviour as well, but it's intermittent.  Sometimes the characteristic is null, and sometimes it has a value.  I have many peripheral devices that are connectable, and once connected, I look for the services, and then characteristics.  If the script disconnects from the peripheral and then reconnects, the characteristic may appear then.

sandeepmistry, 2016-03-28:
@jonflanders 

To debug, I suggest you look at the HCI dump capture or the debug logs.

If you need a hand, please provide the following:
 * A simplified example app to re-create the issue
 * Debug out: ```sudo DEBUG=* node <file>.js```
 * HCI dump capture: ```sudo hcidump -t -x```

rvsrvs, 2016-03-29:
I'm having a similar problem and can supply much more information..  I am advertising a custom service on an iPhone.  The Mac sees the custom service fine, the R/Pi not at all.  If necessary I can provide you with an app which advertises the service on iOS as well.  The results are as follows:

Running the straight peripheral_explorer.js from the distribution on Mac OS/X and R/Pi at the same time produces the following output:

On Mac OS/X: 

sudo node node_modules/noble/examples/peripheral-explorer.js ce2bcf190ad6425abde6196033e97ba1
peripheral with ID ce2bcf190ad6425abde6196033e97ba1 found
  Local Name        = The VPhone 6
  Service Data      = 
  Service UUIDs     = 7c83ee4efc494598aca239a61d6c1c8e

services and characteristics:
d0611e78bbb44591a5f8487910ae4366
  8667556c9a374c9184ed54ee27d90049
    properties  write, notify, extendedProperties
180f (Battery Service)
  2a19 (Battery Level)
    properties  read, notify
    value       64 | 'd'
1805 (Current Time Service)
  2a2b (Current Time)
    properties  read, notify
    value       e007031d0b0137024b02 | '`
                                         7K'
  2a0f (Local Time Information)
    properties  read
    value       ec04 | 'l'
180a (Device Information)
  2a29 (Manufacturer Name String)
    properties  read
    value       4170706c6520496e632e | 'Apple Inc.'
  2a24 (Model Number String)
    properties  read
    value       6950686f6e65372c32 | 'iPhone7,2'
7c83ee4efc494598aca239a61d6c1c8e
/private/tmp/node_modules/noble/examples/peripheral-explorer.js:102
                            userDescriptionDescriptor.readValue(function(error, data) {
                                                      ^

TypeError: userDescriptionDescriptor.readValue is not a function
    at /private/tmp/node_modules/noble/examples/peripheral-explorer.js:102:55
    at /private/tmp/node_modules/async/dist/async.js:2768:29
    at /private/tmp/node_modules/noble/examples/peripheral-explorer.js:98:34
    at wrappedIteratee (/private/tmp/node_modules/async/dist/async.js:2765:17)
    at replenish (/private/tmp/node_modules/async/dist/async.js:808:21)
    at /private/tmp/node_modules/async/dist/async.js:818:15
    at eachOfLimit (/private/tmp/node_modules/async/dist/async.js:823:28)
    at /private/tmp/node_modules/async/dist/async.js:828:20
    at Object.<anonymous> (/private/tmp/node_modules/async/dist/async.js:2783:17)
    at /private/tmp/node_modules/noble/examples/peripheral-explorer.js:95:29

On Raspberry Pi:

pi@raspberrypi:~/noble $ sudo nodejs node_modules/noble/examples/peripheral-explorer.js ce2bcf190ad6425abde6196033e97ba1

No discovery whatsoever.  This is what I see with my own code as well..

when I turn DEBUG=* on on the Mac:

  yosemite-bindings xpcEvent: {
  "kCBMsgId": 37,
  "kCBMsgArgs": {
    "kCBMsgArgAdvertisementData": {
      "kCBAdvDataIsConnectable": 1,
      "kCBAdvDataServiceUUIDs": [
        {
          "type": "Buffer",
          "data": [
            124,
[... 1589 lines cut ...]
  hci 		LE meta event data = 0001d64d3a5224711702010613ff4c000c0e008a40d7466cf0421dbf6b35fdefcc +1ms
  hci 			type = 0 +1ms
  hci 			address = 71:24:52:3a:4d:d6 +0ms
  hci 			address type = random +1ms
  hci 			eir = 02010613ff4c000c0e008a40d7466cf0421dbf6b35fdef +0ms
  hci 			rssi = -52 +1ms
  gap advertisement = {"manufacturerData":{"type":"Buffer","data":[76,0,12,14,0,138,64,215,70,108,240,66,29,191,107,53,253,239]},"serviceData":[],"serviceUuids":[]} +1ms
  hci onSocketData: 043e0c02010401d64d3a52247100cc +152ms
  hci 	event type = 4 +1ms
  hci 	sub event type = 62 +0ms
  hci 		LE meta event type = 2 +1ms
  hci 		LE meta event status = 1 +0ms
  hci 		LE meta event data = 0401d64d3a52247100cc +1ms
  hci 			type = 4 +1ms
  hci 			address = 71:24:52:3a:4d:d6 +0ms
  hci 			address type = random +1ms
  hci 			eir =  +0ms
  hci 			rssi = -52 +0ms
  gap advertisement = {"manufacturerData":{"type":"Buffer","data":[76,0,12,14,0,138,64,215,70,108,240,66,29,191,107,53,253,239]},"serviceData":[],"serviceUuids":[]} +2ms

sudo hcidump -t -x
HCI sniffer - Bluetooth packet analyzer ver 5.23
device: hci0 snap_len: 1500 filter: 0xffffffff

What else do you need?

sandeepmistry, 2016-04-03:
I've answered @rvsrvs's question in #371.

@jonflanders @jeberhardt any progress on what I requested in https://github.com/sandeepmistry/noble/issues/360#issuecomment-202175844.

jeberhardt, 2016-04-15:
While I haven't been able to capture the dump, I have noticed that if I omit the UUIDs for both the service and the characteristic, then it seems to be able to find them right away.  I'll try and get more info to you.

sandeepmistry, 2016-04-18:
@jeberhardt ok, I'm going to close this for now. We can re-open once it can be reproduced and we have more into.

closed 2016-04-18 by sandeepmistry