noble/noble #362: Question: writing to 16-byte characteristic

billsalt, 2016-03-26:
Getting better, but still a bit of a NOOB...  I have a 16-byte long characteristic that I'm writing to (a data buffer in my device).  I have an array of bytes (byte buff) and I want to send out 16 at a time to the 16 byte writeable characteristic eNoteBlockDataCharasteristic.

The following doesn't seem to be doing the job; suggestions?

            var blockbuffer = new Buffer(16);
            for (var block = 0; block < 5808; block = block + 16) {
              blockbuffer.writeUInt8(bytebuff.slice(block, block+16));
              eNoteBlockDataCharacteristic.write(blockbuffer, false);
              console.log('wrote block, slice:', block, bytebuff.slice(block, block+16));
            }

mattgol, 2016-03-27:
Hi!
The writeUInt8 actually only writes a single byte into your target buffer and not the slice. You end up with a buffer of random numbers with a leading zero as the slice is onverted to a zero and inserted as first
entry.
 
And as the slice function directly returns a Buffer object you can assign it directly to blockbuffer. There is no need to prepare an object before the assignment. This boils to down to changing the first line to 
`var blockbuffer;`
and the 3. line to `blockbuffer = bytebuff.slice(block, block+16);`
(or you could directly drop `bytebuff.slice(block, block+16)`in your write statement)

Hope this helps...

billsalt, 2016-03-27:
Thanks @mattgol, but the write doesn't seem to think it is a buffer when I tried your suggestion.  Did I do something wrong here?

//            var blockbuffer = new Buffer(16);
            var blockbuffer;
            for (var block = 0; block < 5808; block = block + 16) {
//              blockbuffer.writeUInt8(bytebuff.slice(block, block+16));
              blockbuffer = bytebuff.slice(block, block+16);
              eNoteBlockDataCharacteristic.write(blockbuffer, false);
//              console.log('wrote block, slice:', block, bytebuff.slice(block, block+16));
            }

**OUTPUT**
/Users/ws/Desktop/eNodeJpgTest/node_modules/noble/lib/characteristic.js:54
      throw new Error('data must be a Buffer');
      ^

Error: data must be a Buffer
    at Characteristic.write (/Users/ws/Desktop/eNodeJpgTest/node_modules/noble/lib/characteristic.js:54:13)
    at /Users/ws/Desktop/eNodeJpgTest/index.js:176:44
    at Service.<anonymous> (/Users/ws/Desktop/eNodeJpgTest/node_modules/noble/lib/service.js:53:7)
    at Service.g (events.js:260:16)
    at emitOne (events.js:77:13)
    at Service.emit (events.js:169:7)
    at Noble.onCharacteristicsDiscover (/Users/ws/Desktop/eNodeJpgTest/node_modules/noble/lib/noble.js:276:13)
    at emitThree (events.js:97:13)
    at emit (events.js:175:7)
    at null.<anonymous> (/Users/ws/Desktop/eNodeJpgTest/node_modules/noble/lib/mac/yosemite.js:492:8)

mattgol, 2016-03-27:
This is strange. Of course, I cannot exactly replicate your setup as I'm missing the hardware. But I wrote to a characteristics directly from slice and also by sending slice to a variable and using it indirectly. Both approaches worked without a hitch.  

Cutting it down to the bare buffers I tried this:
```
var bigbuf = Buffer([0,1,2,3,4,5,6,7,8,9,10]);
var smallbuf = bigbuf.slice(3,6);
console.log(smallbuf);
```
which results in  `<Buffer 03 04 05>`. Even if bytebuff is not big enough slice would simply output an empty buffer.  Maybe, you could output blockbuffer to the console to see what's going on?

BTW: Could it be that you're writing to an Embedded Artists e-ink display - the size of the array looks familiar...

billsalt, 2016-03-27:
Thanks again @mattgol!  It is my own custom device (eNote).  Output from peripheral.explorer.js below.  Also output from console.log for the buffer.  It seems that the .write isn't accepting that the buffer is really a buffer?

--->  from peripheral.explorer
peripheral with ID b2c16aaea5894c76b9ee67c2bcaf16da found
  Local Name        = eNote
  Manufacturer Data = 4c000215dab59c4fa4d6ee286bfe8e0000bbc2bb34127856c3
  Service Data      = 
  Service UUIDs     = 

services and characteristics:
180f (Battery Service)
  2a19 (Battery Level)
    properties  read, notify
    value       64 | 'd'
180a (Device Information)
  2a29 (Manufacturer Name String)
    properties  read
    value       43434e4d | 'CCNM'
bbc21523008efe6b28eed6a44f9cb5da
  bbc21524008efe6b28eed6a44f9cb5da
    properties  read, notify
    value       00 | ''
  bbc21525008efe6b28eed6a44f9cb5da
    properties  read, write
    value       00 | ''
  bbc21526008efe6b28eed6a44f9cb5da
    properties  read, write
    value       0000 | ''
  bbc21527008efe6b28eed6a44f9cb5da
    properties  read, writeWithoutResponse, write
    value       ffffffffffffffffffffffffffffffff | ''
  bbc21528008efe6b28eed6a44f9cb5da
    properties  read, write
    value       02 | ''
  bbc21529008efe6b28eed6a44f9cb5da
    properties  read
    value       000f0100000000000000000000000000 | ''

--> from display.log of the buffer
wrote to block number
eNoteBlockDataCharacteristic, blockbuffer =  bbc21527008efe6b28eed6a44f9cb5da [ 255,
  255,
  255,
  255,
  255,
  255,
  255,
  255,
  255,
  255,
  255,
  255,
  255,
  255,
  255,
  255 ]
/Users/ws/Desktop/eNodeJpgTest/node_modules/noble/lib/characteristic.js:54
      throw new Error('data must be a Buffer');
      ^

Error: data must be a Buffer
    at Characteristic.write (/Users/ws/Desktop/eNodeJpgTest/node_modules/noble/lib/characteristic.js:54:13)

mattgol, 2016-03-28:
@billsalt 
I think I found my wrong assumption here: I thought bytebuff is also a buffer but it looks like an array. In this case you would need the Buffer constructor which can be directly be initialized from an array. How about something like this (again untested due to lack of hardware):
```
        var blockbuffer;
        for (var block = 0; block < 5808; block = block + 16) {
          blockbuffer = new Buffer(bytebuff.slice(block, block+16));
          eNoteBlockDataCharacteristic.write(blockbuffer, false);
          console.log('wrote block, slice:', block, bytebuff.slice(block, block+16));
        }
```

Off Topic: If your eNote is what I think it is I like it already! I've connected an Embedded Artist eink display to a Tinyduino with a Bluegiga BLE112 to build a battery operated display but I'm still fighting with the sleep modes. So I'm not even close to writing the master side code for this one...

billsalt, 2016-03-28:
Thanks @mattgol !  That didn't work, but it _did_ get me on the right track.  Here's the code that seems to be working.  Now on to the next issue; the writes are going too quickly and overrunning.  I either need to slow down or test.  Next bit of coding...

Also @sandeepmistry  - how do I do write without response?

Thanks again for your help.  See www.enote.ink for the device.

            for (var block = 0; block < 5808; block = block + 16) {
              var blockbuffer = Buffer(bytebuff.slice(block, block+16));
              console.log('block, blockbuffer = ', block, blockbuffer);
              eNoteBlockDataCharacteristic.write(blockbuffer, false);
            }

billsalt, 2016-03-28:
@mattgol It is all working.  But, it takes over 10 minutes due to write with response.  I should be able to cut it down to seconds using write-without-response, as we did on iOS.  I'm going to log a question to sandeep.

closed 2016-03-28 by billsalt