noble/noble #363: Persistent services and characteristics across disconnect and reconnect

mattgol, 2016-03-27:
First of all thanks for noble - it's the most mature piece of BLE software out there!  Noble made it very simple to connect to my homemade peripherals and made my project possible in the first place - but now I'm facing an issue regarding battery life: I plan to connect to the device every couple of minutes, read a characteristic, and disconnect. Unfortunately, I have to go through the service discovery process every time I reconnect - which increases the number of necessary bluetooth packages and this means less battery life.

Is there a way to store the GATT structure across a disconnect/reconnect cycle? This would solve my issue.

Right now there are two workarounds I can use:
1. Keep the connection but with long connection interval and slave latency (this can be requested by the slave). This allows me about 30 sec in between packets but not longer as the hardware doesn't accept anything longer as supervision timeout.
2. Use readHandle/writeHandle to read/write directly. This means I have to change the server script every time the GATT structure changes. Also this wouldn't work on OSX but this doesn't hurt me on my linux box.

Anybody out there who had the same issue?

ayhoung, 2016-03-27:
Have you tried storing the peripheral object in cache and reconnecting to it after a while?

ayhoung, 2016-03-28:
@mattgol could you share how you are using readHandle/writeHandle to read/write directly?

mattgol, 2016-03-28:
@ayhoung I dug into the source code of noble-device and noble and also ran everything in the node console interactively to have a look into the objects. I also had btmon running to look at the actual communication with the BT hardware. I could trace the issues down to the internal representation in the moble master object.This is my current status (everything is on linux and looks different in OSX):
- The peripheral object is constructed in the library during the discovery and returned to the user. After service discovery it contains the structure of the services and characteristics. 
- The noble master object contains an array of the handles to the open connections to translate the object id to the handle of the connection. The handle number matches the output of 'hcitool con'. This list is updated after the reconnect so this is not the issue.
- The noble master object also keeps track of the GATT handles. The read method of a characterics object then calls the read method in the noble master object which then looks up the GATT handle number in its list. The actual communication with the BT hardware does not care about the characteristics structure - it only uses the GATT handle numbers. This is by design for BLE communication as it cuts down the transmitted bits to a minimum this way. 
- The noble master object then catches the event with the returned data and relays the data to the user (via the callback). 
- Unfortunately, the list of GATT handles in the master object is removed after a disconnect. My peripheral object still has all the information about the characteristics but I cannot use it anymore since it's lacking the vital GATT handle number. Calling a read on a characteristics after a reconnect throws an error as the GATT list in the master object is empty for my peripheral.
- The GATT structure is specified to be stable and typically only changes after a firmware upgrade of the device. This is also by design to cut down the transmitted bits as the structure of services and characteristics only have to be transmitted once.
- This is also the reason I can get away with using readHandle directly - all I need is the GATT handle number which is stable.

This leaves me currently with two shortterm options if I don't want to keep the connection open indefinitely:
1. Use readHandle and manually supply the handle number. I can read out the numbers with external tool (I use BLE-GUI with Bluegiga hardware). This is what I do currently - not very elegant but works for my devices. 
2. Reach into the internals of noble to take out the list of GATT handles and put it back in later. This is very likely to break with updates of noble as I would be digging very deep into the bowels of noble.

I thought my application profile would be typical for BLE. The design actively encourages my approach of dropping the connection and reconnecting if necessary: It only needs milliseconds after an advertisement package is received to establish the connection and only half a dozen packets over the air for a full cycle of connect-read-disconnect. But please correct me if I'm completely mistaken and should not work like this...

sandeepmistry, 2016-04-03:
@mattgol great to see you dive into the noble source!

Have you looked into reusing the GATT instance inside the HCI socket binding?

It is currently cleaning up the GATT instance [here](https://github.com/sandeepmistry/noble/blob/master/lib/hci-socket/bindings.js#L232-L249) and creating a new one on connection [here](https://github.com/sandeepmistry/noble/blob/master/lib/hci-socket/bindings.js#L182). I believe this was done to cleanup and save memory, in case the peripheral was never re-used. However, for your use case  it makes sense to re-use the old one. If you can experiment with this, I think this is a minimal change, and worth revisiting via a PR/discussion.

mattgol, 2016-04-04:
@sandeepmistry, thanks for the encouragement! 
I played around with noble and noble-device in the interactive mode of node.js to see how the objects interact. I was able to extract the services, characteristics, and descriptors from _gatts and put it back after reconnect. The following code based on noble-device demonstrates that it works. It's a dirty hack as I directly access the internals but it should be possible to keep the 3 objects out of the noble master object and attach it to the peripheral which stays with the user. This way we could have both: A clean master object and persistent service structure - the user would have complete control.
 I forked the repository and will give it a try... I put up a PR when I got something to show.
```
var async = require('async');
var NobleDevice = require('noble-device');

var TestDevice = function(peripheral) {
  NobleDevice.call(this, peripheral);
};

var mygatt = {};

NobleDevice.Util.inherits(TestDevice, NobleDevice);

TestDevice.discover(function(testDevice) {
  console.log('found ' + testDevice.id);

  testDevice.on('disconnect', function() {
    console.log('disconnected!');
    //process.exit(0);
  });

  async.series([
      function(callback) {
        console.log('connect');
        testDevice.connect(callback);
      },
      function(callback) {
        console.log('discoverServicesAndCharacteristics');
        testDevice.discoverServicesAndCharacteristics(callback);
      },
      function(callback) {
	// prepare transplant
	mygatt._services = testDevice._peripheral._noble._bindings._gatts[testDevice.id]._services;
	mygatt._characteristics = testDevice._peripheral._noble._bindings._gatts[testDevice.id]._characteristics;
	mygatt._descriptors = testDevice._peripheral._noble._bindings._gatts[testDevice.id]._descriptors;
        console.log('readDeviceName');
        testDevice.readDeviceName(function(error, deviceName) {
          console.log('\tdevice name = ' + deviceName);
          callback();
        });
      },
      function(callback) {
        console.log('disconnect');
        testDevice.disconnect(callback);
      },
      function(callback) {
        console.log('connect');
        testDevice.connect(callback);
      },
      function(callback) {
	// retransplant
	testDevice._peripheral._noble._bindings._gatts[testDevice.id]._services = mygatt._services;
	testDevice._peripheral._noble._bindings._gatts[testDevice.id]._characteristics = mygatt._characteristics;
	testDevice._peripheral._noble._bindings._gatts[testDevice.id]._descriptors = mygatt._descriptors;
	// and test it
        console.log('readDeviceName');
        testDevice.readDeviceName(function(error, deviceName) {
          console.log('\tdevice name = ' + deviceName);
          callback();
        });
      },
      function(callback) {
        //disconnect
	testDevice.disconnect(callback);
      },
      function(callback) {
        //bye
        process.exit();
      }
      ]
  );
});
```

sandeepmistry, 2016-04-11:
@mattgol I'm going to close this for now, looking forward to your PR.

closed 2016-04-11 by sandeepmistry