noble/noble #364: How do I do write-without-response?

billsalt, 2016-03-28:
I have my code working (woohoo), but due to a large amount of data being transmitted, I need to do write-without-response.  Is this possible with Noble @sandeepmistry ?

Note that this would be good functionality as well for DFU (not my application right now, just a future thought) for mass upgrades of IoT devices.

Thanks!
Bill

mattgol, 2016-03-28:
Hi @billsalt - me again.

The Readme uses other wording but in the source file 'characteristic.js' there is following line of code 
```
Characteristic.prototype.write = function(data, withoutResponse, callback) {
...
```
Could this be the right parameter? It's called 'notify' in the Readme.

billsalt, 2016-03-28:
Thanks again.  I'd seen that earlier, but didn't try it.  Looking at the code it does seem that the second parameter is write-without-response, not notify.  On the assumption that the documentation is in error, I tried it (set the parameter 'true') and did indeed get about 10x improvement in performance. 

@sandeepmistry please confirm.

sandeepmistry, 2016-04-03:
@billsalt @mattgol correct! I've corrected this in the read me in https://github.com/sandeepmistry/noble/commit/47b5431242bfe54055094dc2c6e40b9b0d50eeff.

Please submit a PR if you have any other suggestions on documentation.

closed 2016-04-03 by sandeepmistry