noble/noble #370: Read data on Windows 7

yofun, 2016-03-31:
OS: Windows 7
Sensor: TI CC2650 SensorTag
BLE USB Adapter: CSR8510 A10

Hello,
I'm having some trouble for reading data from my sensor on Windows 7. I didn't see anybody using it on Windows 7 so maybe it's not compatible but I still wanted to try...
So, the problem is that I get nothing for the services and characteristics, so I can't get the data from the sensors. At least, that's my conclusion...

```
C:\Users\John\Documents\noble-master\examples>node peripheral-explorer.js b0b448d31185
peripheral with ID b0b448d31185 found
  Local Name        = CC2650 SensorTag
  Manufacturer Data = 0d00030000
  Service Data      =
  Service UUIDs     = aa80

services and characteristics:
```

Thanks in advance!

rpastel, 2016-04-03:
yofun,

Noble does run on Windows 7, but the current Noble version is temporally broken. See my issue 372. If you do not have an older version of Noble on your own machine, download or clone the Noble repository and step back to a version in early Feb.

Good luck

sandeepmistry, 2016-04-03:
Probably related to #372, I'll close this as a duplicate for now. Let's continue the discussion there.

closed 2016-04-03 by sandeepmistry