noble/noble #371: Difference in behavior between MacOS and Raspberry Pi

rvsrvs, 2016-03-31:
I am advertising a custom BLE service on an iPhone. When running noble on a Mac I see the custom service just fine, on the R/Pi, not at all. If necessary I can provide you with an app which advertises the service on iOS as well. The results are as follows

Running the straight peripheral_explorer.js example from the distribution on Mac OS/X and R/Pi at the same time (so that they see the same advertising packets) produces the following output:

On Mac OS/X:

'''
sudo node node_modules/noble/examples/peripheral-explorer.js ce2bcf190ad6425abde6196033e97ba1
peripheral with ID ce2bcf190ad6425abde6196033e97ba1 found
  Local Name        = The VPhone 6
  Service Data      = 
  Service UUIDs     = 7c83ee4efc494598aca239a61d6c1c8e

services and characteristics:
d0611e78bbb44591a5f8487910ae4366
  8667556c9a374c9184ed54ee27d90049
    properties  write, notify, extendedProperties
180f (Battery Service)
  2a19 (Battery Level)
    properties  read, notify
    value       64 | 'd'
1805 (Current Time Service)
  2a2b (Current Time)
    properties  read, notify
    value       e007031d0b0137024b02 | '`
                                         7K'
  2a0f (Local Time Information)
    properties  read
    value       ec04 | 'l'
180a (Device Information)
  2a29 (Manufacturer Name String)
    properties  read
    value       4170706c6520496e632e | 'Apple Inc.'
  2a24 (Model Number String)
    properties  read
    value       6950686f6e65372c32 | 'iPhone7,2'
7c83ee4efc494598aca239a61d6c1c8e
/private/tmp/node_modules/noble/examples/peripheral-explorer.js:102
                            userDescriptionDescriptor.readValue(function(error, data) {
                                                      ^

TypeError: userDescriptionDescriptor.readValue is not a function
    at /private/tmp/node_modules/noble/examples/peripheral-explorer.js:102:55
    at /private/tmp/node_modules/async/dist/async.js:2768:29
    at /private/tmp/node_modules/noble/examples/peripheral-explorer.js:98:34
    at wrappedIteratee (/private/tmp/node_modules/async/dist/async.js:2765:17)
    at replenish (/private/tmp/node_modules/async/dist/async.js:808:21)
    at /private/tmp/node_modules/async/dist/async.js:818:15
    at eachOfLimit (/private/tmp/node_modules/async/dist/async.js:823:28)
    at /private/tmp/node_modules/async/dist/async.js:828:20
    at Object.<anonymous> (/private/tmp/node_modules/async/dist/async.js:2783:17)
    at /private/tmp/node_modules/noble/examples/peripheral-explorer.js:95:29
'''

On Raspberry Pi:

'''
pi@raspberrypi:~/noble $ sudo nodejs node_modules/noble/examples/peripheral-explorer.js ce2bcf190ad6425abde6196033e97ba1
'''

i.e. no discovery whatsoever. This is what I see with my own code as well..

when I turn DEBUG=* on on the Mac:

'''
  yosemite-bindings xpcEvent: {
  "kCBMsgId": 37,
  "kCBMsgArgs": {
    "kCBMsgArgAdvertisementData": {
      "kCBAdvDataIsConnectable": 1,
[... 1601 lines cut ...]
  hci           rssi = -52 +1ms
  gap advertisement = {"manufacturerData":{"type":"Buffer","data":[76,0,12,14,0,138,64,215,70,108,240,66,29,191,107,53,253,239]},"serviceData":[],"serviceUuids":[]} +1ms
  hci onSocketData: 043e0c02010401d64d3a52247100cc +152ms
  hci   event type = 4 +1ms
  hci   sub event type = 62 +0ms
  hci       LE meta event type = 2 +1ms
  hci       LE meta event status = 1 +0ms
  hci       LE meta event data = 0401d64d3a52247100cc +1ms
  hci           type = 4 +1ms
  hci           address = 71:24:52:3a:4d:d6 +0ms
  hci           address type = random +1ms
  hci           eir =  +0ms
  hci           rssi = -52 +0ms
  gap advertisement = {"manufacturerData":{"type":"Buffer","data":[76,0,12,14,0,138,64,215,70,108,240,66,29,191,107,53,253,239]},"serviceData":[],"serviceUuids":[]} +2ms

'''
sudo hcidump -t -x

'''

HCI sniffer - Bluetooth packet analyzer ver 5.23
device: hci0 snap_len: 1500 filter: 0xffffffff
'''

Any suggestions?

sandeepmistry, 2016-04-03:
@rvsrvs peripheral id's are not consistent between OS X and Linux. Even between two OS X systems they will be different.

I suggest you use the ```advertisement-discovery.js``` example to determine the id.

I think it will be ```61618d2f7189``` for your peripheral, on Linux the address is based on the BT address.

closed 2016-04-03 by sandeepmistry

rvsrvs, 2016-04-03:
Not sure i understand this answer.  Are you saying that peripheral-explorer.js should not work on R/Pi?

sandeepmistry, 2016-04-09:
@rvsrvs compare the output of ```node advertisement-discovery.js``` on both OS X and the RPi, the reported id's for discovered peripherals will differ.

I think this is the arg you need for the RPi:

```sh
sudo node node_modules/noble/examples/peripheral-explorer.js 61618d2f7189
```