noble/noble #372: discoverServices does not return services on Window 7

rpastel, 2016-04-02:
OS: Windows 7
Node v4.1.1
BLE Adopter: Asus BT-400
BLE Server device: Wahoo RPM 

On 4/2/2016 installed noble for new project and ran peripheral-explorer.js, but peripheral-explorer.js does not print the services and the script stalls.

I copied node_modules/noble/ from an old install in a project that I made in 3/19/2016 to the the new project that I made on 4/2/2016. Ran peripheral-explorer.js again the script prints services and exits properly. 

Thanks for a great api!

sandeepmistry, 2016-04-03:
@rpastel hmm, there haven't been any related changes to noble in that time frame.

I wonder if is caused by https://github.com/sandeepmistry/node-bluetooth-hci-socket/commit/39af043593335fd985c01d9e9426c25d5447078e.

You could try downgrading the ```bluetooth-hci-socket``` from 0.4.3 to 0.4.2 to confirm this.

A debug log would also be great between both:
```sh
setenv DEBUG=*
node peripheral-explorer.js
```

rpastel, 2016-04-03:
> I wonder if is caused by sandeepmistry/node-bluetooth-hci-socket@39af043

Hello Sandeep,

I have tried to run the tests you suggested. My bash window does not
recognize "setenv", so I had to add DEBUG to my system and user environment
variables using control panel. Running successfully in debug mode requires
rebooting my computer after running a noble example.

I think I ran each test success fully. After running the noble examples as
is from the download, I changed the bluetooth-hci-socket version
specification in package.json from "~0.4.1" to "0.4.2" and ran npm install
again. I think it still fails.

Console output is attached.

Thank you,
Robert

yofun, 2016-04-04:
Hello,

I tried to use a previous version but it still doesn't print the "services and characteristics". Which version did you use exactly?

rpastel, 2016-04-04:
Hello yofun,
I was hasty suggesting using an older version of noble. See Sandeep comments.

I have tested again and examined my projects' package.json. Downloading the current version noble-master installs

noble: 1.3.0
bluetooth-hci-socket: 0.4.3

This does NOT run examples/peripheral-explorer.js.

I deleted node_modules/bluetooth-hci-socket/ directory and edit the package.json to install blue-hci-socket verision 0.4.2. 

noble: 1.3.0
bluetooth-hci-socket: 0.4.2

This DOES runs examples/peripheral-explorer.js.

I should warn you that currently I have to "restart" my windows machine after every run of a noble example program. (My previous tests were confounded because of that.)

If anyone knows how to fix the "restarting" problem, I'd be grateful. I'll try reinstalling the bluetooth adapter driver.

sandeepmistry, 2016-04-11:
@rpastel thanks for experimenting!

I haven't been able to reproduce locally, but I believe I've found the issue and made the following change: https://github.com/sandeepmistry/node-bluetooth-hci-socket/commit/25999ec523325e43d5d900a7ae8866607f56fd9b

Could you please try out replacing  ```bluetooth-hci-socke``` install with ```npm install sandeepmistry/node-bluetooth-hci-socket```.

rpastel, 2016-04-11:
@sandeepmistry 

OS: Windows 7
Node: v4.1.1
BLE Adopter: Asus BT-400
BLE Server device: Wahoo RPM

I made a new project and package.json and then ran:

install -- save noble

I added DEBUG = * to my environment system variables (set DEBUG=* does not work in my console or bash window.)

I ran advertisement-discovery.js and then peripheral-explore.js. The peripheral-explore.js fails (See attached log.)

I uninstalled bluethooth-hci-socket, deleted the entry from noble/package.json, and installed sandeepmistry/node-bluetooth-hci-socket.

I ran peripheral-explore.js. It WORKED!! (See attached log.)

By the way, I have tested noble/examples/ on my Mac with your current version of noble. They work fine. You probably know that. Just thought you'd still like to hear it. 

On unrelated problem. Working on my electron project using noble, occasionally I get this error using my Windows 7 machine:

`healthy_desk-cadence-app\app\node_modules\noble\lib\hci-socket\smp.js:22 Uncaught TypeError: Cannot read property 'split' of undefined `

The program freezes and never makes the connection with peripheral. Terminating and running the program clears up the problem. The problem does not seem to happen on my Mac. 

Should I make another issue?

Thanks for a great API.

[DiscoverServicesTest-2-logs.txt](https://github.com/sandeepmistry/noble/files/212959/DiscoverServicesTest-2-logs.txt)

rpastel, 2016-04-11:
My mistake. I did not mean to close.

sandeepmistry, 2016-04-12:
@rpastel  thanks for trying, I've published v0.4.4 of ```bluetooth-hci-socket``` with the fix (https://github.com/sandeepmistry/node-bluetooth-hci-socket/commit/25999ec523325e43d5d900a7ae8866607f56fd9b). A fresh install should be fine now, please let me if there are any issues.

> On unrelated problem. Working on my electron project using noble, occasionally I get this error using my Windows 7 machine:
>
> healthy_desk-cadence-app\app\node_modules\noble\lib\hci-socket\smp.js:22 [...]

Yes, please. Please include a minimal sample to reproduce, along with debug logs.

closed 2016-04-12 by sandeepmistry

rpastel, 2016-04-12:
@sandeepmistry 
It'll not be until this weekend that I make the issue. The TypeError comes erratically and seems to be related to how the program previously shut down, so it might be hard to reproduce in a small program.