noble/noble #373: Writewithoutresponse doesn't seem to work on Raspian Jessie

billsalt, 2016-04-04:
Same code that runs fine on OS X El Capitan does not run on Raspberry Pi Raspian Jessie.  Trying to isolate, but thought I'd post this now and see if @sandeepmistry or others have suggestions.

Writewithoutresponse to a 16-byte characteristic.

sandeepmistry, 2016-04-09:
@billsalt please provide more information:

1) Minimal code sample that reproduces the issue
2) HCI dump capture: ```sudo hcidump -t -x```

billsalt, 2016-04-12:
[eNode test archive.zip](https://github.com/sandeepmistry/noble/files/215587/eNode.test.archive.zip)

sandeepmistry, 2016-04-18:
@billsalt 

Here are some snippets of interest:

For whatever reason the connection interval is set to ~1s, which is pretty high:

```
2016-04-12 17:20:10.974174 < HCI Command: LE Connection Update (0x08|0x0013) plen 14
  40 00 10 00 20 03 00 00 90 01 00 00 00 00 
2016-04-12 17:20:10.976384 > HCI Event: Command Status (0x0f) plen 4
    LE Connection Update (0x08|0x0013) status 0x00 ncmd 1
2016-04-12 17:20:11.226853 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 1
2016-04-12 17:20:11.289068 > HCI Event: LE Meta Event (0x3e) plen 10
    LE Connection Update Complete
      status 0x00 handle 64
      interval 997.50ms, latency 0.00ms, superv. timeout 4000.00ms
2016-04-12 17:21:24.296565 < ACL data: handle 64 flags 0x00 den 8
```

Write without response is looking good:

```
016-04-12 17:21:26.101389 > ACL data: handle 64 flags 0x02 dlen 5
    ATT: Write resp (0x13)
2016-04-12 17:21:26.102585 < ACL data: handle 64 flags 0x00 dlen 23
    ATT: Write cmd (0x52)
      handle 0x0019 value  0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2016-04-12 17:21:26.103488 < ACL data: handle 64 flags 0x00 dlen 23
    ATT: Write cmd (0x52)
      handle 0x0019 value  0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2016-04-12 17:21:26.104000 < ACL data: handle 64 flags 0x00 dlen 23
    ATT: Write cmd (0x52)
      handle 0x0019 value  0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2016-04-12 17:21:26.104364 < ACL data: handle 64 flags 0x00 dlen 23
    ATT: Write cmd (0x52)
      handle 0x0019 value  0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2016-04-12 17:21:26.104749 < ACL data: handle 64 flags 0x00 dlen 23
    ATT: Write cmd (0x52)
      handle 0x0019 value  0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2016-04-12 17:21:26.105927 < ACL data: handle 64 flags 0x00 dlen 23
    ATT: Write cmd (0x52)
      handle 0x0019 value  0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2016-04-12 17:21:26.106407 < ACL data: handle 64 flags 0x00 dlen 23
    ATT: Write cmd (0x52)
      handle 0x0019 value  0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2016-04-12 17:21:26.106765 < ACL data: handle 64 flags 0x00 dlen 23
    ATT: Write cmd (0x52)
      handle 0x0019 value  0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2016-04-12 17:21:26.107094 < ACL data: handle 64 flags 0x00 dlen 23
    ATT: Write cmd (0x52)
      handle 0x0019 value  0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2016-04-12 17:21:26.107421 < ACL data: handle 64 flags 0x00 dlen 23
    ATT: Write cmd (0x52)
      handle 0x0019 value  0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2016-04-12 17:21:26.107767 < ACL data: handle 64 flags 0x00 dlen 23
    ATT: Write cmd (0x52)
      handle 0x0019 value  0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2016-04-12 17:21:26.108774 < ACL data: handle 64 flags 0x00 dlen 23
    ATT: Write cmd (0x52)
      handle 0x0019 value  0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2016-04-12 17:21:26.109356 < ACL data: handle 64 flags 0x00 dlen 23
    ATT: Write cmd (0x52)
      handle 0x0019 value  0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xe0 0x1f 0xff 0xff 0xff 0xff 0xff 0xff
2016-04-12 17:21:26.110113 < ACL data: handle 64 flags 0x00 dlen 23
    ATT: Write cmd (0x52)
      handle 0x0019 value  0xff 0xff 0xff 0xff 0xff 0x00 0xfc 0x07 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xfe
2016-04-12 17:21:26.110488 < ACL data: handle 64 flags 0x00 dlen 23
    ATT: Write cmd (0x52)
      handle 0x0019 value  0x01 0xf8 0x07 0xff 0x00 0xff 0xff 0xff 0xff 0x8f 0x1f 0xff 0xff 0xff 0xff 0xff
[... 1022 lines cut ...]
      handle 0x0019 value  0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2016-04-12 17:21:26.197301 < ACL data: handle 64 flags 0x00 dlen 23
    ATT: Write cmd (0x52)
      handle 0x0019 value  0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2016-04-12 17:21:26.197629 < ACL data: handle 64 flags 0x00 dlen 23
    ATT: Write cmd (0x52)
      handle 0x0019 value  0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2016-04-12 17:21:26.197801 < ACL data: handle 64 flags 0x00 dlen 23
    ATT: Write cmd (0x52)
      handle 0x0019 value  0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2016-04-12 17:21:26.197959 < ACL data: handle 64 flags 0x00 dlen 23
    ATT: Write cmd (0x52)
      handle 0x0019 value  0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2016-04-12 17:21:26.198117 < ACL data: handle 64 flags 0x00 dlen 23
    ATT: Write cmd (0x52)
      handle 0x0019 value  0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2016-04-12 17:21:26.198287 < ACL data: handle 64 flags 0x00 dlen 23
    ATT: Write cmd (0x52)
      handle 0x0019 value  0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
2016-04-12 17:21:26.198434 < ACL data: handle 64 flags 0x00 dlen 23
    ATT: Write cmd (0x52)
      handle 0x0019 value  0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
```

What's your concern? Is it slow or is data being dropped?

billsalt, 2016-04-18:
Hi Sandeep -- the writes are actually happening -after- I try to disconnect, having given up.  I definitely need to change the connection interval; it should be in the 20-30 mS range.  Yes, this is waaaay slow and that may be the whole issue; I'm not waiting several minutes for it to complete.  On iOS, with a 20mS connection interval I can do the entire write in a couple of seconds.  The fastest I've been able to do it with noble on my Mac is about 1 minute.

Can you provide a method to adjust the bluetooth stack parameters, or is there a work-around?  The interval is absolutely key to the performance.  I see that there have been other requests, but don't see a resolution (maybe I'm missing it).

Thanks!!

billsalt, 2016-04-18:
@sandeepmistry I did some more reading of the history of issues on this with noble.  I don't think that the current behavior is good enough.  My peripheral is set up for the proper connection update, but I don't think it is getting called; as I understand that is optional on the Central side as to when (if ever) it does it.  I think we need to have a way of setting up the defaults, and it would be really nice to expose that in noble if at all possible.

billsalt, 2016-04-20:
@sandeepmistry reading more; there seem to be lots of issues here.  On fix that seems to be working for some is to open an HCI socket and then an LCAP socket.  Question is, does noble work at the same time with bluetooth-hci?

sandeepmistry, 2016-04-25:
@billsalt I think another kernel work around is needed in `bluetooth-hci-socket`, to prevent the kernel from sending:

```
2016-04-12 17:20:10.974174 < HCI Command: LE Connection Update (0x08|0x0013) plen 14
  40 00 10 00 20 03 00 00 90 01 00 00 00 00 
```

as there is nothing in noble that sends this type of command.

billsalt, 2016-04-25:
Thanks @sandeepmistry, but that connection interval update event doesn't look correct at all.  My device is set up to request a connection update with a very short connection interval, but it would appear that something in the stack is not accepting the update.  Somehow this doesn't make sense.  Something is not right with the intervals... 

> #define BLE_GAP_CP_MIN_CONN_INTVL_MIN            0x0006  /**< Lowest mimimum connection interval permitted, in units of 1.25 ms, i.e. 7.5 ms. */
> #define BLE_GAP_CP_MIN_CONN_INTVL_MAX            0x0C80  /**< Highest minimum connection interval permitted, in units of 1.25 ms, i.e. 4 s. */

sandeepmistry, 2016-05-01:
@billsalt yep, I think we are in agreement.

I think if you share a full HCI dump, you'll see a connection update request been sent out by the kernel (there's nothing in noble to do this).

As mentioned in https://github.com/sandeepmistry/noble/issues/373#issuecomment-214076716 we'll have to figure a work around to add in bluetooth-hci-socket. Similar to https://github.com/sandeepmistry/node-bluetooth-hci-socket/commit/c5d7f8455a33d1da3ccd3915dc20a32f38c91182.

billsalt, 2016-05-03:
Thanks @sandeepmistry.  I'll take a better look and compare it with the OS X version which works fine.  I'll be offline for the next 2 weeks on vacation, but will get to it when I return.

sandeepmistry, 2016-06-06:
@billsalt ping ...

billsalt, 2016-06-06:
@sandeepmistry Thanks for the reminder; I was on vacation and needed to work on other projects when I returned.  I should get you something in the next week or so.  Meanwhile, there has been a new version of Raspian that I'll test and see if there is any different behavior.

sandeepmistry, 2016-09-26:
@billsalt any updates on this? Maybe https://github.com/sandeepmistry/noble/issues/293#issuecomment-242659350 has some tips.

billsalt, 2016-09-27:
> @billsalt <https://github.com/billsalt> any updates on this? Maybe #293 (comment) <https://github.com/sandeepmistry/noble/issues/293#issuecomment-242659350> has some tips.

Thanks Sandeep,
  I’ve moved on to other methods for now since I didn’t see any good means of resolving this, but would really like for it to work.  

The resolution in #293 has (as they note) the issue that the handle will change depending on a lot of things.  Could this process be reliably implemented from within noble so that it has a correct handle?

Thanks!
Bill

sandeepmistry, 2016-10-10:
I'm closing this, I think it's a mis-configuration of the peripheral device.

```
2016-04-12 17:20:10.974061 > ACL data: handle 64 flags 0x02 dlen 16
    L2CAP(d): cid 0x0005 len 12 [psm 0]
      12 02 08 00 10 00 20 03 00 00 90 01 
2016-04-12 17:20:10.974156 < ACL data: handle 64 flags 0x00 dlen 10
    L2CAP(d): cid 0x0005 len 6 [psm 0]
      13 02 02 00 00 00 
2016-04-12 17:20:10.974174 < HCI Command: LE Connection Update (0x08|0x0013) plen 14
  40 00 10 00 20 03 00 00 90 01 00 00 00 00 
2016-04-12 17:20:10.976384 > HCI Event: Command Status (0x0f) plen 4
    LE Connection Update (0x08|0x0013) status 0x00 ncmd 1
2016-04-12 17:20:11.226853 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 1
2016-04-12 17:20:11.289068 > HCI Event: LE Meta Event (0x3e) plen 10
    LE Connection Update Complete
      status 0x00 handle 64
      interval 997.50ms, latency 0.00ms, superv. timeout 4000.00ms
```

Here's a breakdown of the L2CAP signalling connection parameter update request from the peripheral.

```
12 02 08 00
10 00 - min connection interval 20 ms
20 03  - max connection interval 1000 ms
00 00 - latency 0 ms
90 01 - supervision timeout 4000 ms
```

So it's requesting between 20ms and 1000 ms, which the Linux kernel agrees to and responds. Then they decide on 997.50ms which is between the range. OS X CoreBluetooth/blued has a different negotiating scheme, but both are out of the control of noble, you should configure the peripheral to have a lower max connection interval.

closed 2016-10-10 by sandeepmistry

billsalt, 2016-10-10:
Hi Sandeep,
  I regret that I haven’t spent further investigation efforts on this.  The need disappeared, but I hope to pick it up in the future.

  The bottom line is that writewithoutresponse is poorly supported and seems to function differently on different platforms due to the specifics on how the OS handles it.  Anything you can do to provide access to configuration parameters will help on this, but I understand there are limitations.

Thank you and keep up the great work!
Bill