noble/noble #374: Intel Edison Systemd and Noble - "noble: unknown peripheral null connected!"

navesta, 2016-04-04:
Hi
Running Pokey 2.1, 159.devkit, with node 4.4.0, am having the following problem: I am able to run my node application from command line , without any problems; however, when I attempt to make a service out of it, I no longer am able to set notifier/read handlers. Specifically, upon calling discoverSomeServicesAndCharacteristics, the handler gets called multiple times. Furthermore, upon connection, the following message is displayed:

`noble: unknown peripheral null connected!`

Here is my service file:
```
[Unit]
Description=node server daemon
After=syslog.target network.target 

[Service]
User=root
Group=root
IOSchedulingClass=realtime
IOSchedulingPriority=0
CPUSchedulingPolicy=fifo
CPUSchedulingPriority=90
Nice=-20
OOMScoreAdjust=-500
ExecStart=/usr/local/bin/npm run hub
WorkingDirectory=/opt/bws/Moon
ExecReload=/bin/kill -HUP $MAINPID
KillMode=process
Restart=always
RestartSec=10s
StandardOutput=syslog
StandardError=syslog
SyslogIdentifier=hub-node

[Install]
WantedBy=multi-user.target
```
Any thoughts?
Cheers,

sandeepmistry, 2016-04-09:
@navesta 

> Any thoughts?

Have you checked to see if only one instance of the script is running?

navesta, 2016-04-11:
It is one instance of node, and here the general layout of the code:

sandeepmistry, 2016-04-18:
@navesta and this is fine without the service?

Would you be able to provide an HCI dump capture? ```hcidump -t -x```

navesta, 2016-04-25:
Correct works fine without service. Here is the hcidump capture; this is a live capture, while 4-5 devices are advertising, and there is an attempt to connect to a device with MAC address "D6:9F:9C:F2:79:D5":
http://pastebin.com/p5tX0tMv

sandeepmistry, 2016-05-01:
I see one connection attempt:

```
2016-04-25 09:33:08.271696 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr D6:9F:9C:F2:79:D5 type 1
    interval 96 window 48 initiator_filter 0
    own_bdaddr_type 0 min_interval 6 max_interval 12
    latency 0 supervision_to 200 min_ce 4 max_ce 6
2016-04-25 09:33:08.272590 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 cmd 1
```

Then a bunch of notifications, is this what should happen? If not could you please try to line up the HCI dump with your code, and annotate what you expect vs what is happening.

sandeepmistry, 2016-05-16:
I'm closing this for now, due to lack of activity. Please re-open with requested info if you are still interested ....

closed 2016-05-16 by sandeepmistry