noble/noble #375: Multiple USB BLE adapters

pdaddy26, 2016-04-06:
Hi, 

I'm planning to connect to multiple sensors, like 20+. So I'll probably have multiple USB BLE dongle connected to the computer to allow for 20+ BLE sensors. 

Do you know if this setup can work with noble? Or any issues that can arise? 

Thanks,

deadprogram, 2016-04-06:
Hi @pdaddy26 as of right now, a Node program that uses Noble can only use a single USB adaptor at a time. You can specify which adaptor to use on Linux as documented at https://github.com/sandeepmistry/noble#multiple-adapters

pdaddy26, 2016-04-07:
So then I should be able run multiple node app and each one with a specific NOBLE_REPORT_ALL_HCI_EVENTS. That should work right?

deadprogram, 2016-04-07:
I've not tried that configuration, it might?

pdaddy26, 2016-04-07:
@sandeepmistry can you tell me if that would work?

sandeepmistry, 2016-04-10:
@deadprogram thanks for helping out!

> can you tell me if that would work?

@pdaddy26 I haven't tried it out, but I don't see any reason why it would not work.

Try it out and let us know :)

closed 2016-04-10 by sandeepmistry

jeberhardt, 2016-04-13:
@pdaddy26, I believe in your above comment you meant to say you'd run each node program with a specific NOBLE_HCI_DEVICE_ID parameter.

I have tried this, and it does work (Raspberry Pi).  However, the CPU usage was so high that it wasn't very feasible.