noble/noble #377: Noble + Bleno: Error with noble rejecting unknown commands

popasquat89, 2016-04-08:
Recently, a change was added to the hci gatt file that rejects commands that are unknown to noble. Anytime that a central goes to discover services on the bleno peripheral, while noble is running on the same device, noble rejects the unknown command. This causes the central that's attempting to discover the services to receive an error, specifically a REQ_NOT_SUPPORTED error.

Commit 6701fc61c5c0f44b948d90f12013ea464baffc80

This is a buzzkill, as we use noble for scanning beacons while at the same time using bleno as a peripheral. Perhaps, we can roll back this commit?

Thanks,
Bryce

sandeepmistry, 2016-04-11:
> Perhaps, we can roll back this commit?

@popasquat89 rolling back is not likely, however we could introduce a flag to disable this behaviour (maybe an environment variable, that bleno sets).

popasquat89, 2016-04-11:
I'll nest the conditional logic under the flag NOBLE_MULTI_ROLE, so that any other conditional logic in the future for making the two work nicely together can be supported without new ENV flags.

Bryce

popasquat89, 2016-04-11:
@sandeepmistry See #379

sandeepmistry, 2016-04-21:
Closed via #379

closed 2016-04-21 by sandeepmistry