noble/noble #378: xpcError: connection interrupted on MacOS

mulderp, 2016-04-10:
Hi,

I am using an Intel Edison with Bleno as peripheral, but when I start scanning for the peripheral with Noble, I get this error:

```
 node simple_write.js
start
scanning...
Starting to scan peripheral
Stopping peripheral scan
found peripheral: { localName: 'edison',
  txPowerLevel: undefined,
  manufacturerData: undefined,
  serviceData: [],
  serviceUuids: [ '12341234123412341234123412341234' ] }
xpcError: connection interrupted
```

The Bleno peripheral stays up:

```
root@edison:~/dev_bleno# node simple_char.js
{ uuid: '12341234123412341234123412341234',
  characteristics:
   [ { uuid: '01010101010101010101010101524742',
       properties: [Object],
       secure: [],
       value: null,
       descriptors: [Object],
       _events: [Object],
       robot: {} } ] }
12341234123412341234123412341234
on -> advertisingStart: success
```

On the MacBook I see the in the menu bar that the laptop paired with the Edison peripheral. Maybe this caused the BLE transmitter of the laptop to hang?

Code for the example is here: https://github.com/embeddednodejs/ch_10_physical_internet/tree/master/bleno_peripheral

sandeepmistry, 2016-04-11:
@mulderp have you followed the bleno setup instructions?  Specifically stopping ```bluetoothd```.

https://github.com/sandeepmistry/bleno#linux

GregDracoulis, 2016-04-18:
This is happening to me too, but only on OSX. When I run OSX as the peripheral and the central with `noble` on Debian, everything works great.

@sandeepmistry it doesn't seem like there is any way to stop `bluetoothd` on OSX. It also doesn't seem like that should be required as far as I know? (see https://github.com/sandeepmistry/bleno#os-x)

I looked into using `launchctl`, which seems to be the OSX equivalent to `systemctl` to stop Bluetooth... but I was only able to locate `com.apple.bluetoothUIServer` which doesn't seem like it would be causing the issue here.

sandeepmistry, 2016-04-24:
@GregDracoulis please open a new issue and provide debug logs and simplified code example of how to reproduce the issue: ```DEBUG=* node <file.js>```

sandeepmistry, 2016-05-02:
I'm closing this due to lack of activity.

closed 2016-05-02 by sandeepmistry