noble/noble #381: RangeError: index out of range

ank-khandelwal, 2016-04-14:
Hi,
I started getting this error recently on raspberry pi 
```
buffer.js:619
    throw new RangeError('index out of range');
    ^
RangeError: index out of range
    at checkOffset (buffer.js:619:11)
    at Buffer.readUInt16LE (buffer.js:665:5)
    at Gap.onHciLeAdvertisingReport (/home/pi/node_modules/noble/lib/hci-socket/gap.js:100:31)
    at emitMany (events.js:108:13)
    at emit (events.js:182:7)
    at Hci.processLeAdvertisingReport (/home/pi/node_modules/noble/lib/hci-socket/hci.js:575:8)
    at Hci.processLeMetaEvent (/home/pi/node_modules/noble/lib/hci-socket/hci.js:534:10)
    at Hci.onSocketData (/home/pi/node_modules/noble/lib/hci-socket/hci.js:429:12)
    at emitOne (events.js:77:13)
    at BluetoothHciSocket.emit (events.js:169:7)
```

Apart from this error I am also not able to read the advertisement  of one of the beacon which you can see in the debug output which caused the error. [Attached]
[debug.txt](https://github.com/sandeepmistry/noble/files/218454/debug.txt)

sandeepmistry, 2016-04-24:
@ank-khandelwal thanks. I think the EIR is invalid, see:

```
02 01 06 - flags 0x06
03 03 aaf1 - Complete List of 16-bit Service Class UUIDs:  0xf1aa
14 02 2032 2d86 2127 00bc fed5 ff38 3a59 2189 0100 00 - Incomplete List of 16-bit Service Class UUID
```

There's an extra 00 or missing byte in the last 16-bit UUID ... what peripheral is it?

ank-khandelwal, 2016-04-25:
Hi, 
It is a custom peripheral with a custom advertising packet. The problem was I didn't set the payload type for the custom data packet. Now I can scan without any problem. Thanks!

closed 2016-04-25 by ank-khandelwal

seshu-varma, 2017-04-09:
hi,
how you get rid of this problem

ank-GML, 2017-04-13:
Hi @seshu-varma , 
As I pointed out it was the custom advertising packet. The BLE advertising data should be packed in the following manner [payload_length] [payload_type] [payload]. I had forgotten to set the payload type, that is why I was not able to read it.

ank-khandelwal, 2017-04-13:
Hi @seshu-varma ,
As I pointed out it was the custom advertising packet. The BLE advertising data should be packed in the following manner [payload_length] [payload_type] [payload]. I had forgotten to set the payload type, that is why I was not able to read it.