noble/noble #382: Uncaught TypeError in hci-socket/smp.js  after closing browser window

rpastel, 2016-04-15:
OS: Windows 7
Node: v4.1.1
Bluetooth Adaptor: Assus BTE400
Bluetooth Sensor: Whaoo RPM
Noble version: "1.3.0"
electron-prebuilt v0.35.4

Closing the a browser window by using the "X" in the upper-right corner of the browser window, frequently causes an "uncaught TypeError"  after running an electron project using noble. 

E:\Data-2012\Research\Healthy Desk\TypeError Debug\healthy_desk-cadence-app\app\node_modules\noble\lib\hci-socket\smp.js:22 Uncaught TypeError: Cannot read property 'split' of undefined

Error only occurs using the "X" in the upper-right corner of the browser window, I have not seen the error if I use "crtl-c" in the command window. Consequently, I do not think that I can make small program demonstrating the error. 

Note that I have not seen the "uncaught TypeError" running a Mac build of the electron project. 

Attached are two log files:

TypeError-HD-Log-2.txt - is with bluetooth-hci-socket version: "0.4.2"
[TypeError-HD-Log-2.txt](https://github.com/sandeepmistry/noble/files/221111/TypeError-HD-Log-2.txt)

TypeError-HD-Log-3.txt - is with bluetooth-hci-socket version: "0.4.4"
[TypeError-HD-Log-3.txt](https://github.com/sandeepmistry/noble/files/221113/TypeError-HD-Log-3.txt)

Both version of Bluetooth-hci-socket produce the error.

Each log contains three 

runs of the program. The first run is the run closing the browser window using the "X" in the upper-right in browser window. The second run of the program is the run that stalls with the "uncaught TypeError". The third run is a successful run after restarting the program.

Both logs have annotation delineating the runs and errors. 

Thank you for a great API.

sandeepmistry, 2016-04-24:
@rpastel would it be possible to create a pure Node.js (no electron) app to reproduce this and then provide debug logs? On first glance it's hard to line up the type error with the debug logs.

sandeepmistry, 2016-05-09:
@rpastel ping

rpastel, 2016-05-09:
Sorry. Have not had time yet.

rpastel, 2016-05-18:
I have been able to get an "Type:Error" using only node script. I run the attached script candenceNotification.js and crtl-c the process early. When I start the script again I get the "TypeError"

Two debug logs are attached which includes the run before crtl-c and after crtl-c.

If I run the script after the error it runs fine.

[cadReconnect.zip](https://github.com/sandeepmistry/noble/files/270947/cadReconnect.zip)

[Uncaught_TypeError-log-1.txt](https://github.com/sandeepmistry/noble/files/270950/Uncaught_TypeError-log-1.txt)
[Uncaught_TypeError-log-2.txt](https://github.com/sandeepmistry/noble/files/270951/Uncaught_TypeError-log-2.txt)

sandeepmistry, 2016-06-19:
@rpastel thanks.

This is really strange:

```
hci-usb reset +0ms
  hci-usb write: 01030c00 +4ms
  noble stateChange poweredOff +3ms
  hci-usb HCI event: 3e130100400000017f8857fcd1f50c00 +2ms
  hci-usb HCI event: 0000c80005 +0ms
  hci onSocketData: 043e130100400000017f8857fcd1f50c000000c80005 +1ms
  hci   event type = 4 +0ms
  hci   sub event type = 62 +0ms
  hci           LE meta event type = 1 +0ms
  hci           LE meta event status = 0 +0ms
  hci           LE meta event data = 400000017f8857fcd1f50c000000c80005 +1ms
  hci                   handle = 64 +0ms
  hci                   role = 0 +1ms
  hci                   address type = random +0ms
  hci                   address = f5:d1:fc:57:88:7f +0ms
  hci                   interval = 15 +0ms
  hci                   latency = 0 +0ms
  hci                   supervision timeout = 2000 +1ms
  hci                   master clock accuracy = 5 +0ms
  hci set scan enabled - writing: 010c20020001 +1ms
  hci-usb write: 010c20020001 +0ms
E:\Data-2012\Research\Healthy Desk\RPM_ED2F-Spike-2\node_modules\noble\lib\hci-socket\smp.js:22
  this._ia = new Buffer(localAddress.split(':').reverse().join(''), 'hex');
```

The HCI stack in `node-bluetooth-chi-socket` has started to reset the local bluetooth adapter (as it's does every startup). However, it looks like a connection event happens during this. I don't see the pairing request though.

During the initialization process, it also reads the local BT adapter address. However, since this has not happened yet, `localAddress` is undefined. I can add a check for this condition, however this doesn't solve the root cause.

Are you by chance using anything that is using noble or the BT adapter at the same time as this is happening?

sandeepmistry, 2016-09-05:
@rpastel ok, sounds like we'll have to do my suggestion from https://github.com/sandeepmistry/noble/issues/382#issuecomment-227027871

> However, since this has not happened yet, localAddress is undefined. I can add a check for this condition, however this doesn't solve the root cause.

Do you think you can try to make this change and submit a pull request?

rpastel, 2016-09-05:
> suggestion from #382 (comment)

Hi Sandeep,

I really do not feel qualified to make the change. Sorry.

Sincerely,
Robert