noble/noble #386: Error when installing noble

murlionline, 2016-04-23:
I just got my Raspberry Pi 3 and installed nodejs/npm. I try to install noble and it throws the below error. Is this specific to Raspberry Pi 3 ?

make: Entering directory '/home/pi/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
  AR(target) Release/obj.target/usb.a
  COPY Release/usb.a
  CXX(target) Release/obj.target/usb_bindings/src/node_usb.o
In file included from ../src/helpers.h:3:0,
                 from ../src/node_usb.h:21,
                 from ../src/node_usb.cc:1:
../../nan/nan.h:328:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
   static const unsigned kReplaceInvalidUtf8 = v8::String::REPLACE_INVALID_UTF8;
                                               ^
usb_bindings.target.mk:93: recipe for target 'Release/obj.target/usb_bindings/src/node_usb.o' failed
make: *** [Release/obj.target/usb_bindings/src/node_usb.o] Error 1
make: Leaving directory '/home/pi/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.1.18-v7+

murlionline, 2016-04-23:
I noticed this was an issue specific to the nodejs which came with my Raspberry PI. I uninstalled it and installed v4.0.0. npm version is 2.14.2. Also when installing nobel, I got errors which I was able to overcome by using the additional command like npm install noble --unsafe-perm -g

sandeepmistry, 2016-04-24:
@murlionline I would suggest you report this to the node-usb dependency repo: https://github.com/nonolith/node-usb

closed 2016-04-24 by sandeepmistry