noble/noble #387: Issues discovering peripherals on raspberry pi 2/bluez 5.38

lessonpix, 2016-04-23:
Hi Sandeep,

First, before I whine, thank you for the great Node.js interface.  I've been so frustrated w/bluez and the dbus interface (through Python) that I rewrote my project with Noble and it feels like a much better fit.  The Node async approach fits what I'm trying to do and Noble seems to be a great design.

That said, I'm still frustrated, this time with something simple: I cannot seem to consistently discover all of my devices.  I have 7 BLE light bulbs and a custom board (Adafruit Feather BlueFruit 32u4) and I cannot (ever) get all 7 bulbs to respond appropriately.  Right now I'm getting 5 (the first 5 it finds) and the last two get
noble warning: unknown peripheral [ADDRESS]

`^Cpi@raspberrypi:~/fadecandy/examples/node $ sudo node blescan.js
b4:99:4c:64:7d:bd
b4:99:4c:64:85:01
b4:99:4c:64:83:35
b4:99:4c:64:84:cb
b4:99:4c:55:38:d1
b4:99:4c:64:83:f4
b4:99:4c:64:85:26
Found > 7 devices, start changing them all
peripheral with ID b4994c647dbd found
  Local Name        = RGBLightOne
peripheral with ID b4994c648501 found
  Local Name        = RGBLightOne
peripheral with ID b4994c648335 found
  Local Name        = RGBLightOne
peripheral with ID b4994c6484cb found
  Local Name        = RGBLightOne
peripheral with ID b4994c5538d1 found
  Local Name        = RGBLightOne
peripheral with ID b4994c6483f4 found
  Local Name        = RGBLightOne
peripheral with ID b4994c648526 found
Changing value
Changing value
Changing value
Changing value
noble warning: unknown peripheral b4994c6483f4
noble warning: unknown peripheral b4994c648526
Changing value
`
Now, I'm fully happy to be told that I am doing something in Node.js wrong (I've not been using it long).  But honestly, I can't see it.  

I've adapted your example and have attached it.  I would LOVE to get through to success here.  I've spent an enormous amount of time on this project and have had little success. :-/

I know that you generally ask for hcidump output so it's attached as well.

Thanks for the help
Bill

[bledump.txt](https://github.com/sandeepmistry/noble/files/233178/bledump.txt)

[blescan.txt](https://github.com/sandeepmistry/noble/files/233171/blescan.txt)

sandeepmistry, 2016-04-25:
@lessonpix 

I think you are hitting the connection limit for your Bluetooth adapter. Do you have a CSR based adapter?
https://github.com/sandeepmistry/noble#maximum-simultaneous-connections

Output of ```lsusb``` would provide this info. Maybe try a Broadcom based adapter? https://github.com/sandeepmistry/noble/wiki/Compatible-devices#linux

```
2016-04-23 22:16:17.315683 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x09 ncmd 1
    Error: Connection Limit Exceeded
2016-04-23 22:16:17.329684 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x09 ncmd 1
    Error: Connection Limit Exceeded
```

lessonpix, 2016-04-25:
It says CSR 4.0 on the plastic - I'm not at my office but will check tomorrow.  

If it is a limit, are there better choices?  Can I install two?  What about the RPi 3's internal BT adapter....

I'll read your notes above and will check out the tomorrow.  Thanks for the response.

lessonpix, 2016-04-25:
Actually, while it's not what I want, it should be possible to connect to each bulb, change the color and disconnect (right?).  When I try, it seems that the disconnect's callback is being called before the disconnect is complete.  Is that right?

sandeepmistry, 2016-05-01:
@lessonpix it would be great it you could share a simplify code example of how to recreate the issue along with a HCI dump capture.

lessonpix, 2016-05-04:
I will take a look at this (again) on Friday - thanks for the response!

deadprogram, 2016-05-10:
Isn't @lessonpix just scanning and not connecting here? The connection limit applies to connections only, AFAIK. @sandeepmistry please correct me here if I'm missing something.

Interesting that @lessonpix can even scan using bluez 5.x, since I cannot even get that far https://github.com/sandeepmistry/noble/issues/348 which might be related to bluez 5 as well.

sandeepmistry, 2016-05-15:
> Isn't @lessonpix just scanning and not connecting here? The connection limit applies to connections only, AFAIK. @sandeepmistry please correct me here if I'm missing something.

@deadprogram from my understanding the issue is not with scanning by attempting to connect to over 5 peripheral simultaneously, I'll rename the issue accordingly.

deadprogram, 2016-05-15:
OK, my bad, I just saw that the blescan.js code was posted and that it was also connecting. Sorry to interject, please ignore.

sandeepmistry, 2016-06-06:
@lessonpix I'm closing this due to lack of activity, please re-open with requested info. if you are still interested.

closed 2016-06-06 by sandeepmistry

lessonpix, 2016-06-29:
Hi Sandeep - I just got my Pi3 and will be looking at this again.  

To clarify, it is Connecting to > 5 that is the problem, scanning is fine.  If I can't get past this limitation, I will need to add a dongle + the internal BT adapter on the Pi3 and work out how to manage two interfaces.  

One way or another I'll get this working this weekend and will update the thread.

lessonpix, 2016-06-30:
So, this should, indeed be closed.

With the new Raspberry Pi 3's builtin bluetooth interface, I have succeeded in connecting to 8 bluetooth bulbs without hitting this issue.  It looks like it is simply a limitation of the dongle I was using.  I suppose it would be nice if there were a more obvious error message, but I'm not complaining - you're help was much appreciated.

Thank you!

mbifulco, 2016-07-27:
Hi @sandeepmistry - is there a way to circumvent the limit for CSR radios?  I'm using a long-range BLE radio for the project I'm working on, and it appears to use a Cambridge Silicon Radio (that is the CSR being referenced, right? 😄 )

Additionally, is there any way to output the HCI device that is being used by noble at runtime? I saw the comment about setting an environment variable, but I'd like to be sure I'm sure.

sandeepmistry, 2016-07-28:
@mbifulco 

> is there a way to circumvent the limit for CSR radios? 

Not that I'm aware of. This is a fixed hardware or firmware limitation. You could add more adapters.

> Additionally, is there any way to output the HCI device that is being used by noble at runtime? I saw the comment about setting an environment variable, but I'd like to be sure I'm sure.

See PR #432, maybe something like `noble._bindings._deviceId` if you use the master version.