noble/noble #388: Move Buffer check when writing to characteristic to bindings level

rclai, 2016-04-24:
I'm trying to integrate `noble` into React Native by simply installing this library and then feeding the Noble constructor with custom bindings, however, the React Native bridge doesn't allow bridging of Buffer objects and must first be sent as base-64 encoded strings to the iOS Objective-C world/Android Java world and be decoded into NSData/[Java Equivalent NSData].

I know I could just pass a Buffer, convert to base-64 string in the custom bindings then send it to the bridge, but that would be unnecessary overhead.

That said, would you be okay with remove the Buffer type check from `Characteristic.prototype.write` and move it to the Buffer check to be at the bindings level?

sandeepmistry, 2016-04-24:
@rclai I'm not too keen about this.

What's your goal for porting noble to React Native? Requiring a base-64 string for data args would introduce incompatibilities, if people wanted to run existing noble code. I think converting buffer's to/from base64 in the binding layer is ok.

rclai, 2016-04-24:
No, I'm not saying to pass data always as base-64 strings. I'm saying to just simply remove [this check](https://github.com/sandeepmistry/noble/blob/67a00d1eddb7339ba5d135f9830d7a79441545a5/lib/characteristic.js#L52-L56) so that you can pass whatever you want. The other bindings will continue to require passing in Buffers.

rclai, 2016-04-24:
Actually, never mind this. I did not have a good understanding of Buffer objects and how they're converted to NSData. Now I understand.

rclai, 2016-04-25:
Hey @sandeepmistry, sorry I'd like to revisit this again but this time it's for a different purpose.

So I figured out how to create a Buffer object in React Native, however, I found out that React Native doesn't have a global `Buffer` object, so every time it hits this [code](https://github.com/sandeepmistry/noble/blob/67a00d1eddb7339ba5d135f9830d7a79441545a5/lib/characteristic.js#L52-L56), it breaks.

So I had to overwrite the prototype function to not check for it like [this](https://github.com/Profab/react-native-ble/commit/cffe68d430b9a1649e4122250bd7961b60febce2). I am not fond of this, but would you be open to adding a `Buffer` shim at the top of the `characteristic.js` file which will do something like this:

```js
var Buffer = (window || global).Buffer || require('buffer').Buffer;
```

rclai, 2016-04-26:
Ping @sandeepmistry.

sandeepmistry, 2016-04-28:
I'm not opposed to merging this, but we should take a look at the bigger picture.

Is there any way the Reactive Native bride could inject Buffer into global/window scope? This would be a big win for the existing packages that depend on noble: https://www.npmjs.com/browse/depended/noble -  they in theory would work as is.

cc/ @jacobrosenthal

jacobrosenthal, 2016-04-29:
I think the issue is the the react package manager can only alias with the browser keyword 1 level deep, unlike the webpack which will swap packages n-deep. the Please upvote both of these:
https://productpains.com/post/react-native/packager-support-resolvealias-ala-webpack/ https://github.com/facebook/react-native/issues/6253

sandeepmistry, 2016-05-09:
@jacobrosenthal upvoted. I'm going to close this for now.

closed 2016-05-09 by sandeepmistry

windhost, 2016-09-02:
@rclai I meet the same problem, could you tell me how to create a Buffer object in React Native?

windhost, 2016-09-02:
@rclai I solved it ,by 

 `var Buffer = (window || global).Buffer || require('buffer').Buffer;`

thanks a lot