noble/noble #392: 'discover' not seeing manufacturerData on OSX Yosemite

farfromrefug, 2016-04-28:
Hi,

I am building an few apps around BLE:
- BLE server on rPi using bleno
- iOS / Android native app to connect to that BLE device
- electron app for OSX / Windows using noble

Now everything works perfectly between the rPi and iOS / Android
I have an issue with noble though.
When discovering the devices noble does not report the ```manufacturerData``` on OSX
Though i see it well from And and iOS (which as a very similar code to noble).

on mi Pi i see that i create the adv correctly
```
creating BLEDevice 12334 12wef3
createData: {
  "name": "NeOsmia 12wef3",
  "manufacturerData": [
    3,
    186
  ],
  "services": [
    "6e400001b5a3f393e0a9e50e24dcca9e"
  ]
}
adv [ <Buffer 02 01 06 11 06 9e ca dc 24 0e e5 a9 e0 93 f3 a3 b5 01 00 40 6e>,
  <Buffer 0f 08 4e 65 4f 73 6d 69 61 20 31 32 77 65 66 33 03 ff 03 ba> ]
on -> stateChange: poweredOn true
startAdvertising [ <Buffer 02 01 06 11 06 9e ca dc 24 0e e5 a9 e0 93 f3 a3 b5 01 00 40 6e>,
  <Buffer 0f 08 4e 65 4f 73 6d 69 61 20 31 32 77 65 66 33 03 ff 03 ba> ]
on -> advertisingStart: success
services [ '6e400001b5a3f393e0a9e50e24dcca9e' ]
```
i use
```
                bleno.startAdvertisingWithEIRData(this.advertisement[0], this.advertisement[1]);
```

Now on noble when get the ```discover``` event i get this:
```
parseAdvertisingData { localName: 'raspberrypi',
  txPowerLevel: undefined,
  manufacturerData: undefined,
  serviceData: [],
  serviceUuids: [ '6e400001b5a3f393e0a9e50e24dcca9e' ] }
```
Where manufacturerData is undefined.

Do you have any idea?

Thanks

farfromrefug, 2016-04-28:
Actually just found something really weird. Using your examples, look what i see
```
peripheral discovered (3dfcac45fee74f3f8176f9e6b409a746 with address <5c:f3:70:68:17:2a, unknown>, connectable true, manufacturerData undefined, RSSI -59:
	hello my local name is:
		raspberrypi
	can I interest you in any of the following advertised services:
		["6e400001b5a3f393e0a9e50e24dcca9e"]

peripheral discovered (3dfcac45fee74f3f8176f9e6b409a746 with address <5c:f3:70:68:17:2a, unknown>, connectable true, manufacturerData �, RSSI -56:
	hello my local name is:
		raspberrypi
	can I interest you in any of the following advertised services:
		["6e400001b5a3f393e0a9e50e24dcca9e"]
	here is my manufacturer data:
		"03ba"
```
The second event (which i don't know when it actually happened, i did not restart the device) show the correct manufacturerData ...

farfromrefug, 2016-04-28:
Found the bug.
If i start ```advertisement-discovery.js```  with my mac's bluetooth on i get this 
```
peripheral discovered (3dfcac45fee74f3f8176f9e6b409a746 with address <5c:f3:70:68:17:2a, unknown>, connectable true, manufacturerData undefined, RSSI -59:
    hello my local name is:
        raspberrypi
    can I interest you in any of the following advertised services:
        ["6e400001b5a3f393e0a9e50e24dcca9e"]
```
if i then stop the bluetooth and start it again i get this
```
peripheral discovered (3dfcac45fee74f3f8176f9e6b409a746 with address <5c:f3:70:68:17:2a, unknown>, connectable true, manufacturerData �, RSSI -56:
    hello my local name is:
        raspberrypi
    can I interest you in any of the following advertised services:
        ["6e400001b5a3f393e0a9e50e24dcca9e"]
    here is my manufacturer data:
        "03ba"
```
It seems that for some reason if noble is started with the bluetooth on, i don't get the manufacturerData
Any idea?

sandeepmistry, 2016-05-01:
@farfromrefug could you please shared debug output: ```DEBUG=* node <file>.js```

I suspect blued is not sending the message through ...

farfromrefug, 2016-05-02:
[Here](http://pastebin.com/teWdyVtV) is the log.
It appears noble sees the manufacturerData but dont forward it

sandeepmistry, 2016-05-09:
@farfromrefug I think the best way to move forward is if you dive into the code and add some more long statements. Areas of interest would be:

 * https://github.com/sandeepmistry/noble/blob/master/lib/noble.js#L118-L123
 * https://github.com/sandeepmistry/noble/blob/master/lib/mac/yosemite.js#L128-L191

If you share the bleno code used on the Raspberry Pi, I add it to my (every growing) list of things to look at.

farfromrefug, 2016-05-11:
i created a PR for that issue. Can you look at it?
https://github.com/sandeepmistry/noble/pull/395

brandon-collins, 2017-05-16:
It would be nice if this was patched. I ran into the same issue today. I figured out how to patch it in my code but it would be nice to have this lib patched.

stevelibuzz, 2018-05-08:
I have the same situation, is it possible to merge the PR?