noble/noble #394: peripheral disappears from 'discover' after being connected..

nicwsm, 2016-05-05:
Hi,

I have a scenario now where I have a sensor in the peripheral+broadcaster role, meaning to say that my sensor will continue to emit iBeacon packets even after it's being connected to a central.

When this sensor is not connected to any central, my scanning device (Intel Edison) has no issue to read the iBeacon packets transmitted from this sensor using noble. However, once this sensor is connected to another BLE central, somehow the noble.on('discover') failed to read the iBeacon packets anymore. 

I have confirmed using my phone iBeacon tracking App that the sensor is actually still transmitting the iBeacon packets, but in a reduced rate. But using noble, I cannot detect it at all. Any explanation on this? Please advise. Thanks!

sandeepmistry, 2016-05-09:
@nicwsm could you please share your code for scanning as well as an HCI dump? `hcidump -t -x

nicwsm, 2016-05-10:
@sandeepmistry I found out what's the issue. Apparently this code in gap.js is causing the iBeacon advertising packets to be muted from the events.

if (type === 0x04 || (discoveryCount > 1 && !hasScanResponse) || process.env.NOBLE_REPORT_ALL_HCI_EVENTS)

If I run my code enabling NOBLE_REPORT_ALL_HCI_EVENTS=1 then it works. Do you mind elaborating why is that IF condition added in? Thanks!

sandeepmistry, 2016-05-15:
> Do you mind elaborating why is that IF condition added in?

noble is waiting for both an advertisement discovery event and scan response. It looks like your device does not have a scan response.

It might be better to have a timer, so that if a X milliseconds pass without a scan response, report the device discovery event ...