noble/noble #397: Peripheral address

joustava, 2016-05-12:
I am interested in using this library, so I had a quick look at the discovery example. Basically I edited like this:
```
var noble = require('noble');

noble.on('stateChange', function(state) {
  if (state === 'poweredOn') {
    noble.startScanning();
  } else {
    noble.stopScanning();
  }
});

noble.on('discover', function(peripheral) {

  console.log('Address', peripheral._noble.address);

  console.dir(peripheral, {depth: null, colors: true});
  
  console.log();
});
```

This seems to work just fine and discovers the BLE devices I have configured.
However, I noticed that `peripheral.address` is not set although noble seems to know about it, as shown in the output below (from my setup)

```
Address 00:1a:7d:da:71:13
Peripheral {
  _noble:
   Noble {
     state: 'poweredOn',
     address: '00:1a:7d:da:71:13',
     _bindings:
      EventEmitter {
        _peripherals:
         { a349f63a36b14079be845cf0752f0fcf:
            { uuid:
               Buffer {
                 '0': 163,
                 '1': 73,
                 '2': 246,
                 '3': 58,
                 '4': 54,
                 '5': 177,
                 '6': 64,
                 '7': 121,
                 '8': 190,
                 '9': 132,
                 '10': 92,
                 '11': 240,
                 '12': 117,
                 '13': 47,
                 '14': 15,
                 '15': 207,
                 isUuid: true },
              connectable: true,
              advertisement:
               { localName: undefined,
                 txPowerLevel: undefined,
                 manufacturerData:
                  Buffer {
                    '0': 76,
                    '1': 0,
                    '2': 9,
                    '3': 6,
                    '4': 3,
                    '5': 24,
                    '6': 192,
                    '7': 168,
                    '8': 222,
                    '9': 106 },
                 serviceData: [],
                 serviceUuids: [] },
              rssi: -65,
              address: 'unknown',
              addressType: 'unknown' } },
        _xpcConnection:
         XpcConnection {
           _events: { error: [Function: bound ], event: [Function: bound ] },
           _eventsCount: 2 },
        _events:

       ...removed the rest 
```

Is this a bug or am I missing something? I am running this on OSX El Capitan v 10.11.4.

sandeepmistry, 2016-05-15:
> Is this a bug or am I missing something? I am running this on OSX El Capitan v 10.11.4.

`noble.address` is for the local (central) BT address.

`peripheral.address` is for the remote peripheral address. In your case it is set to "unknown". On OS X this is expected behaviour until a connection has been established. After this an non-unknown value should be reported the next time.

```
address: 'unknown',
addressType: 'unknown'
```

closed 2016-05-15 by sandeepmistry

joustava, 2016-05-16:
Thanks, confusion arose because of having dongles on central and peripheral with the same `BD Address`.