noble/noble #398: Node 6 issue

nemanja-tosic, 2016-05-13:
Hello,

When i run the module i get the following:

==== C stack trace ===============================

(empty)
(node) v8::ObjectTemplate::Set() with non-primitive values is deprecated
(node) and will stop working in the next major release.
2016-05-12 20:46:01: App name:supernode-supervisor id:2 online

==== JS stack trace =========================================

Security context: 0x5c22b8a1 <JS Object>#0#
    1: .node [module.js:568] [pc=0x423fdc64] (this=0x50b1abb9 <an Object with map 0x34924641>#1#,module=0x3b2e79dd <a Module with map 0x3492492d>#2#,filename=0x3b2e79c1 <String[94]: /root/node-hub/node_modules/noble/node_modules/bluetooth-hci-socket/build/Release/binding.node>)
    2: load [module.js:456] [pc=0x33dc3398] (this=0x3b2e79dd <a Module with map 0x3492492d>#2#,filename=0x3b2e79c1 <String[94]: /root/node-hub/node_modules/noble/node_modules/bluetooth-hci-socket/build/Release/binding.node>)
    3: tryModuleLoad(aka tryModuleLoad) [module.js:415] [pc=0x33dc2e88] (this=0x5c20809d <undefined>,module=0x3b2e79dd <a Module with map 0x3492492d>#2#,filename=0x3b2e79c1 <String[94]: /root/node-hub/node_modules/noble/node_modules/bluetooth-hci-socket/build/Release/binding.node>)
    4: /* anonymous */ [module.js:407] [pc=0x33dbeb54] (this=0x50b1abf9 <JS Function Module (SharedFunctionInfo 0x31e9f2a5)>#3#,request=0x3d7fc3f1 <String[29]: ../build/Release/binding.node>,parent=0x3b2e5f91 <a Module with map 0x3492492d>#4#,isMain=0x5c2081f5 <false>)
    5: _load [/root/node-hub/node_modules/pmx/lib/transaction.js:62] [pc=0x42348270] (this=0x50b1abf9 <JS Function Module (SharedFunctionInfo 0x31e9f2a5)>#3#,file=0x3d7fc3f1 <String[29]: ../build/Release/binding.node>)
    6: arguments adaptor frame: 3->1
    7: require [module.js:~463] [pc=0x423772c4] (this=0x3b2e5f91 <a Module with map 0x3492492d>#4#,path=0x3d7fc3f1 <String[29]: ../build/Release/binding.node>)
    8: require(aka require) [internal/module.js:20] [pc=0x33dccd8c] (this=0x5c20809d <undefined>,path=0x3d7fc3f1 <String[29]: ../build/Release/binding.node>)
    9: /* anonymous */ [/root/node-hub/node_modules/noble/node_modules/bluetooth-hci-socket/lib/native.js:3] [pc=0x423fd900] (this=0x3b2e5fb9 <an Object with map 0x3490933d>#5#,exports=0x3b2e5fb9 <an Object with map 0x3490933d>#5#,require=0x3b2e6b51 <JS Function require (SharedFunctionInfo 0x31ed4ed9)>#6#,module=0x3b2e5f91 <a Module with map 0x3492492d>#4#,__filename=0x3b2e5f75 <String[81]: /root/node-hub/node_modules/noble/node_modules/bluetooth-hci-socket/lib/native.js>,__dirname=0x3b2e6b1d <String[71]: /root/node-hub/node_modules/noble/node_modules/bluetooth-hci-socket/lib>)
   10: _compile [module.js:541] [pc=0x33dcaeb0] (this=0x3b2e5f91 <a Module with map 0x3492492d>#4#,content=0x3b2e6649 <String[378]\: var events = require('events');\n\nvar binding = require('../build/Release/binding.node');\nvar BluetoothHciSocket = binding.BluetoothHciSocket;\n\ninherits(BluetoothHciSocket, events.EventEmitter);\n\n// extend prototype\nfunction inherits(target, source) {\n  for (var k in source.prototype) {\n    target.prototype[k] = source.prototype[k];\n  }\n}\n\nmodule.exports = BluetoothHciSocket;\n>,filename=0x3b2e5f75 <String[81]: /root/node-hub/node_modules/noble/node_modules/bluetooth-hci-socket/lib/native.js>)
   11: .js [module.js:550] [pc=0x33dc4bb4] (this=0x50b1abb9 <an Object with map 0x34924641>#1#,module=0x3b2e5f91 <a Module with map 0x3492492d>#4#,filename=0x3b2e5f75 <String[81]: /root/node-hub/node_modules/noble/node_modules/bluetooth-hci-socket/lib/native.js>)
   12: load [module.js:456] [pc=0x33dc3398] (this=0x3b2e5f91 <a Module with map 0x3492492d>#4#,filename=0x3b2e5f75 <String[81]: /root/node-hub/node_modules/noble/node_modules/bluetooth-hci-socket/lib/native.js>)
   13: tryModuleLoad(aka tryModuleLoad) [module.js:415] [pc=0x33dc2e88] (this=0x5c20809d <undefined>,module=0x3b2e5f91 <a Module with map 0x3492492d>#4#,filename=0x3b2e5f75 <String[81]: /root/node-hub/node_modules/noble/node_modules/bluetooth-hci-socket/lib/native.js>)
   14: /* anonymous */ [module.js:407] [pc=0x33dbeb54] (this=0x50b1abf9 <JS Function Module (SharedFunctionInfo 0x31e9f2a5)>#3#,request=0x3d7fbf6d <String[12]: ./lib/native>,parent=0x3b2e4839 <a Module with map 0x3492492d>#7#,isMain=0x5c2081f5 <false>)
   15: _load [/root/node-hub/node_modules/pmx/lib/transaction.js:62] [pc=0x42348270] (this=0x50b1abf9 <JS Function Module (SharedFunctionInfo 0x31e9f2a5)>#3#,file=0x3d7fbf6d <String[12]: ./lib/native>)
   16: arguments adaptor frame: 3->1
   17: require [module.js:~463] [pc=0x423772c4] (this=0x3b2e4839 <a Module with map 0x3492492d>#7#,path=0x3d7fbf6d <String[12]: ./lib/native>)
   18: require(aka require) [internal/module.js:20] [pc=0x33dccd8c] (this=0x5c20809d <undefined>,path=0x3d7fbf6d <String[12]: ./lib/native>)
   19: /* anonymous */ [/root/node-hub/node_modules/noble/node_modules/bluetooth-hci-socket/index.js:8] [pc=0x423fd5c8] (this=0x3b2e4861 <an Object with map 0x3490933d>#8#,exports=0x3b2e4861 <an Object with map 0x3490933d>#8#,require=0x3b2e5301 <JS Function require (SharedFunctionInfo 0x31ed4ed9)>#9#,module=0x3b2e4839 <a Module with map 0x3492492d>#7#,__filename=0x3b2e481d <String[76]: /root/node-hub/node_modules/noble/node_modules/bluetooth-hci-socket/index.js>,__dirname=0x3b2e52cd <String[67]: /root/node-hub/node_modules/noble/node_modules/bluetooth-hci-socket>)
   20: _compile [module.js:541] [pc=0x33dcaeb0] (this=0x3b2e4839 <a Module with map 0x3492492d>#7#,content=0x3b2e4e85 <String[307]\: var os = require('os');\n\nvar platform = os.platform();\n\nif (process.env.BLUETOOTH_HCI_SOCKET_FORCE_USB || platform === 'win32') {\n  module.exports = require('./lib/usb.js');\n} else if (platform === 'linux') {\n  module.exports = require('./lib/native');\n} else {\n  throw new Error('Unsupported platform');\n}\n>,filename=0x3b2e481d <String[76]: /root/node-hub/node_modules/noble/node_modules/bluetooth-hci-socket/index.js>)
   21: .js [module.js:550] [pc=0x33dc4bb4] (this=0x50b1abb9 <an Object with map 0x34924641>#1#,module=0x3b2e4839 <a Module with map 0x3492492d>#7#,filename=0x3b2e481d <String[76]: /root/node-hub/node_modules/noble/node_modules/bluetooth-hci-socket/index.js>)
   22: load [module.js:456] [pc=0x33dc3398] (this=0x3b2e4839 <a Module with map 0x3492492d>#7#,filename=0x3b2e481d <String[76]: /root/node-hub/node_modules/noble/node_modules/bluetooth-hci-socket/index.js>)
   23: tryModuleLoad(aka tryModuleLoad) [module.js:415] [pc=0x33dc2e88] (this=0x5c20809d <undefined>,module=0x3b2e4839 <a Module with map 0x3492492d>#7#,filename=0x3b2e481d <String[76]: /root/node-hub/node_modules/noble/node_modules/bluetooth-hci-socket/index.js>)
   24: /* anonymous */ [module.js:407] [pc=0x33dbeb54] (this=0x50b1abf9 <JS Function Module (SharedFunctionInfo 0x31e9f2a5)>#3#,request=0x3d7e7f05 <String[20]: bluetooth-hci-socket>,parent=0x3b2d77a5 <a Module with map 0x3492492d>#10#,isMain=0x5c2081f5 <false>)
   25: _load [/root/node-hub/node_modules/pmx/lib/transaction.js:62] [pc=0x42348270] (this=0x50b1abf9 <JS Function Module (SharedFunctionInfo 0x31e9f2a5)>#3#,file=0x3d7e7f05 <String[20]: bluetooth-hci-socket>)
   26: arguments adaptor frame: 3->1
   27: require [module.js:~463] [pc=0x423772c4] (this=0x3b2d77a5 <a Module with map 0x3492492d>#10#,path=0x3d7e7f05 <String[20]: bluetooth-hci-socket>)
   28: require(aka require) [internal/module.js:20] [pc=0x33dccd8c] (this=0x5c20809d <undefined>,path=0x3d7e7f05 <String[20]: bluetooth-hci-socket>)
   29: /* anonymous */ [/root/node-hub/node_modules/noble/lib/hci-socket/hci.js:6] [pc=0x423fbaa0] (this=0x3b2d77cd <an Object with map 0x3490933d>#11#,exports=0x3b2d77cd <an Object with map 0x3490933d>#11#,require=0x3b2e129d <JS Function require (SharedFunctionInfo 0x31ed4ed9)>#12#,module=0x3b2d77a5 <a Module with map 0x3492492d>#10#,__filename=0x3b2d7789 <String[55]: /root/node-hub/node_modules/noble/lib/hci-socket/hci.js>,__dirname=0x3b2e1269 <String[48]: /root/node-hub/node_modules/noble/lib/hci-socket>)
   30: _compile [module.js:541] [pc=0x33dcaeb0] (this=0x3b2d77a5 <a Module with map 0x3492492d>#10#,content=0x3b2d7db5 <Very long string[18135]>#13#,filename=0x3b2d7789 <String[55]: /root/node-hub/node_modules/noble/lib/hci-socket/hci.js>)
   31: .js [module.js:550] [pc=0x33dc4bb4] (this=0x50b1abb9 <an Object with map 0x34924641>#1#,module=0x3b2d77a5 <a Module with map 0x3492492d>#10#,filename=0x3b2d7789 <String[55]: /root/node-hub/node_modules/noble/lib/hci-socket/hci.js>)
[... cut ...]
  105: emit [events.js:188] [pc=0x33d9511c] (this=0x3d71b3cd <a WriteStream with map 0x34941de1>#38#,type=0x31e7845d <String[4]: open>)
  106: arguments adaptor frame: 2->1
  107: /* anonymous */ [fs.js:1843] [pc=0x42372388] (this=0x3d71b3cd <a WriteStream with map 0x34941de1>#38#,er=0x5c208089 <null>,fd=14)
  108: oncomplete [fs.js:117] [pc=0x42374b00] (this=0x3d71b4b1 <an FSReqWrap with map 0x34941d05>#40#)
  109: arguments adaptor frame: 2->0

=====================

Im running node nodejs 6.1.0 on archlinux ARM.

Thanks,
Nemanja

sandeepmistry, 2016-08-22:
@nemanja-tosic does this still happen with the latest Node.js 6 release?

If so please close this issue and open again in https://github.com/sandeepmistry/node-bluetooth-hci-socket

sandeepmistry, 2016-09-17:
Closing this due to lack of activity.

closed 2016-09-17 by sandeepmistry