noble/noble #402: Discover delay between duplicate

jbeaudoin11, 2016-05-20:
Hi !

I wanna no if there is a way to configure the discover time for the same device ?
My goal is to detect when the device enter and leave the area, like one of the example in the git.

For now it's pretty inconsistent...

```
id : d7f9135ed7d4 rssi : -78 delay : 5035
id : d7f9135ed7d4 rssi : -78 delay : 20091
id : d7f9135ed7d4 rssi : -77 delay : 2009
id : d7f9135ed7d4 rssi : -79 delay : 3019
id : d7f9135ed7d4 rssi : -78 delay : 1008
id : d7f9135ed7d4 rssi : -76 delay : 2015
id : d7f9135ed7d4 rssi : -76 delay : 2007
id : d7f9135ed7d4 rssi : -77 delay : 2009
id : d7f9135ed7d4 rssi : -78 delay : 4012
id : d7f9135ed7d4 rssi : -76 delay : 2017
id : d7f9135ed7d4 rssi : -79 delay : 3015
id : d7f9135ed7d4 rssi : -79 delay : 13056
```

 As you can see it's between 1s to 5s in general.. sometime it's higher. It's not really viable for my use :/

sandeepmistry, 2016-05-24:
@jbeaudoin11 why not lower the grace period?

https://github.com/sandeepmistry/noble/blob/master/examples/enter-exit.js#L12

You could also try changing the scan parameters: https://github.com/sandeepmistry/noble/blob/master/lib/hci-socket/hci.js#L238-L242

Btw, you might also want to try decreasing the advertising interval on the peripheral (device).

closed 2016-05-24 by sandeepmistry

jbeaudoin11, 2016-05-24:
There was no grace period in my test.

I read more about the subject and the problem is the advertising interval like you said !

Unfortunately we have no control on the ble device we bought (was not a good choice) :/

All good from your lib ! Thanks.

sandeepmistry, 2016-05-24:
@jbeaudoin11 cool, thanks for the update.