noble/noble #403: Multiple connection from 2 different processes with cc2650

Lahorde, 2016-05-20:
I'm trying to connect multiple peripheral and I encountered several issues. 

First test : try to connect 2 sensortag V2
Steps : 
 * launch a modified noble-device/examples/basic.js (remove disconnection)
 * start first sensortag v2: it connects successfully and service discovery ok
 * launch a modified noble-device/examples/basic.js (remove disconnection)
 * start second sensortag : it connects successfully but service discovery fails.

I've sniffed ble packets with wireshark, in second connection, 2 MTU responses are received, consequently, first is correctly handled from gatt.js command queue, but second MTU response data is set to next popped command data in gatt.js command queue, consequently, First 'read by group type request' response data is set to MTU response data and discovery stops for second device as GATT is unable to parse next handles to request;

I think opcode must be checked when receiving in order to check received data corresponds to request.

I've attached wiresharks logs.

[ble_dual_connection.pcapng.gz](https://github.com/sandeepmistry/noble/files/274821/ble_dual_connection.pcapng.gz)

sandeepmistry, 2016-06-20:
@Lahorde would it be possible for you to provided an HCI dump capture using `hcidump -t -x`?

jasoncodes, 2016-08-03:
I’ve ran into this same problem trying to connect to two BTLE tags from seperate processes. I can connect to both with the same noble instance fine.

linux 4.4.13+ (raspberry pi)
bluez 5.23-2+rpi2
noble 1.6.0
CSR8510

Test script: [test.js](https://github.com/sandeepmistry/noble/files/400461/test.js.txt)

Test procedure:

1. Start `hcidump -t -x`
2. Start first instance of test script
3. Turn on first tag
4. Start second instance of test script
5. Turn on second tag

Test script output 1:

```
poweredOn
connecting to ffff0000e80f
connected to ffff0000e80f
9 services
25 characteristics
```

Test script output 2:

```
poweredOn
connecting to ffff0000b802
connected to ffff0000b802
```

The second script never receives the list of services and characteristics. No error is displayed.

HCP dump output:

```
HCI sniffer - Bluetooth packet analyzer ver 5.23
device: hci0 snap_len: 1500 filter: 0xffffffff
2016-08-04 08:47:47.428397 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2016-08-04 08:47:47.438387 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2016-08-04 08:47:47.458402 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0x22bb
    LMP Version: 4.0 (0x6) LMP Subversion: 0x22bb
    Manufacturer: Cambridge Silicon Radio (10)
2016-08-04 08:47:47.460411 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    00 
2016-08-04 08:47:47.461391 > HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
    00 01 00 
2016-08-04 08:47:47.466418 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr 00:1A:7D:DA:71:0A
2016-08-04 08:47:47.537418 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2016-08-04 08:47:47.542411 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2016-08-04 08:47:47.702414 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-08-04 08:47:58.185608 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
2016-08-04 08:47:58.451665 > HCI Event: Command Status (0x0f) plen 4
    LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 0
2016-08-04 08:47:58.499645 > HCI Event: Command Status (0x0f) plen 4
    Unknown (0x00|0x0000) status 0x00 ncmd 1
2016-08-04 08:47:58.510626 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-08-04 08:48:06.464532 > HCI Event: Command Status (0x0f) plen 4
    LE Connection Update (0x08|0x0013) status 0x00 ncmd 1
2016-08-04 08:48:10.811048 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2016-08-04 08:48:10.819047 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2016-08-04 08:48:10.826063 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0x22bb
    LMP Version: 4.0 (0x6) LMP Subversion: 0x22bb
    Manufacturer: Cambridge Silicon Radio (10)
2016-08-04 08:48:10.834068 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    00 
2016-08-04 08:48:10.840061 > HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
    00 01 00 
2016-08-04 08:48:10.847063 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr 00:1A:7D:DA:71:0A
2016-08-04 08:48:10.909067 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2016-08-04 08:48:10.919067 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2016-08-04 08:48:11.079062 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-08-04 08:48:15.066526 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
2016-08-04 08:48:15.322561 > HCI Event: Command Status (0x0f) plen 4
    LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 1
2016-08-04 08:48:15.329564 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-08-04 08:48:23.342441 > HCI Event: Command Status (0x0f) plen 4
    LE Connection Update (0x08|0x0013) status 0x00 ncmd 1
```

sandeepmistry, 2016-09-05:
Hi @jasoncodes,

Is there anything after this?

```
2016-08-04 08:48:23.342441 > HCI Event: Command Status (0x0f) plen 4
    LE Connection Update (0x08|0x0013) status 0x00 cmd 1
```

I'm not sure if the capture is cut off or if there are no further events ...

jasoncodes, 2016-09-05:
I am pretty sure that’s the complete capture. That was the complete capture up until the end of the test script output. I left it a few seconds then stopped capturing before I killed both processes. If this output is not making sense I can run the test again.

sandeepmistry, 2016-09-05:
@jasoncodes yes please, I'm confused why `LE Connection Update` is the last event. Thanks.

jasoncodes, 2016-09-07:
I don’t have the same two devices I used last time (one seems to have gone kaput) but have reproduced the issue this time using a heart rate monitor.

The last few lines of output before I kill the test processes is the same. As before, running either test app individually works fine.

<details>
Start first app:
```
HCI sniffer - Bluetooth packet analyzer ver 5.23
device: hci0 snap_len: 1500 filter: 0xffffffff
2016-09-08 08:46:11.993826 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2016-09-08 08:46:12.007821 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2016-09-08 08:46:12.024845 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0x22bb
    LMP Version: 4.0 (0x6) LMP Subversion: 0x22bb
    Manufacturer: Cambridge Silicon Radio (10)
2016-09-08 08:46:12.026816 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    00 
2016-09-08 08:46:12.037807 > HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
    00 01 00 
2016-09-08 08:46:12.042807 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr 00:1A:7D:DA:71:0A
2016-09-08 08:46:12.136820 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2016-09-08 08:46:12.142814 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2016-09-08 08:46:12.324844 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
```
Turn on first device:
```
2016-09-08 08:46:19.198643 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
2016-09-08 08:46:19.458672 > HCI Event: Command Status (0x0f) plen 4
    LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 0
2016-09-08 08:46:19.544667 > HCI Event: Command Status (0x0f) plen 4
    Unknown (0x00|0x0000) status 0x00 ncmd 1
2016-09-08 08:46:19.560659 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-09-08 08:46:27.469556 > HCI Event: Command Status (0x0f) plen 4
    LE Connection Update (0x08|0x0013) status 0x00 ncmd 1
```
Start second app:
```
2016-09-08 08:46:30.212916 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2016-09-08 08:46:30.229916 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2016-09-08 08:46:30.237889 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0x22bb
    LMP Version: 4.0 (0x6) LMP Subversion: 0x22bb
    Manufacturer: Cambridge Silicon Radio (10)
2016-09-08 08:46:30.251909 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    00 
2016-09-08 08:46:30.268908 > HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
    00 01 00 
2016-09-08 08:46:30.276914 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr 00:1A:7D:DA:71:0A
2016-09-08 08:46:30.354929 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2016-09-08 08:46:30.373933 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2016-09-08 08:46:30.553952 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
```
Turn on second device:
```
2016-09-08 08:46:32.964230 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
2016-09-08 08:46:32.990219 > HCI Event: Command Status (0x0f) plen 4
    LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 1
2016-09-08 08:46:33.002237 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-09-08 08:46:43.055342 > HCI Event: Command Status (0x0f) plen 4
    LE Connection Update (0x08|0x0013) status 0x00 ncmd 1
```
Kill test apps:
```
2016-09-08 08:47:33.243169 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2016-09-08 08:47:33.262180 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2016-09-08 08:47:33.758209 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 63 reason 0x16
    Reason: Connection Terminated by Local Host
2016-09-08 08:47:35.674444 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2016-09-08 08:47:35.692456 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2016-09-08 08:47:35.964603 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 68 reason 0x16
    Reason: Connection Terminated by Local Host
```
</details>

sandeepmistry, 2016-09-11:
@jasoncodes ok, I think this might be caused by a kernel disconnect workaround in `node-bluetooth-hci-socket`.

Could you please try:
 1. Removing the call to `this->kernelDisconnectWorkArounds(length, data);` in https://github.com/sandeepmistry/node-bluetooth-hci-socket/blob/master/src/BluetoothHciSocket.cpp#L243
 1. Running `node-gyp rebuild` in your `bluetooth-hci-socket` folder
 1. Then re-running your test.

Even if this does fix the issue, it will re-introduce the issue where the kernel auto disconnects BLE peripherals with some adapters.

jasoncodes, 2016-09-12:
The output from my test apps is the same but the `hcidump` output is now a little different.

<details>
Start first app:
```
2016-09-12 19:11:26 HCI sniffer - Bluetooth packet analyzer ver 5.23
2016-09-12 19:11:26 device: hci0 snap_len: 1500 filter: 0xffffffff
2016-09-12 19:11:26 2016-09-12 19:11:26.202652 > HCI Event: Command Complete (0x0e) plen 4
2016-09-12 19:11:26     Set Event Mask (0x03|0x0001) ncmd 1
2016-09-12 19:11:26     status 0x00
2016-09-12 19:11:26 2016-09-12 19:11:26.219665 > HCI Event: Command Complete (0x0e) plen 4
2016-09-12 19:11:26     LE Set Event Mask (0x08|0x0001) ncmd 1
2016-09-12 19:11:26     status 0x00
2016-09-12 19:11:26 2016-09-12 19:11:26.236662 > HCI Event: Command Complete (0x0e) plen 12
2016-09-12 19:11:26     Read Local Version Information (0x04|0x0001) ncmd 1
2016-09-12 19:11:26     status 0x00
2016-09-12 19:11:26     HCI Version: 4.0 (0x6) HCI Revision: 0x22bb
2016-09-12 19:11:26     LMP Version: 4.0 (0x6) LMP Subversion: 0x22bb
2016-09-12 19:11:26     Manufacturer: Cambridge Silicon Radio (10)
2016-09-12 19:11:26 2016-09-12 19:11:26.237648 > HCI Event: Command Complete (0x0e) plen 4
2016-09-12 19:11:26     Write LE Host Supported (0x03|0x006d) ncmd 1
2016-09-12 19:11:26     00 
2016-09-12 19:11:26 2016-09-12 19:11:26.239646 > HCI Event: Command Complete (0x0e) plen 6
2016-09-12 19:11:26     Read LE Host Supported (0x03|0x006c) ncmd 1
2016-09-12 19:11:26     00 01 00 
2016-09-12 19:11:26 2016-09-12 19:11:26.242650 > HCI Event: Command Complete (0x0e) plen 10
2016-09-12 19:11:26     Read BD ADDR (0x04|0x0009) ncmd 1
2016-09-12 19:11:26     status 0x00 bdaddr 00:1A:7D:DA:71:0A
2016-09-12 19:11:26 2016-09-12 19:11:26.320678 > HCI Event: Command Complete (0x0e) plen 4
2016-09-12 19:11:26     LE Set Scan Enable (0x08|0x000c) ncmd 1
2016-09-12 19:11:26     status 0x0c
2016-09-12 19:11:26     Error: Command Disallowed
2016-09-12 19:11:26 2016-09-12 19:11:26.330674 > HCI Event: Command Complete (0x0e) plen 4
2016-09-12 19:11:26     LE Set Scan Parameters (0x08|0x000b) ncmd 1
2016-09-12 19:11:26     status 0x00
2016-09-12 19:11:26 2016-09-12 19:11:26.493707 > HCI Event: Command Complete (0x0e) plen 4
2016-09-12 19:11:26     LE Set Scan Enable (0x08|0x000c) ncmd 1
2016-09-12 19:11:26     status 0x00
```

Turn on first device:
```
2016-09-12 19:11:37 2016-09-12 19:11:37.712959 > HCI Event: Command Status (0x0f) plen 4
2016-09-12 19:11:37     LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
2016-09-12 19:11:37 2016-09-12 19:11:37.971000 > HCI Event: Command Status (0x0f) plen 4
2016-09-12 19:11:37     LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 0
2016-09-12 19:11:38 2016-09-12 19:11:38.078008 > HCI Event: Command Status (0x0f) plen 4
2016-09-12 19:11:38     Unknown (0x00|0x0000) status 0x00 ncmd 1
2016-09-12 19:11:38 2016-09-12 19:11:38.092005 > HCI Event: Command Complete (0x0e) plen 4
2016-09-12 19:11:38     LE Set Scan Enable (0x08|0x000c) ncmd 1
2016-09-12 19:11:38     status 0x00
```

Start second app:
```
2016-09-12 19:11:45 2016-09-12 19:11:45.983886 > HCI Event: Command Status (0x0f) plen 4
2016-09-12 19:11:45     LE Connection Update (0x08|0x0013) status 0x00 ncmd 1
2016-09-12 19:11:49 2016-09-12 19:11:49.631340 > HCI Event: Command Complete (0x0e) plen 4
2016-09-12 19:11:49     Set Event Mask (0x03|0x0001) ncmd 1
2016-09-12 19:11:49     status 0x00
2016-09-12 19:11:49 2016-09-12 19:11:49.642453 > HCI Event: Command Complete (0x0e) plen 4
2016-09-12 19:11:49     LE Set Event Mask (0x08|0x0001) ncmd 1
2016-09-12 19:11:49     status 0x00
2016-09-12 19:11:49 2016-09-12 19:11:49.653346 > HCI Event: Command Complete (0x0e) plen 12
2016-09-12 19:11:49     Read Local Version Information (0x04|0x0001) ncmd 1
2016-09-12 19:11:49     status 0x00
2016-09-12 19:11:49     HCI Version: 4.0 (0x6) HCI Revision: 0x22bb
2016-09-12 19:11:49     LMP Version: 4.0 (0x6) LMP Subversion: 0x22bb
2016-09-12 19:11:49     Manufacturer: Cambridge Silicon Radio (10)
2016-09-12 19:11:49 2016-09-12 19:11:49.666323 > HCI Event: Command Complete (0x0e) plen 4
2016-09-12 19:11:49     Write LE Host Supported (0x03|0x006d) ncmd 1
2016-09-12 19:11:49     00 
2016-09-12 19:11:49 2016-09-12 19:11:49.675340 > HCI Event: Command Complete (0x0e) plen 6
2016-09-12 19:11:49     Read LE Host Supported (0x03|0x006c) ncmd 1
2016-09-12 19:11:49     00 01 00 
2016-09-12 19:11:49 2016-09-12 19:11:49.685327 > HCI Event: Command Complete (0x0e) plen 10
2016-09-12 19:11:49     Read BD ADDR (0x04|0x0009) ncmd 1
2016-09-12 19:11:49     status 0x00 bdaddr 00:1A:7D:DA:71:0A
2016-09-12 19:11:49 2016-09-12 19:11:49.750343 > HCI Event: Command Complete (0x0e) plen 4
2016-09-12 19:11:49     LE Set Scan Enable (0x08|0x000c) ncmd 1
2016-09-12 19:11:49     status 0x0c
2016-09-12 19:11:49     Error: Command Disallowed
2016-09-12 19:11:49 2016-09-12 19:11:49.764407 > HCI Event: Command Complete (0x0e) plen 4
2016-09-12 19:11:49     LE Set Scan Parameters (0x08|0x000b) ncmd 1
2016-09-12 19:11:49     status 0x00
2016-09-12 19:11:49 2016-09-12 19:11:49.928360 > HCI Event: Command Complete (0x0e) plen 4
2016-09-12 19:11:49     LE Set Scan Enable (0x08|0x000c) ncmd 1
2016-09-12 19:11:49     status 0x00
```

Turn on second device:
```
2016-09-12 19:11:58 2016-09-12 19:11:58.358314 > HCI Event: Command Status (0x0f) plen 4
2016-09-12 19:11:58     LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
2016-09-12 19:12:03 2016-09-12 19:12:03.451897 > HCI Event: Command Status (0x0f) plen 4
2016-09-12 19:12:03     LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 0
2016-09-12 19:12:03 2016-09-12 19:12:03.532910 > HCI Event: Command Status (0x0f) plen 4
2016-09-12 19:12:03     Unknown (0x00|0x0000) status 0x00 ncmd 1
2016-09-12 19:12:03 2016-09-12 19:12:03.538906 > HCI Event: Command Complete (0x0e) plen 4
2016-09-12 19:12:03     LE Set Scan Enable (0x08|0x000c) ncmd 1
2016-09-12 19:12:03     status 0x00
```
</details>

roeckandre, 2016-10-05:
Any update on this?
I am facing some similar problems trying to connect to multiple peripherals simultaneously.
Sometimes it fails on discoverServices and the next time I connect to the device that failed, it calls the callback of that failed.
I tried to fix this creating a timeout function in my app, and then removing the all the listeners everytime I run the timeout function.
It fixes the wrong callback, but about the discoverServices failing, any ideas?

sandeepmistry, 2016-10-10:
@roeckandre please open a new issue with your code, debug logs (`sudo DEBUG=* node <file>.js`) and HCI dump capture (`sudo hcidump -t -x`). As you issue appears to be unrelated to this topic.

sandeepmistry, 2016-10-10:
@jasoncodes can you please try with adding a `noble.stopScanning()` before the connect?

I just tried it on a Raspberry Pi 3 with the [`peripheral-explorer.js` example](https://github.com/sandeepmistry/noble/blob/master/examples/peripheral-explorer.js) + removing the `peripheral.disconnect();` line. I was about to launching two instances and everything worked.

arunbangari, 2016-12-13:
@sandeepmistry I too have seen similar issue. If I am running a node-red-sensortag which uses noble device and then run a separate node js app which uses noble, no services are discovered. If I stop the node-red then, everything works fine.

I have noble.stopScanning before connect in my application.

If you want to investigate this further, let me know and I can provide you hci dump.
Br,Arun

sandeepmistry, 2016-12-18:
@arunbangari sure, if you had time a HCI dump and debug logs would be useful.

bitraven, 2018-03-15:
I think I'm having this same issue with my project.  

I have two Noble processes, the first is Scanner, which loops, scanning for advertising devices and sorting them.  Scanner then forks a second process, Reader with a target device ID.  

Reader then connects to the target device and pulls data off (while Scanner continues to scan).  What I see is Reader gets a valid connection, however never receives any discovery information.  If I kill Scanner and manually launch Reader, it behaves properly.

I'd rather not disable scanning, because the ultimate goal is to fork multiple Reader processes, handling different device connections at the same time.

The same code works perfectly on OSX, this issue is only seen on RPi

Has there been any further thoughts or developments on this issue?

bitraven, 2018-03-15:
Follow-up :  I tried what was suggested here:  https://github.com/noble/noble/issues/403#issuecomment-246206813

The behavior that resulted was the Reader connection became terminated by Local Host

bitraven, 2018-03-15:
I see the same behavior if I run advertisement-discovery and then peripheral-explorer ... peripheral explorer self-terminates after logging the line: "services and characteristics:"

in all of my experiments so far, the hcidump logs basically match the previously commented hcidump logs

luan007, 2018-10-11:
Just a quick note.. stumbled across issue like this today, used 4 CSR dongles in one rpi.
Later found out it was caused by duplicate mac addresses of the dongles.. Mine had same mac across 4 dongles.. 

By using bdaddr and changed mac of the dongles into different ones, mine problem was gone.

Getting bdaddr:
`
sudo apt-get install libbluetooth-dev
wget -U firefox http://www.petrilopia.net/wordpress/wp-content/uploads/bdaddrtar.bz2
bzip2 -d bdaddrtar.bz2 && tar xf bdaddrtar
cd bdaddr && make
`

Addr change:
`
sudo ./bdaddr -i hci0 -r 00:11:22:33:44:55
`

(tutorial from https://scribles.net/changing-a-bluetooth-device-address-on-raspberry-pi/)
(bdaddr from http://blog.petrilopia.net/linux/change-your-bluetooth-device-mac-address/)

hope it helps