noble/noble #405: Out of Range Issue:

sthanh, 2016-05-21:
Hi there,
I have this out of range issue and I it happens intermittently. If I wait and re-run the code it works from time to time but most of the time it ends up with the error. Could you help?

Thank you! 

`pi@raspberrypi:~/Desktop $ node beaconscanv2.js 
buffer.js:615
    throw new RangeError('index out of range');
    ^

RangeError: index out of range
    at checkOffset (buffer.js:615:11)
    at Buffer.readUInt16BE (buffer.js:669:5)
    at Noble.<anonymous> (/home/pi/Desktop/beaconscanv2.js:25:58)
    at emitOne (events.js:77:13)
    at Noble.emit (events.js:169:7)
    at Noble.onDiscover (/home/pi/node_modules/noble/lib/noble.js:135:10)
    at emitMany (events.js:108:13)
    at emit (events.js:182:7)
    at NobleBindings.onDiscover (/home/pi/node_modules/noble/lib/hci-socket/bindings.js:169:10)
    at emitMany (events.js:108:13)
pi@raspberrypi:~/Desktop $ `

sthanh, 2016-05-21:
Here's the hci dump.

pi@raspberrypi:~/Desktop $ sudo hcidump -t -x
HCI sniffer - Bluetooth packet analyzer ver 5.39
device: hci0 snap_len: 1500 filter: 0xffffffff
2016-05-21 00:30:05.594247 < HCI Command: Set Event Mask (0x03|0x0001) plen 8
    Mask: 0xfffffbff07f8bf3d
2016-05-21 00:30:05.595330 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2016-05-21 00:30:05.595384 < HCI Command: LE Set Event Mask (0x08|0x0001) plen 8
    mask 0x1f00000000000000 (Reserved)
2016-05-21 00:30:05.595874 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2016-05-21 00:30:05.596153 < HCI Command: Read Local Version Information (0x04|0x0001) plen 0
2016-05-21 00:30:05.596624 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.1 (0x7) HCI Revision: 0xb6
    LMP Version: 4.1 (0x7) LMP Subversion: 0x2209
    Manufacturer: Broadcom Corporation (15)
2016-05-21 00:30:05.596909 < HCI Command: Write LE Host Supported (0x03|0x006d) plen 2
  01 00 
2016-05-21 00:30:05.597343 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    00 
2016-05-21 00:30:05.597611 < HCI Command: Read LE Host Supported (0x03|0x006c) plen 0
2016-05-21 00:30:05.598060 > HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
    00 01 00 
2016-05-21 00:30:05.598285 < HCI Command: Read BD ADDR (0x04|0x0009) plen 0
2016-05-21 00:30:05.598828 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr B8:27:EB:B7:2B:5E
2016-05-21 00:30:06.444517 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2016-05-21 00:30:06.445132 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2016-05-21 00:30:06.445331 < HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
    type 0x01 (active)
    interval 10.000ms window 10.000ms
    own address: 0x00 (Public) policy: All
2016-05-21 00:30:06.445826 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2016-05-21 00:30:06.452421 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
2016-05-21 00:30:06.453046 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-05-21 00:30:06.876192 > HCI Event: LE Meta Event (0x3e) plen 42
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr D2:49:30:B4:BA:67 (Random)
      Flags: 0x06
      Unknown type 0xff with 25 bytes data
      RSSI: -76
2016-05-21 00:30:07.139116 > HCI Event: LE Meta Event (0x3e) plen 27
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr F4:F9:51:D1:9E:76 (Public)
      Flags: 0x1a
      Unknown type 0xff with 10 bytes data
      RSSI: -96
2016-05-21 00:30:07.139712 > HCI Event: LE Meta Event (0x3e) plen 12
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr F4:F9:51:D1:9E:76 (Public)
      RSSI: -96
2016-05-21 00:30:07.151854 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2016-05-21 00:30:07.154270 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00

sthanh, 2016-05-24:
When I try to do Debug mode, it seems the program keep hanging at

hci set scan enabled - writing: 010c20020001 +38s

Any ideas what's going on? Anybody?

peterschmiz, 2016-05-28:
I have the same issue, it happens after running my application for 2-3 minutes.

sandeepmistry, 2016-05-31:
@sthanh what's in you ``beaconscanv2.js`` file?

sthanh, 2016-06-01:
@sandeepmistry it's just a beacon scan for specific service and characteristic, in this case for the battery.
Thank you for your help.

Here's the code:

`var noble = require('noble');
var async = require('async');
var http = require('http');
var request = require('request');

noble.on('stateChange', function(state) {
	if(state === 'poweredOn'){
		noble.startScanning();
	}
	else{noble.stopScanning();}
	});

var companyName = "MiniBeacon";
var mainUUID = ['74278bdab64445208f0c720eaf059937'];
var batteryPower = '';

noble.on('discover', function(peripheral){
	 var macAddress = peripheral.uuid;
	 var rss = peripheral.rssi;
	 var localName = peripheral.advertisement.localName;
	 var txPower = peripheral.advertisement.txPowerLevel;
	 var manuData = JSON.stringify(peripheral.advertisement.manufacturerData.toString('hex'));
	 var uuid = peripheral.advertisement.manufacturerData.slice(4, 20).toString('hex');
  	 var major = peripheral.advertisement.manufacturerData.readUInt16BE(20);
  	 var minor = peripheral.advertisement.manufacturerData.readUInt16BE(22);
  	 var measuredPower = peripheral.advertisement.manufacturerData.readInt8(24);

	 if (String(localName).indexOf(companyName) >= 0){
	 
	    noble.stopScanning();
	   
        peripheral.connect(function(error) {
    
           peripheral.discoverServices(['180f'], function(error, services) {
          var deviceInformationService = services[0];
          deviceInformationService.discoverCharacteristics(['2a19'], function(error, characteristics) {
               var manufacturerNameCharacteristic = characteristics[0];
               manufacturerNameCharacteristic.read(function(error, data) {
          // data is a buffer
                 batteryPower = data.toString('hex');
               var sendata = {
                           
                           "beacon" : 
                           
                           {
                            'UUID': uuid,
                            'Major': major,
                            'Minor': minor,
                            
               },
                           "hub": {"HubID":"923456789"},
                            'RSSI': rss };
               
                sendme = JSON.stringify(sendata);

                         console.log(sendme);
                         
                         var options = {
                         method: 'POST',
                         url: ' http://localhost:3000/api/v1/scan',
                        
                         headers: {
                         //   "Content-Type": "application/json",
                          
                         },
                         json: sendata

                     };
  function callback(error, response, body) {
      if (!error) {
         // var info = JSON.parse(body);
         // console.log(info);
      }
      else {
         console.log('Error happened: '+ error);
     }
  };

// //send request
 request(options, callback);
  
             
              
                });
                
             });
         });
    });

sandeepmistry, 2016-06-06:
@sthanh I would say the issue is in your code:

```javascript
	 var uuid = peripheral.advertisement.manufacturerData.slice(4, 20).toString('hex');
  	 var major = peripheral.advertisement.manufacturerData.readUInt16BE(20);
  	 var minor = peripheral.advertisement.manufacturerData.readUInt16BE(22);
  	 var measuredPower = peripheral.advertisement.manufacturerData.readInt8(24);
```

`peripheral.advertisement.manufacturerData` is not guaranteed to contain iBeacon data, other BLE peripherals in the vicinity could be broadcasting other data. I suggest you have a guard like: https://github.com/sandeepmistry/node-bleacon/blob/master/lib/bleacon.js#L78-L82

closed 2016-06-06 by sandeepmistry

jpalo, 2019-11-23:
Added try/catch when handling EVT_LE_ADVERTISING_REPORT in hci.js to stop my node apps from crashing.

```
Hci.prototype.processLeMetaEvent = function (eventType, status, data) {
  if (eventType === EVT_LE_CONN_COMPLETE) {
    this.processLeConnComplete(status, data);
  } else if (eventType === EVT_LE_ADVERTISING_REPORT) {
    try {
      this.processLeAdvertisingReport(status, data);
    } catch (err) {
      debug('processLeAdvertisingReport crash prevented');
    }
  } else if (eventType === EVT_LE_CONN_UPDATE_COMPLETE) {
    this.processLeConnUpdateComplete(status, data);
  }
};
```