noble/noble #407: Question - Subscribe to a Characteristic

newbreedofgeek, 2016-06-01:
Hello,

This is more of a question than an issue.

I've pretty new to the BLE / noble scene and have been following the documentation and examples but have hit an issue when trying to connect and read from a BLE weight scale peripheral.

I've been having issues getting data from a peripheral that has a Service Characteristic with "indicate".

This is the Characteristic I get back from the peripheral:

```
var myChar = Characteristic {
  _noble:
   Noble {
     state: 'poweredOn',
     address: 'f4:5c:89:ab:44:6a',
     _bindings:
      EventEmitter {
        _peripherals: [Object],
        _xpcConnection: [Object],
        _events: [Object],
        _eventsCount: 38,
        sendCBMsg: [Function],
        init: [Function],
        startScanning: [Function],
        stopScanning: [Function],
        connect: [Function],
        disconnect: [Function],
        updateRssi: [Function],
        discoverServices: [Function],
        discoverIncludedServices: [Function],
        discoverCharacteristics: [Function],
        read: [Function],
        write: [Function],
        broadcast: [Function],
        notify: [Function],
        discoverDescriptors: [Function],
        readValue: [Function],
        writeValue: [Function],
        readHandle: [Function],
        writeHandle: [Function] },
     _peripherals:
      { '90836f2c78ff4c6281b4678c5ece9170': [Object],
        e9da9f08428c4c308ac7b0bb4c938ad7: [Object],
        '04ef0187ec39492b96a60540fe714278': [Object] },
     _services:
      { '90836f2c78ff4c6281b4678c5ece9170': {},
        e9da9f08428c4c308ac7b0bb4c938ad7: {},
        '04ef0187ec39492b96a60540fe714278': [Object] },
     _characteristics:
      { '90836f2c78ff4c6281b4678c5ece9170': {},
        '3805e319d4944ff3aebd8ed27c40baf8': {},
        e9da9f08428c4c308ac7b0bb4c938ad7: {},
        '04ef0187ec39492b96a60540fe714278': [Object] },
     _descriptors:
      { '90836f2c78ff4c6281b4678c5ece9170': {},
        e9da9f08428c4c308ac7b0bb4c938ad7: {},
        '04ef0187ec39492b96a60540fe714278': [Object] },
     _discoveredPeripheralUUids:
      [ '90836f2c78ff4c6281b4678c5ece9170',
        '3805e319d4944ff3aebd8ed27c40baf8',
        '04ef0187ec39492b96a60540fe714278' ],
     _events:
      { warning: [Function: bound ],
        discover: [Function],
        stateChange: [Function] },
     _eventsCount: 3,
     _allowDuplicates: true },
  _peripheralId: '04ef0187ec39492b96a60540fe714278',
  _serviceUuid: '181d',
  uuid: '2a9d',
  name: 'Weight Measurement',
  type: 'org.bluetooth.characteristic.weight_measurement',
  properties: [ 'indicate' ],
  descriptors: null }
```

And I'm trying to subscribe to its updates like so (in order to get the data):

```
myChar.on('read', function(data, isNotification) {
    // peripheral data is read here        
});

myChar.subscribe(function(err) {});
```

But I get an error saying "TypeError: myChar.subscribe is not a function"

What am I doing wrong?

Thanks very much.

newbreedofgeek, 2016-06-01:
Sorry for troubling you guys. I worked it out.

For anyone else who maybe interested, you don't "subscribe" to Characteristics that Indicate, you instead have to:

```
characteristic.notify(true, function(error) {});
```

and then:

```
characteristic.on('data', function(data, isNotification) {});
```

closed 2016-06-01 by newbreedofgeek

sandeepmistry, 2016-06-06:
@newbreedofgeek sorry for the confusion, as indicated in #408: https://github.com/sandeepmistry/noble/commit/472c6036eaa4e9c236e91e23987289d1c074731f was in Github master but not published to npm.

v1.6.0 is on npm and includes the new `myChar.subscribe` API.

newbreedofgeek, 2016-06-06:
@sandeepmistry thanks :)