noble/noble #409: Testing on pi3?

aronsemle, 2016-06-01:
Has this been tested on the raspberry pi3? It has built in bluetooth.

I wrote some code on Windows with a dongle (scan, connect, stream data, disconnect) and it's working great. I ported it over to the pi3 and I'm seeing random issues.

1. I get random disconnects right after a connect (disconnect callback is called). Doesn't happen all the time. Maybe every 3rd try
2. After a disconnect, scan won't find the device again (even if duplicates are enabled)

I just tried this out and haven't really started debugging it, but I wanted to see if pi3 has known issues.

ajwells, 2016-06-03:
The only weird thing I have experienced on the Pi 3 are disconnects not right after connect but about a minute or so afterwards.

sandeepmistry, 2016-06-06:
@aronsemle 

> Has this been tested on the raspberry pi3?

Yes, I use noble on a Raspberry Pi 3.

HCI dump capture will help to understand what's going on: `sudo hcidump -t -x`

aronsemle, 2016-06-06:
Thanks for the tip!

Unfortunately I don't have that tool installed on the pi at the moment, and can't get it connected to the internet to install.

It does look like the problem is only occurring when I continue to scan for devices after connecting. It will disconnect right away. If I stop the scan once the device is discovered and then connect, it seems to work fine.

I'll keep digging. I want to continue scanning for devices in my application.

ajwells, 2016-06-06:
@aronsemle 

I used the following guide to get one of latest version of Bluez running on the Pi 3 it includes some Bluetooth utilities 

https://github.com/ukBaz/python-bluezero/issues/30

peturh, 2016-06-07:
@ajwells It doesn't handle raspberry pi's disconnects right? Any progress on this? I'm experiencing the exact same issue.

peturh, 2016-06-10:
I have a scenario where I connect to devices and try to send data to it. If new devices are created in our environment, they don't show up and random disconnects occur.

Here is a scenario from a dump:
[hciLog.txt](https://github.com/sandeepmistry/noble/files/308335/hciLog.txt)

sandeepmistry, 2016-06-20:
@peturh what peripheral are you using?

```
2016-06-07 15:58:58.094195 < HCI Command: LE Connection Update (0x08|0x0013) plan 14

...

2016-06-07 15:58:58.387627 > HCI Event: LE Meta Event (0x3e) plen 10
    LE Connection Update Complete
      status 0x00 handle 64
      interval 997.50ms, latency 0.00ms, superv. timeout 10000.00ms

...

2016-06-07 15:59:19.335809 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x08
    Reason: Connection Timeout
```

The disconnect happens shortly after the Linus kernel sends a request for a connection update (change connection parameters).

taktran, 2016-06-22:
Hi, I'm getting problems with disconnecting on a rpi3 too.

Some devices disconnect and connect correctly, but then some don't connect again at all. When they don't connect, they don't show up in `sudo hcidump -t -x` at all. However, I have noticed that when I open the [Bluetooth manager](https://thepihut.com/blogs/raspberry-pi-tutorials/17841464-bluetooth-installing-and-using-bluetooth-on-the-raspberry-pi) > Add Device... dialog, it kick starts the scan of noble, and it can be found again - do you know if there is a command that can kick start the scan again?

Sometimes the devices also disconnect after a couple of minutes.

For reference, these were the dumps using `sudo hcidump -t -x`:

* when the device was working: https://gist.github.com/anonymous/69862ae4c6d38c8ca0939bbd53ee7a48
* when the device wasn't detected, but then got kick started with bluetooth manager: https://gist.github.com/anonymous/374e33910dd5bbc95e85247343cdd981

Also, I'm using noble v1.0.2. I tried to use v1.6.0, but the app would keep disconnecting.

taktran, 2016-06-22:
I just caught the devices turning off and on again with the `hcidump`: https://gist.github.com/anonymous/4e8c4e08a27e1cfc11075bbfb5cf755a

I **didn't** have bluetooth manager on in this case, and 2 came back on and 2 didn't. There was no activity between the devices for a couple of minutes before they turned off.

sandeepmistry, 2016-06-26:
> Also, I'm using noble v1.0.2. I tried to use v1.6.0, but the app would keep disconnecting.

@taktran I don't "support" old versions of noble. We can discuss setting up a consulting agreement via email if you have no plans to upgrade to the latest.

taktran, 2016-07-22:
Hi @sandeepmistry, we are planning to upgrade, just haven't had the time to do it yet. Just thought it might be useful know, the issue happens on older version noble.

Will let you know my findings when I do a successful upgrade.

sandeepmistry, 2016-09-26:
I'm closing this due to lack of activity, please try the latest version of noble, and open a new issue if you still have problems.

closed 2016-09-26 by sandeepmistry