noble/noble #411: Is it possible to flush the command queue?

Dan-Azure, 2016-06-06:
Hi, Sandeep. I have a problem, which is not really yours, but I'll pose it anyway:

When I use noble to connect to my device, I will often see this behaviour:

< HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr ** type 1
    interval 96 window 48 initiator_filter 0
    own_bdaddr_type 0 min_interval 6 max_interval 12
    latency 0 supervision_to 200 min_ce 4 max_ce 6
> HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
> HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 64, role master
      bdaddr ** (Random)
< ACL data: handle 64 flags 0x00 dlen 7
    ATT: MTU req (0x02)
      client rx mtu 256
< ACL data: handle 64 flags 0x00 dlen 7
    ATT: MTU req (0x02)
      client rx mtu 256
< ACL data: handle 64 flags 0x00 dlen 7
    ATT: MTU req (0x02)
      client rx mtu 256
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 3
> HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x13
    Reason: Remote User Terminated Connection
 
or to paraphrase: Our remote device terminates in the middle of the MTU exchange. What happens in the master end is then seemingly that noble has queued a request for MTU callback inside a command queue. When we give up at the high level and create a new connection, the request still lingers and the new connection is then passed back into the lingering callback and doesn't get to us.
 
Now, obviously our first priority is to stop the slave device from disconnecting during handshaking, which is clearly our own problem.
 
But do you know of a way we can work around the noble side? To somehow pop this dead callback or check for its existence from the top level?

Thanks,

Dan

sandeepmistry, 2016-06-20:
@Dan-Azure could you please share a more complete HCI dump capture for more context. Just to track down the root cause. (Also, adding `-t -x` arguments would be great when running `hcidump`).

Dan-Azure, 2016-06-21:
Hi, Sandeep.

With pleasure:

    LE Advertising Report                                                       
      ADV_IND - Connectable undirected advertising (0)                          
      bdaddr DEVICE (Random)                                         
      Shortened local name: 'myname'                                          
      Flags: 0x06                                                               
      Unknown type 0x06 with 16 bytes data                                      
      RSSI: -48                                                                 
2016-06-21 00:42:16.518519 > HCI Event: LE Meta Event (0x3e) plen 12            
    LE Advertising Report                                                       
      SCAN_RSP - Scan Response (4)                                              
      bdaddr DEVICE (Random)                                         
      RSSI: -48                                                                 
2016-06-21 00:42:16.521939 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 
    value 0x00 (scanning disabled)                                              
    filter duplicates 0x01 (enabled)                                            
2016-06-21 00:42:16.523457 > HCI Event: Command Complete (0x0e) plen 4          
    LE Set Scan Enable (0x08|0x000c) ncmd 1                                     
    status 0x00                                                                 
2016-06-21 00:42:16.523570 < HCI Command: LE Create Connection (0x08|0x000d) ple
    bdaddr DEVICE type 1                                            
    interval 96 window 48 initiator_filter 0                                   
    own_bdaddr_type 0 min_interval 6 max_interval 12                           
    latency 0 supervision_to 200 min_ce 4 max_ce 6                             
2016-06-21 00:42:16.524506 > HCI Event: Command Status (0x0f) plen 4            
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1                       
2016-06-21 00:42:18.212168 > HCI Event: LE Meta Event (0x3e) plen 19            
    LE Connection Complete                                                      
      status 0x00 handle 64, role master                                        
      bdaddr DEVICE (Random)                                         
2016-06-21 00:42:18.224355 < ACL data: handle 64 flags 0x00 dlen 7              
    ATT: MTU req (0x02)                                                         
      client rx mtu 256                                                         
2016-06-21 00:42:18.252957 > ACL data: handle 64 flags 0x02 dlen 7              
    ATT: MTU resp (0x03)                                                        
      server rx mtu 23                                                          
2016-06-21 00:42:18.412839 > HCI Event: Number of Completed Packets (0x13) plen 5 
    handle 64 packets 1                                                         
2016-06-21 00:42:21.525952 < HCI Command: Disconnect (0x01|0x0006) plen 3       
    handle 64 reason 0x13                                                       
    Reason: Remote User Terminated Connection                                   
2016-06-21 00:42:21.532374 > HCI Event: Command Status (0x0f) plen 4           
    Disconnect (0x01|0x0006) status 0x00 ncmd 1                                
2016-06-21 00:42:21.532589 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)                                               
    filter duplicates 0x00 (disabled)                                           
2016-06-21 00:42:21.533254 > HCI Event: Command Complete (0x0e) plen 4          
    LE Set Scan Enable (0x08|0x000c) ncmd 1                                     
    status 0x00                                                                 
2016-06-21 00:42:21.540108 > HCI Event: Disconn Complete (0x05) plen 4          
    status 0x00 handle 64 reason 0x16                                           
    Reason: Connection Terminated by Local Host                                 
2016-06-21 00:42:21.551731 > HCI Event: LE Meta Event (0x3e) plen 43            
    LE Advertising Report                                                       
      ADV_IND - Connectable undirected advertising (0)                          
      bdaddr DEVICE (Random)                                         
      Shortened local name: 'myname'                                          
      Flags: 0x06                                                               
      Unknown type 0x06 with 16 bytes data                                      
      RSSI: -48                                                                

A short break with a lot of ADV_INDs followed by

2016-06-21 00:42:16.439656 > HCI Event: LE Meta Event (0x3e) plen 43            
    LE Advertising Report                                                       
      ADV_IND - Connectable undirected advertising (0)                          
      bdaddr OTHERDEVICE (Random)                                         
[... 81 lines cut ...]
    LE Set Scan Enable (0x08|0x000c) ncmd 1                                     
    status 0x0c                                                                 
    Error: Command Disallowed                                                   
2016-06-21 00:42:21.779741 > HCI Event: LE Meta Event (0x3e) plen 27            
[... cut ...]
      Flags: 0x06                                                               
      Unknown type 0x06 with 16 bytes data                                      
      RSSI: -77                                                                 
      - etc.                         
                                                              
This is a full cut. No lines or messages in between except for the big chunk of advertising messages in the middle. I've removed the MAC's, but please trust me when I say they are perfectly normal.

Thanks,

Dan

sandeepmistry, 2016-06-26:
@Dan-Azure I might be missing something, but I don't see a disconnect event between MTU exchanges anywhere:

```
2016-06-21 00:42:16.523570 < HCI Command: LE Create Connection (0x08|0x000d) ple
    bdaddr DEVICE type 1                                            
    interval 96 window 48 initiator_filter 0                                   
    own_bdaddr_type 0 min_interval 6 max_interval 12                           
    latency 0 supervision_to 200 min_ce 4 max_ce 6                             
2016-06-21 00:42:16.524506 > HCI Event: Command Status (0x0f) plen 4            
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1                       
2016-06-21 00:42:18.212168 > HCI Event: LE Meta Event (0x3e) plen 19            
    LE Connection Complete                                                      
      status 0x00 handle 64, role master                                        
      bdaddr DEVICE (Random)                                         
2016-06-21 00:42:18.224355 < ACL data: handle 64 flags 0x00 dlen 7              
    ATT: MTU req (0x02)                                                         
      client rx mtu 256                                                         
2016-06-21 00:42:18.252957 > ACL data: handle 64 flags 0x02 dlen 7              
    ATT: MTU resp (0x03)                                                        
      server rx mtu 23                                                          
2016-06-21 00:42:18.412839 > HCI Event: Number of Completed Packets (0x13) plen 5 
    handle 64 packets 1                                                         
2016-06-21 00:42:21.525952 < HCI Command: Disconnect (0x01|0x0006) plen 3       
    handle 64 reason 0x13                                                       
    Reason: Remote User Terminated Connection                                   
2016-06-21 00:42:21.532374 > HCI Event: Command Status (0x0f) plen 4           
    Disconnect (0x01|0x0006) status 0x00 cmd 1

...

2016-06-21 00:42:21.540108 > HCI Event: Disconn Complete (0x05) plen 4          
    status 0x00 handle 64 reason 0x16                                           
    Reason: Connection Terminated by Local Host    
```

```
2016-06-21 00:42:16.523570 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr DEVICE type 1                                             
    interval 96 window 48 initiator_filter 0                                    
    own_bdaddr_type 0 min_interval 6 max_interval 12                            
    latency 0 supervision_to 200 min_ce 4 max_ce 6                              
2016-06-21 00:42:16.524506 > HCI Event: Command Status (0x0f) plen 4            
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1                       
2016-06-21 00:42:18.212168 > HCI Event: LE Meta Event (0x3e) plen 19            
    LE Connection Complete                                                      
      status 0x00 handle 64, role master                                        
      bdaddr DEVICE (Random)                                         
2016-06-21 00:42:18.224355 < ACL data: handle 64 flags 0x00 dlen 7              
    ATT: MTU req (0x02)                                                         
      client rx mtu 256                                                         
2016-06-21 00:42:18.252957 > ACL data: handle 64 flags 0x02 dlen 7              
    ATT: MTU resp (0x03)                                                        
      server rx mtu 23                                                          
2016-06-21 00:42:18.412839 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 1                                                         
2016-06-21 00:42:21.525952 < HCI Command: Disconnect (0x01|0x0006) plen 3       
    handle 64 reason 0x13                                                       
    Reason: Remote User Terminated Connection                                   
2016-06-21 00:42:21.532374 > HCI Event: Command Status (0x0f) plen 4            
    Disconnect (0x01|0x0006) status 0x00 cmd 1

...

2016-06-21 00:42:21.540108 > HCI Event: Disconn Complete (0x05) plen 4          
    status 0x00 handle 64 reason 0x16                                           
    Reason: Connection Terminated by Local Host
```

However, I do see a local disconnect command, is this what you are concerned about?

sandeepmistry, 2016-08-22:
@Dan-Azure ping ...

Dan-Azure, 2016-08-22:
Hi, Sandeep. Apologies for not getting back to you. My problem fixed itself and I kinda forgot about you.

However, I do want to share what fixed my problem:

hci.js: Hci.prototype.createLeConn

The default supervision timeout is really short. Too short for us to reliably succeed in connecting. We looked at Apple's default standard and they give something like 5 times the period that noble gives us.

We changed it from c8 and are running it at 2ee right now and everything seems happy.

sandeepmistry, 2016-10-10:
[image]

@Dan-Azure thanks for the follow up. On my Mac it's using 0x00c8 for the supervision timeout, which maps to 200 * 10 = 2000 ms.

closed 2016-10-10 by sandeepmistry