noble/noble #412: Unable to scan.

akayami, 2016-06-11:
I've been attempting to run examples/enter-exit.js as root
The program generates no output at all. 

hcidump -t -x:
```
HCI sniffer - Bluetooth packet analyzer ver 5.40
device: hci0 snap_len: 1500 filter: 0xffffffffffffffff
2016-06-10 22:41:42.687700 < HCI Command: Set Event Mask (0x03|0x0001) plen 8
    Mask: 0xfffffbff07f8bf3d
2016-06-10 22:41:42.689146 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2016-06-10 22:41:42.689160 < HCI Command: LE Set Event Mask (0x08|0x0001) plen 8
    mask 0x1f00000000000000 (Reserved)
2016-06-10 22:41:42.690157 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2016-06-10 22:41:42.690171 < HCI Command: Read Local Version Information (0x04|0x0001) plen 0
2016-06-10 22:41:42.702162 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0x22bb
    LMP Version: 4.0 (0x6) LMP Subversion: 0x22bb
    Manufacturer: Cambridge Silicon Radio (10)
2016-06-10 22:41:42.702197 < HCI Command: Write LE Host Supported (0x03|0x006d) plen 2
  01 00 
2016-06-10 22:41:42.703157 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    00 
2016-06-10 22:41:42.703170 < HCI Command: Read LE Host Supported (0x03|0x006c) plen 0
2016-06-10 22:41:42.704146 > HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
    00 01 00 
2016-06-10 22:41:42.704169 < HCI Command: Read BD ADDR (0x04|0x0009) plen 0
2016-06-10 22:41:42.705148 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr 00:1A:7D:DA:71:11
2016-06-10 22:41:42.705168 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2016-06-10 22:41:42.706144 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2016-06-10 22:41:42.706163 < HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
    type 0x01 (active)
    interval 10.000ms window 10.000ms
    own address: 0x00 (Public) policy: All
2016-06-10 22:41:42.707157 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2016-06-10 22:41:42.707534 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x00 (disabled)
2016-06-10 22:41:42.709157 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
```

I'm pretty new to this stuff, and not used to read this output. I suspect this part of it might have something to do with the issue: 
```
2016-06-10 22:41:42.706144 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
```

Just FYI, I'm able to do everything using bluetoothctl, so all involved devices are to be running correctly, but I've not been able to see any output while attempting to scan. 

Additionally, I've tried a very simple test program of my own:
```
var noble = require('noble');

noble.on('stateChange', function(state) {
	if (state === 'poweredOn') {
		console.log('scanning');
		noble.startScanning([], false, function(err) {
			if(err) {
				console.error('Error');
				console.error(err);
			}
		});
	} else {
		console.log('not scanning');
		noble.stopScanning();
	}
});

noble.on('discover', function(peripheral) {
	console.log('Discovered');
	console.log(peripheral);
})

noble.on('warning', function(message) {
	console.log('Warn');
	console.warn(message);
});
```

The output I receive is the following:
```
scanning
Error
true
```
which would logically mean the callback of startScanning is called with an error set to "true". I'm not sure what is going on there, but from the documentation it looks like the only thing that can be passed to this callback is an error. 

System:
Node: v6.2.1
4.5.4-1-ARCH

Thank you in advance

skerit, 2016-06-17:
I've got the same problem. But, once I've created a connection to a device, enabling SCAN *does* work.

akayami, 2016-06-18:
Hmm should not you first scan for stuff before connecting ? It does not
make sense to me.

sandeepmistry, 2016-06-20:
@skerit 

> But, once I've created a connection to a device, enabling SCAN does work.

What Bluetooth adapter are you using?

@akayami 

> which would logically mean the callback of startScanning is called with an error set to "true". I'm not sure what is going on there, but from the documentation it looks like the only thing that can be passed to this callback is an error.

Yes, there is something wrong here, an error should be passed into the callback.

sandeepmistry, 2016-10-10:
I'm closing this due to lack of activity.

closed 2016-10-10 by sandeepmistry

bassamseif, 2016-10-30:
I'm having the same issue mentioned before, here's the dump
```
2016-10-30 20:04:18.961023 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2016-10-30 20:04:18.963788 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2016-10-30 20:04:18.966297 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.1 (0x7) HCI Revision: 0xb6
    LMP Version: 4.1 (0x7) LMP Subversion: 0x2209
    Manufacturer: Broadcom Corporation (15)
2016-10-30 20:04:18.970168 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    00 
2016-10-30 20:04:18.972580 > HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
    00 01 00 
2016-10-30 20:04:18.975057 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr B8:27:EB:B9:D0:1D
2016-10-30 20:04:18.998138 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2016-10-30 20:04:18.999298 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2016-10-30 20:04:19.007773 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
```
using node 5.9.1 and running the `enter-exit.js` script in the examples folder

sandeepmistry, 2016-10-30:
@bassamseif 

The last scan enable looks ok:

```
2016-10-30 20:04:19.007773 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
```

bassamseif, 2016-10-30:
I see, but it does not detect any bluetooth devices although i can detect them using `bluetoothctl`

sandeepmistry, 2016-10-30:
Are you sure the device you are scanning for is a BLE device? noble only supports BLE and not regular Bluetooth classic devices.

If you provide a HCI dump capture of what happens when you use `bluetoothctl`, I can take a quick look.

bassamseif, 2016-10-31:
here is a dump from an inquiry with `bluetoothctl`
```
2016-10-31 08:37:08.660448 > HCI Event: Command Status (0x0f) plen 4
    Inquiry (0x01|0x0001) status 0x00 ncmd 1
2016-10-31 08:37:15.887097 > HCI Event: Extended Inquiry Result (0x2f) plen 255
    bdaddr C8:A8:23:2A:11:A0 mode 1 clkoffset 0x72a3 class 0x5a020c rssi -59
    Complete local name: 'Galaxy S6'
    Complete service classes: 0x1105 0x110a 0x110c 0x1112 0x1115 0x1116 0x111f 0x112d 0x112f 0x1200 0x1132
    Unknown type 0x05 with 0 bytes data
    Unknown type 0x07 with 0 bytes data
    Unknown type 0xff with 39 bytes data
2016-10-31 08:37:18.903117 > HCI Event: Inquiry Complete (0x01) plen 1
    status 0x00
```

sandeepmistry, 2016-10-31:
Yep, the not BLE. Event type would be:
```
HCI Event: LE Meta Event (0x3e) plen 23
    LE Advertising Report
...
```

bassamseif, 2016-10-31:
oh okay, thanks for the clarification!