noble/noble #413: noble: unknown peripheral undefined RSSI update!

skerit, 2016-06-14:
I'm periodically trying to get an update on the RSSI value, like so:

```
peripheral.updateRssi(callback);
```

But all I get as output is:

    noble: unknown peripheral undefined RSSI update!

What's going wrong there?

sandeepmistry, 2016-06-20:
@skerit we're going to need more info. to track this down ...

 * What OS are you using noble on?
 * Could you please share a minimal Node.js app to re-create the problem?
 * Also, provided debug logs from running above, `DEBUG=* node <file>.js`

sandeepmistry, 2016-08-22:
@skerit ping ...

sandeepmistry, 2016-09-11:
I'm closing this due to lack of activity, please re-open with requested info. if you are still interested.

closed 2016-09-11 by sandeepmistry

zangdar64, 2016-11-30:
I have the same problem on my linux (Rpi3) board using estimote tags.

`	for(uuid in noble._peripherals){
		noble.updateRssi(uuid);
	}`
print this error `noble: unknown peripheral undefined RSSI update!`

using console.log i make some test
`peripheralUuid` is undefined in`Noble.prototype.onRssiUpdate = function(peripheralUuid, rssi) {` call (in  noble/lib/noble.js)
also in  noble/lib/hci-socket/hci.js L600
`this.emit('rssiRead', handle, rssi);` the "handle and rssi values are 0 on my board.

this is some of my debug
>   hci read rssi - writing: 010514020000 +6ms
  hci onSocketData: 040e0701051402000000 +2ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 5125 +1ms
  hci           status = 2 +0ms
  hci           result = 000000 +0ms
  hci                   handle = 0 +0ms
  hci                   rssi = 0 +0ms
noble: unknown peripheral undefined RSSI update!

(sorry for my bad english)

sandeepmistry, 2016-12-02:
> for(uuid in noble._peripherals){ noble.updateRssi(uuid); }

This is not a public facing API. Please see: https://github.com/sandeepmistry/noble#update-rssi

zangdar64, 2016-12-02:
This is the same problem with peripheral.updateRssi()
I test with noble test.js and it fail too
`sudo node node_modules/noble/test.js
noble
on -> stateChange: poweredOn
on -> scanStart
on -> discover: {"id":"dd4fc890b5c8","address":"dd:4f:c8:90:b5:c8","addressType":"random","connectable":true,"advertisement":{"manufacturerData":{"type":"Buffer","data":[93,1,1,148,247,195,240,0,65,107,120,4,133,4,225,116,253,255,193,0,22,83]},"serviceData":[],"serviceUuids":["180f"],"solicitationServiceUuids":[],"serviceSolicitationUuids":[]},"rssi":-88,"state":"disconnected"}
on -> scanStop
on -> connect
on -> disconnect
noble: unknown peripheral undefined RSSI update!
`
There is a disconnect problem in updateRssi function

sandeepmistry, 2016-12-05:
`test.js` from noble is not a really useful test app for you to use.

Please provide a minimal sample app that can be used to reproduce the issue along with an HCI dump capture `sudo hcidump -t -x`

malikolivier, 2017-03-27:
Hi Sandeep,

I've got the same issue than these people on Intel Edison.

See for yourself a sample code that reproduces the issue on my Edison and an HCI dump:
https://gist.github.com/malikolivier/66e571fd639e35597e6ad180df31872b
My edison had its bluetooth deamon disabled a long time ago with `systemctl disable bluetooth`.

Hope this will helps.

By the way, here is BlueZ version:
```
root@edison:~# bluetoothctl --version
5.24
```

sandeepmistry, 2017-04-09:
@malikolivier please open a new issue to track this. Thanks.