noble/noble #415: configuration property NOBLE_HCI_DEVICE_ID=1

vdharmon, 2016-06-18:
I'm working with a device that has multiple BLE's.  One is attached via USB and another one is on the board.  I'm trying to pin the Noble code to the UART (on board) BLE.  From my testing it's not pinning the Noble code to the device id I set the property to.  
Here's how I'm testing.

1) I have a USB BLE plugged in.  Hciconfig shows the USB as hci0 and the UART (on board) as hci1.

2) I set the NOBLE_HCI_DEVICE_ID = 0 and execute code to scan for devices. I see BLE devices displayed.  Also I can see the USB BLE light up when activity is happening.  The USB lights up as expected.  If I don't set the NOBLE_HCI_DEVICE_ID I get the same results, which is expected.

3) I set the NOBLE_HCI_DEVICE_ID = 1 and execute code to scan for devices.  I see BLE devices displayed, but the USB lights up which is not expected because the code should be running on the UART on the board.

4) I unplug the USB BLE so there's only the hci1, there's no hci0 listed.  

5) I set NOBLE_HCI_DEVICE_ID = 0, to see if that forces the Noble code to fail.  I execute code to scan for devices and BLE devices are showing up.

From what I can see the Noble code always pickups the first device it sees in the hci listing and never pins.  I see the same results testing the Bleno also.

Let me know what I can do to help address.

Regards,
Vince

sandeepmistry, 2016-06-20:
@vdharmon this probably related to the discussion in https://github.com/sandeepmistry/node-bleacon/issues/47#issuecomment-225391552 by @jfathman.

He noticed the following:

1. When there are no external USB adapters plugged in the UART interface is hci0
2. When an external USB adapter is plugged in the UART becomes hci1 then the USB adapter is hci0

Since the kernel decided which HCI # to use for each adapter, this is out of control of noble.

closed 2016-06-20 by sandeepmistry

vdharmon, 2016-06-20:
Actually your code is working correctly.  

I'm passing in NOBLE_HCI_DEVICE_ID = 0 and BLENO_HCI_DEVICE_ID = 1 when I have a USB with the UART.  What happened is I then reversed these to do some validation and for some reason the NOBLE_HCI_DEVICE_ID = 1 wasn't pinning the Noble code to the UART.  It ended up being how the variable was getting set.  I modified the hci.js file to expose a this._deviceId so I could determine which hci the code was getting pinned to and found the error.  

It would be great if you would expose the deviceId in both Bleno and Noble so we can write it out to find these types of issues.

Your code has worked extremely well and I thank you for all the hard work.

Regards,
Vince