noble/noble #416: No state change in strict mode

BasKiers, 2016-06-20:
I am unable to use Noble in strict mode. Since I am using node 4.4.5 I want to use strict mode to take advantage of const/let and classes. Example code that does not work for me:

```
'use strict';

var noble = require('noble');
noble.on('stateChange', function(state){
        console.log(state);
    }
}
``` 

When I remove `'use strict';` the `stateChange` event does get fired.

sandeepmistry, 2016-06-26:
@BasKiers seem ok to me on OS X with Node.js v4.4.6:

```javascript
'use strict';

var noble = require('noble');

noble.on('stateChange', function(state){
  console.log(state);
});
```

What OS are you using? Could you also provide debug output? ```DEBUG=* node <file>.js`

sandeepmistry, 2016-08-22:
@BasKiers ping ...

sandeepmistry, 2016-08-28:
I'm closing this due to lack of activity.

closed 2016-08-28 by sandeepmistry

BasKiers, 2016-08-29:
I'm sorry for my inactivity on this issue. This was before my holiday and I cannot reproduce the issue I had back then so this was probably not a bug. Thanks.