noble/noble #418: Using noble together with bleno results in no `stateChange` to `poweredOn`

corneyl, 2016-06-29:
I'm trying to use noble and bleno in one project, as e.g. is done in the project [sandeepmistry/node-bleacon](/sandeepmistry/node-bleacon), but as soon as I `import noble from 'noble'`, bleno does not start advertising. But this should be possible right?

This is my debug output when using `$env:DEBUG='noble,bleno'` It seems like bleno does not get a stateChange to poweredOn.

```ps
PS D:\test> node .\src\node\ble\index.js
  bleno platform win32 +0ms
  bleno stateChange poweredOff +6ms
  noble stateChange poweredOff +24ms
  bleno addressChange 00:1a:7d:da:71:0b +96ms
  bleno addressChange 00:1a:7d:da:71:0b +890ms
  noble stateChange poweredOn +2ms
noble -> stateChange: poweredOn
```
When I import bleno first instead of noble, this happens:

```ps
PS D:\Werk\LMS\e-marine-settings-app> node .\src\node\ble\index.js
  noble stateChange poweredOff +0ms
  bleno platform win32 +20ms
  bleno stateChange poweredOff +3ms
  noble addressChange 00:1a:7d:da:71:0b +987ms
  bleno stateChange poweredOn +2ms
bleno -> stateChange: poweredOn
```

I want to create a central device which acts as server, which should be [possible](https://devzone.nordicsemi.com/question/232/what-is-a-client-and-server-in-ble/?answer=233#post-id-233).

To reconstruct this issue, you only have to load the libraries, e.g. using ES6:
```js
import noble from 'noble';
import bleno from 'bleno';
```
And one or both of the libs will not change state to `poweredOn`.

corneyl, 2016-07-06:
Ok thanks, maybe it might be handy to throw an error. :smiley:

sandeepmistry, 2016-07-13:
@corneyl first we have to find a way to programmatically detect both are running in the same process :) Do you have time to look into this?

elovin, 2017-10-17:
I'm trying to develop a Bluetooth mesh network using noble and bleno on each device (connecting to the network and extending it), the app should run on linux, osx and windows.
So has this issue been resolved by now?