noble/noble #419: Exception when trying to enable notify on a characteristics

papajim, 2016-07-02:
I use the following code to enable notify on a characteristic, but it results in an uncaughtException with the following message "Range Error: index out of range". I'm not sure whether this is a bug or me not using the library as expected.

```javascript
if(connectedDevice.state==="connected"){
    ...
    connectedDevice.discoverSomeServicesAndCharacteristics(null, characteristicUUIDs, function(error, services, characteristics){
        characteristics[0].notify(true, function(error){
            console.log("Notification Enabled");
        });
    });
    ...
}
```
I have also tried the following code, but it results in the same error.
```javascript
if(connectedDevice.state==="connected"){
    ...
    connectedDevice.discoverServices(serviceUUIDs, function(error, services) {
        var service = services[0];
        service.discoverCharacteristics(characteristicsUUID, function(error, characteristics) {
            var characteristic = characteristics[0];
            characteristic.notify(true, function(error) {
                console.log("Notification Enabled");
            });
        });
    });
   ....
}
```
I use noble.js 1.3.0, this code is used on the central device and characteristics[0] object is not null.
I have also tried noble.js 1.6.0 with the new way of enabling notifications "characteristic.subscribe", but I still get the same error.

sandeepmistry, 2016-07-03:
Hi @papajim,

noble should definitely handle this case in a nicer way than crashing.

Could you please share the full exception and which OS you are using? As well as what the value of `characteristics[0].uuid` and `characteristics[0].properties` is?

Maybe you are trying to subscribe to a characteristic that does not have a `notify` property?

papajim, 2016-07-03:
Hello @sandeepmistry thanks for your reply.

I'm using Windows 7 and I'm trying to enable notify on a custom characteristic of a custom service. characteristics[0].uuid = 19202122232425262728293031323334 (hex)
characteristics[0].properties = ["read", "notify"]

[image]
[image]

papajim, 2016-07-25:
*Bump*
Hello @sandeepmistry do you have any idea what's going on ?

sandeepmistry, 2016-07-27:
@papajim could you please provide debug logs. You need to set the `DEBUG` environment variable to `*`. I think `setenv DEBUG=*` is the way to do this.

sandeepmistry, 2016-08-22:
@papajim ping ...

sandeepmistry, 2016-08-28:
I'm closing this due to lack of activity.

closed 2016-08-28 by sandeepmistry