noble/noble #420: Help connecting to known devices

lessonpix, 2016-07-02:
Hi folks,

I've had much more luck since moving from the CSR dongle to the RPi3 internal adapter (yay).  

What I'm seeing now is more of a timing issue that may be related to my newness to Node-style coding.  I'd like some help getting this done right, and I'll be happy to write up a guide in the Wiki when it's working.

I have 8 devices I'm working with: 7 BLE bulbs and an adafruit Bluefruit chip (which is irrelevant here).  I know all of their UUIDs and I have logical Ids (positions in the room) that map to them.  Here's my current approach:

1) Start Discovering at process startup
2) In on("discover) I connect(), then discoverServices() to see if it has a bulb Characteristic and, if so, I store the Peripheral & Characteristic objects in a map and set that entry to "valid".
3) I set a timer to periodically adjust the lights (per logic that's not relevant) every 200ms
4) When I see I have all the devices, I stopScanning().

What I'm seeing is that if I do the logic in #2 above, I miss the discovery event for some of the lights.  So I am re-working this to JUST store the peripheral object in the map and then loop through those after I have all 7 bulbs.  However, I'm finding that EVEN THAT connect() call is enough to make me miss discover events.

Given that I have the addresses, is there a better approach?  I don't see a way to connect() to peripherals by UUID (unless I'm missing something), and I LIKE the discover() approach because it would work well if the lights are turned off and on (they'd be removed & replaced in the map).

Sample code is attached... it relies on onecolor, but that could easily be removed

When the connectAndSetup() call (line44) is commented out, I get this output:
```
pi@pi3:~/fadecandy/examples/node $ node test.js
State Changed: poweredOn
discovered: b4:99:4c:55:38:d1
discovered: b4:99:4c:64:85:26
discovered: b4:99:4c:64:84:cb
discovered: b4:99:4c:64:83:f4
discovered: b4:99:4c:64:7d:bd
discovered: b4:99:4c:64:85:01
discovered: b4:99:4c:64:83:35
discovered: f2:bb:cf:78:2c:85
```

When I uncomment it, I get this:
```
Cpi@pi3:~/fadecandy/examples/node $ node test.js
State Changed: poweredOn
discovered: b4:99:4c:64:7d:bd
discovered: b4:99:4c:55:38:d1
discovered: b4:99:4c:64:83:f4
Connecting to b4:99:4c:64:7d:bd
null
Found a Bulb - b4:99:4c:64:7d:bd
checking for the map
Connecting to b4:99:4c:55:38:d1
null
Found a Bulb - b4:99:4c:55:38:d1
checking for the map
Connecting to b4:99:4c:64:83:f4
null
Found a Bulb - b4:99:4c:64:83:f4
checking for the map
#the output stops here
```
It looks like if that the first attempt to connect causes the scanning/disconnect to fail: I've attached the hcidump output of that second attempt.

Any thoughts here?  Again, this seems like a usage that will be pretty common - people who know what peripherals to expect and want to configure them on discovery.

Thanks and again, I'll be happy to write 
[test.js.txt](https://github.com/sandeepmistry/noble/files/344752/test.js.txt)

[hcidump.out.txt](https://github.com/sandeepmistry/noble/files/344754/hcidump.out.txt)

up what the end result is as a guide.

Bill

lessonpix, 2016-07-02:
After adding lots of safety checks and JUST saving the peripheral, waiting until all had been connected,  I have this connected to 6 of 7 90+% of the time. and 7 of 7 50% of the time.  When it fails I see this message in hcidump:
```
> HCI Event: Command Status (0x0f) plen 4
    LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 1
> HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 68 reason 0x3e
    Reason: Connection Failed to be Established
```
Now, some googling tells me that that error 0x3e can "just happen" sometimes (yuk).  However, don't see this error being returned (all 'error' parameters are null), so I don't know how to retry this...

Any help would be appreciated.  I've included the new code for reference

Bill
[test.js.txt](https://github.com/sandeepmistry/noble/files/344774/test.js.txt)

sandeepmistry, 2016-10-10:
> Given that I have the addresses, is there a better approach? I don't see a way to connect() to peripherals by UUID (unless I'm missing something), and I LIKE the discover() approach because it would work well if the lights are turned off and on (they'd be removed & replaced in the map).

This is the right approach, noble is based on the OS X/iOS CoreBluetooth stack which requires scanning + discovering prior to connecting.

> Any thoughts here? Again, this seems like a usage that will be pretty common - people who know what peripherals to expect and want to configure them on discovery.

I would suggest trying out calling `noble.startScanning(...)` again after the connect is done so that any devices that were missed can be discovered again.

> Now, some googling tells me that that error 0x3e can "just happen" sometimes (yuk). However, don't see this error being returned (all 'error' parameters are null), so I don't know how to retry this...

Please provide a full HCI dump to share more context.

sandeepmistry, 2016-10-30:
I'm closing this due to lack of activity. If you are still interested please re-open with the requested info.

closed 2016-10-30 by sandeepmistry