noble/noble #422: Unable to get services or characteristics

Casper-Oakley, 2016-07-05:
Hey!
I've recently been trying to setup noble on a raspberry pi 3 using an external bluetooth dongle. Installation was all fine and noble was able to run without errors. However, after connecting (successfully) to a peripheral, it isn't able to find any services. It is worth mentioning that I can read and write characteristics using gatttool. The version of bluez I'm running is 5.37 in case that affects anything.

pizza example output:

`scanning...
found peripheral: { localName: 'PizzaSquat',
  txPowerLevel: undefined,
  manufacturerData: undefined,
  serviceData: [],
  serviceUuids: [ '13333333333333333333333333333337' ] }
`

Hcidump when running pizza example:

`HCI sniffer - Bluetooth packet analyzer ver 5.40
device: hci0 snap_len: 1500 filter: 0xffffffff
2016-07-05 14:52:31.420811 < HCI Command: Set Event Mask (0x03|0x0001) plen 8
    Mask: 0xfffffbff07f8bf3d
2016-07-05 14:52:31.422047 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2016-07-05 14:52:31.422102 < HCI Command: LE Set Event Mask (0x08|0x0001) plen 8
    mask 0x1f00000000000000 (Reserved)
2016-07-05 14:52:31.424043 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2016-07-05 14:52:31.424100 < HCI Command: Read Local Version Information (0x04|0x0001) plen 0
2016-07-05 14:52:31.458054 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0x1d86
    LMP Version: 4.0 (0x6) LMP Subversion: 0x1d86
    Manufacturer: Cambridge Silicon Radio (10)
2016-07-05 14:52:31.458115 < HCI Command: Write LE Host Supported (0x03|0x006d) plen 2
  01 00 
2016-07-05 14:52:31.459042 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    00 
2016-07-05 14:52:31.459105 < HCI Command: Read LE Host Supported (0x03|0x006c) plen 0
2016-07-05 14:52:31.461040 > HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
    00 01 00 
2016-07-05 14:52:31.461092 < HCI Command: Read BD ADDR (0x04|0x0009) plen 0
2016-07-05 14:52:31.463038 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr 00:15:83:D1:BF:90
2016-07-05 14:52:31.463069 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2016-07-05 14:52:31.464045 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2016-07-05 14:52:31.464097 < HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
    type 0x01 (active)
    interval 10.000ms window 10.000ms
    own address: 0x00 (Public) policy: All
2016-07-05 14:52:31.466048 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2016-07-05 14:52:31.489670 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
2016-07-05 14:52:31.493047 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-07-05 14:52:31.573050 > HCI Event: LE Meta Event (0x3e) plen 33
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
[... 108 lines cut ...]
    handle 41 packets 1
2016-07-05 14:52:32.064367 > ACL data: handle 41 flags 0x02 dlen 9
    ATT: Error (0x01)
      Error: Request not supported (6)
      Error (0x01) on handle 0x0000
2016-07-05 14:52:32.094123 > HCI Event: LE Meta Event (0x3e) plen 10
    LE Connection Update Complete
      status 0x00 handle 41
      interval 70.00ms, latency 0.00ms, superv. timeout 2000.00ms
2016-07-05 14:52:33.221946 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2016-07-05 14:52:33.223294 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2016-07-05 14:52:33.223364 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 41 reason 0x13
    Reason: Remote User Terminated Connection
2016-07-05 14:52:33.226281 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2016-07-05 14:52:33.285287 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 41 reason 0x16
    Reason: Connection Terminated by Local Host
`

Casper-Oakley, 2016-07-07:
Ignore this. Turns out the error was on the peripheral's side. Closing the issue.

closed 2016-07-07 by Casper-Oakley

LebDev, 2016-12-23:
So what was the problem? I'm facing the same issue, using mac OSX10.12 as central(Noble) and RPI3 as peripheral(Bleno).