noble/noble #425: How do we contribute (monetarily, hardware, etc)?

zrecore, 2016-07-10:
Hi Sandeep,

NobleJS is a wonderful package. I've gotten BLE comms into UE4, ThreeJS, and other platforms thanks to the hardwork put into this package. (I also noticed some big names in the maker field using NobleJS for desktop apps)

Is there any official way of contributing money and/or equipment to support this package?

Thanks in advance.

- Alex

sandeepmistry, 2016-07-13:
Hi @zrecore,

Thanks for the positive feedback! It's great to hear about how you've successfully used noble in your projects.

I have received various equipment to test with from others through people contacting me via email.

Do you have any examples on how other open source projects have setup "official ways of contributing money and/or equipment"? Or do you have any suggestions on how to do this?

Thanks!

zrecore, 2016-07-13:
Usually, I've seen PayPal buttons (or payment page) - where said funds are
used for hosting, equipment, etc.

A list of hardware that the NobleJS project can benefit from would be great
(along w/ some way of sending stuff).

I am grateful for your work on NobleJS, as it becomes even more relevant to
an expanded set of platforms - enabling BLE communication into environments
where it may have been near impossible in the past.

I also believe in supporting open source work in a meaningful matter.

Thanks to your NobleJS package, and the imuduino-3js server (thanks
@Nodebotanist and @RussellHay), I was able to connect UE4 via websockets to
NobleJS for BLE communications on Windows 10. See
https://youtu.be/3G4ZTAhhAcQ

- Alex

mbifulco, 2016-08-11:
Echoing this.  This is a fantastic library, and very well documented and supported.  Would love to show some appreciation, @sandeepmistry!

sandeepmistry, 2016-08-21:
@zrecore @mbifulco thanks!

Does anyone have experience with something like OpenCollective? https://opencollective.com

zrecore, 2016-08-22:
Oh cool! I've never heard of it - but if that's something you want to try,
I'm chipping in some loot.

mbifulco, 2016-08-22:
Ditto.  Seems legit to me - like a Patreon for OSS.

sandeepmistry, 2016-09-05:
Here's a link to OpenCollective's TOS: https://docs.google.com/document/d/1-hajYd7coL05z2LTCOKXTYzXqNp40kPuw0z66kEIY5Y/pub

@jacobrosenthal what do you think?

jacobrosenthal, 2016-09-05:
Gulp recently started using them. My understanding is fundraising has been very slow going so as usual with these things don't actually expect much income, but I heard good things about their responsiveness when they were setting it up.

sandeepmistry, 2016-09-11:
I've submitted the form to start a collective, stay tuned!

> Dear Sandeep Mistry,
>
> Thanks for signing up your GitHub repository: noble. We are reviewing new requests as fast as we can.
>
> While you wait, check out other open source collectives and see how they are spending the money they have raised.
>
> Stay tuned for updates.
>
> Warmly,
>
> – Team Open Collective

zrecore, 2016-09-11:
Awesome! Keep us posted as soon as it's ready. :-)

Alex Albino
Sr. Fullstack Developer

(707) 246-6864
LinkedIn: linkedin.com/in/alexalbino

sandeepmistry, 2016-09-17:
noble has been approved for Open Collective!

https://opencollective.com/noble

Badges have been added to the read me in #464 as well.

closed 2016-09-17 by sandeepmistry