noble/noble #427: Running Noble on a Raspberry PI 3 where there are 2 BLE devices

vdharmon, 2016-07-12:
I'm seeing an issue on a Raspberry PI 3.  

Here's the configuration.
1) There is a USB attached to the device.  So hciconfig will show 2 ble devices.  USB is on port 0 and UART is on port 1.
2) I've configured Noble to run on hci0. Using NOBLE_HCI_DEVICE_ID=0.
3) Both hci devices are in the 'UP' state.
4) I have the latest version of the PI firmware. (4.4.15-v7+)

I'm using the include example file: **peripheral-explorer.js**.  I added more console.log() so I could determine code execution steps.

Executing from the command line: **sudo node peripheral-explorer.js e7b1090d2606** where the e7b1090d2606 is a device I have locally.

When I execute the above command with hci1 -> UP the code will run until it hits the **peripheral.discoverServices([], function (error, services)**.  Once it hits this line of code, the application stops running.  No exceptions or errors are generated.  The only way to stop the application is to kill it.

I then stop hci1, so hci1 -> DOWN.  I then execute the above command and everything works perfectly.

Thoughts on how to address the issue?

Regards,
Vince

sandeepmistry, 2016-07-18:
Hi @vdharmon,

Could you please provide an HCI dump capture and debug logs? ```sudo hcidump -t -x```, ```sudo DEBUG=* node <file>.js```

vdharmon, 2016-07-22:
Here's the information you requested.  

The dump.txt file is running **sudo hcidump -t -x -w dump.txt**
The node.txt file is running **sudo DEBUG=* node peripheral-explorer.js e7b1090d2606**

[dump.txt](https://github.com/sandeepmistry/noble/files/379190/dump.txt)
[node.txt](https://github.com/sandeepmistry/noble/files/379193/node.txt)

Regards,
Vince

vdharmon, 2016-07-23:
I've tried to take more time and get you more complete results.  I'm using Noble and Bleno on same PI but each is tied to it's own BLE.  They never use the same BLE.  

If I run the code for Bleno or Noble alone tied to either the USB or UART everything works fine.  Having the USB plugged in doesn't seem to cause any issues.  It's only when both are running at the same time.  

I've shared the tests I've ran below with result data.

**Test 1:**
When using Noble to call an external peripheral to get services and characteristics none are returned (fails).
When using a BLE such as LightBlue on IOs (tried with Android app also) I can connect to bleno and see services and characteristics on the PI (success).

Configuration: Using a USB and UART ble.
Running Bleno on USB hci0
Running Noble on UART hci1

sudo DEBUG=* NOBLE_HCI_DEVICE_ID=1 node peripheral-explorer.js c5bbf4af01d1  >> debugT1.txt

sudo hcidump -i hci1 -t -x -w hci1T1.txt
sudo hcidump -i hci0 -t -x -w hci0T1.txt

The result files:
[debugT1.txt](https://github.com/sandeepmistry/noble/files/379800/debugT1.txt)
[hci0T1.txt](https://github.com/sandeepmistry/noble/files/379802/hci0T1.txt)
[hci1T1.txt](https://github.com/sandeepmistry/noble/files/379804/hci1T1.txt)

==============================================

**Test 2:**
When using Noble to call an external peripheral to get services and characteristics are returned (success).
When using a BLE such as LightBlue on IOs (tried with Android app also) I can NOT connect to bleno and see services and characteristics on the PI (fails).

Configuration: Using 2 USB and UART ble.
Running Bleno on USB hci0
Running Noble on USB hci2

sudo DEBUG=* NOBLE_HCI_DEVICE_ID=2 node peripheral-explorer.js c5bbf4af01d1 >> debugT2.txt

sudo hcidump -i hci2 -t -x -w hci2T2.txt
sudo hcidump -i hci0 -t -x -w hci0T2.txt

The result files:
[debugT2.txt](https://github.com/sandeepmistry/noble/files/379806/debugT2.txt)
[hci0T2.txt](https://github.com/sandeepmistry/noble/files/379807/hci0T2.txt)
[hci2T2.txt](https://github.com/sandeepmistry/noble/files/379808/hci2T2.txt)

==============================================

If you need anymore details please let me know.

Regards,
Vince

sandeepmistry, 2016-09-07:
Hi @vdharmon,

Could you please try the latest master version `npm install sandeepmistry/noble`. It uses the latest `bluetooth-hci-socket` version that includes the following fix: https://github.com/sandeepmistry/node-bluetooth-hci-socket/commit/14dc3660f9e86695c6c75b92dc2222da2905549e

Let me know if it works, then I can publish a new version to npm.

vdharmon, 2016-09-20:
I'm still seeing the same issue.  Here's what I did.

1) I installed your code above for Noble.
2) Running bleno on the UART.
3) Running noble on the USB.
4) I can connect to bleno and see the characteristics using LightBlue.
5) Noble cannot connect to another BLE, it hangs on getting services.

The code fix was for Noble, but wouldn't there need to be a change for the Bleno code base also?

Regards,
Vince

sandeepmistry, 2016-09-21:
Hi @vdharmon,

Please make sure you are using the latest bleno and noble releases. Also, you need `bluetooth-hci-socket` v0.5.1, since it has the fix.

vdharmon, 2016-09-26:
I've updated all my code.  I'm still seeing an issue but slightly different.

The original issue where I could not connect to bleno when noble has a connection or noble connect to an external BLE when bleno is connected has been addressed.  Meaning I can connect.
 
1) I'm running **sudo DEBUG=* BLENO_HCI_DEVICE_ID=1 node peripheral.js >> bleno.log** .
2) I'm running **sudo DEBUG=* BLENO_HCI_DEVICE_ID=0 node peripheral-explorer.js e7b1090d2606 >> noble.log**.  I've slightly modified the code to set a timeout in the disconnect and restart scanning after disconnect to make sure the noble side will continuously trying to connect and read information from the external BLE.  I've ran this by itself (without bleno running) for an hour to make sure it functions properly.

Issue:
The noble code will continue to connect to external device, as it should whether another device is connected to the bleno or not as it should.  But now the bleno will not return the services it is broadcasting.  I've seen in the dump on the noble side **noble: unknown peripheral 7e26eeaacecb disconnected!** which is the external device connected to the bleno.  So it looks like noble is disconnecting the device connected to bleno which explains why the services are not returned.

Below are the debug dumps of the bleno and noble code.  
1) blenogood.txt contains running bleno without noble and connecting and seeing services.
2) noble.txt contains the running noble code during issue.
3) bleno.txt contains the running bleno code during issue.

[bleno.txt](https://github.com/sandeepmistry/noble/files/494058/bleno.txt)
[noble.txt](https://github.com/sandeepmistry/noble/files/494059/noble.txt)
[blenogood.txt](https://github.com/sandeepmistry/noble/files/494065/blenogood.txt)

If you need anymore details please let me know.

Regards,
Vince

sandeepmistry, 2016-09-26:
> 2) I'm running sudo DEBUG=* BLENO_HCI_DEVICE_ID=0 node peripheral-explorer.js e7b1090d2606 >> noble.log. 

Is this correct? You should be using `NOBLE_HCI_DEVICE_ID=0` for noble, let me know.

vdharmon, 2016-09-27:
Good catch.  

I've gone through and retested making sure I have the proper parameters set.  I'm still seeing an issue.

Test:
1) Running **sudo DEBUG=* BLENO_HCI_DEVICE_ID=1 node peripheral.js**.
2) Running **sudo DEBUG=* NOBLE_HCI_DEVICE_ID=0 node peripheral-explorer.js e7b1090d2606**.
3) Using my IPad (LightBlue BLE explorer) to connect to bleno.  I also tested with an Android.
4) The peripheral-explorer.js file has been modified to run continuously, with a delay of 5 seconds after a disconnect.  I've included it.

Issue:
Connect the IPad to bleno.  Noble will run reading services and characteristics properly.  Disconnect the IPad from the bleno.  Noble will read a portion of the services and then get stuck.  After a long delay, around a minute or so noble disconnects and starts reading again, but it then gets stuck at the same point over and over again.

As long as I leave the IPad connected to bleno, noble works fine but as soon as it's disconnected the first time I see the intermittent noble reading error.

If I then stop bleno from advertising noble again works properly.

Below are the debug dumps of the bleno and noble code.

1) blenogood.txt contains running bleno without noble and connecting and seeing services.
2) noble.txt contains the running noble code during issue.
3) bleno.txt contains the running bleno code during issue.
4) peripheral-explorer is the modified noble example file

[bleno.txt](https://github.com/sandeepmistry/noble/files/495724/bleno.txt)
[noble.txt](https://github.com/sandeepmistry/noble/files/495725/noble.txt)
[blenogood.txt](https://github.com/sandeepmistry/noble/files/495726/blenogood.txt)
[peripheral-explorer.txt](https://github.com/sandeepmistry/noble/files/495735/peripheral-explorer.txt)

If you need anymore details please let me know.

Regards,
Vince

sandeepmistry, 2016-10-10:
Hi @vdharmon,

Could you please also provide HCI dump captures (`sudo hcidump -t -x -i hci0` and `sudo hcidump -t -x -i hci1`)?

I'm also wondering if https://github.com/sandeepmistry/node-bluetooth-hci-socket/pull/39 might help ...

vdharmon, 2016-10-11:
Here's the files.

[bleno.txt](https://github.com/sandeepmistry/noble/files/521739/bleno.txt)
[noble.txt](https://github.com/sandeepmistry/noble/files/521741/noble.txt)
[hci0.txt](https://github.com/sandeepmistry/noble/files/521742/hci0.txt)
[hci1.txt](https://github.com/sandeepmistry/noble/files/521740/hci1.txt)

sandeepmistry, 2016-10-15:
@vdharmon thanks, did you get a chance to try out the PR I linked to?

If not please see instructions in https://github.com/sandeepmistry/noble/issues/474#issuecomment-252724816.

vdharmon, 2016-10-17:
Yes I tried using the package **npm install sandeepmistry/node-bluetooth-hci-socket#rework-kernel-workarounds**

I get worse results.  Noble will scan and connect once then after that it gets stuck and moves no farther.  Here's the log files.

[hci0.txt](https://github.com/sandeepmistry/noble/files/533989/hci0.txt)
[hci1.txt](https://github.com/sandeepmistry/noble/files/533987/hci1.txt)
[noble.txt](https://github.com/sandeepmistry/noble/files/533988/noble.txt)
[bleno.txt](https://github.com/sandeepmistry/noble/files/533990/bleno.txt)