noble/noble #428: additional filtering options in noble.startScanning()

monteslu, 2016-07-13:
Currently we can pass an array of service UUIDs to search for in `noble.startScanning();`.

In cases where the device doesn't broadcast the service we want to search for we either pass in an empty array or no parameters to search for any BLE device and manually filter for device we want.

While this works for most platforms, WebBluetooth does not currently allow for an open scan.  Instead we can pass a device `name`, `namePrefix`, or a `services` array: 
https://webbluetoothcg.github.io/web-bluetooth/#dom-requestdeviceoptions-filters

I propose that `noble.startScanning()` accept an options object as well as the array we're currently using (for backwards compatibility). If the parameter is an object, it should respect the filtering criteria listed above.  If the parameter is an array, or empty it should behave as it currently does.

Thoughts?

sandeepmistry, 2016-07-14:
@monteslu this sounds good to me, just to confirm the structure of the options object would be the following?

```javascript
{
  name: <string>, // optional
  namePrefix: <string>, // optional
  services: [<string>, ...] // optional
}
```

sandeepmistry, 2016-10-10:
@monteslu ping ^