noble/noble #434: Requiring noble package causes errors on npm install

mbifulco, 2016-07-26:
Hi there - on any project including noble, I'm getting errors when I try to install the noble package.  It looks like it's because of a dependency on node-usb, specifically this issue: https://github.com/nonolith/node-usb/issues/109

This is a bit of an odd place to report the error, I suppose, but according to that issue, it looks like the author of that library will no longer be supporting it.  

Steps to reproduce:
`npm init` a new node project
`npm install --save noble`

You'll see errors having to do with 

`usb@1.1.2 install: node-pre-gyp install --fallback-to-build`

Error produced on:
- Raspbian for Raspberry Pi 3

sandeepmistry, 2016-07-27:
@mbifulco could you please share also console output of the failure.

Also, did you follow the read me: https://github.com/sandeepmistry/noble#ubuntudebianraspbian

`node-usb` is optional on Linux anyways, just needed for Windows.

mbifulco, 2016-07-27:
@sandeepmistry so it looks like I missed installing `libudev-dev` from the prereqs in the readme.  Now the only output I get from `npm install` that looks funny is the following:

```
npm WARN optional Skipping failed optional dependency /noble/xpc-connection:
npm WARN notsup Not compatible with your operating system or architecture: xpc-connection@0.1.4
```

But I assume that's okay because I'm running on a pi, right?

sandeepmistry, 2016-07-28:
> But I assume that's okay because I'm running on a pi, right?

Yep, `xpc-connection` is only needed for OS X.

closed 2016-07-28 by sandeepmistry