noble/noble #436: Access to all advertising and scan response packet data

bluetooth-mdw, 2016-07-27:
It would be handy sometimes if there was a way to have the entire advertising/scan response content passed to an application which is scanning rather than the specific fields which are currently parsed. Right now, to give an example, I'm interested in ADV packets which include the "List of 16 bit Solicitation UUIDs" AD type.

So ideally, parsing and formulation of JSON advertisement objects could support all types defined by the SIG's Core Spec Supplement or alternatively, simply include a raw copy of the entire packet as an array property of the current JSON object.

Thanks

sandeepmistry, 2016-07-28:
As I mentioned in https://github.com/sandeepmistry/noble/pull/426#issuecomment-234811146 I'm not keen on exposing raw EIR data.

No issues adding support to parse solicited service UUID's.  OS X also supports them via the `CBAdvertisementDataSolicitedServiceUUIDsKey` key ([more details](https://developer.apple.com/library/mac/documentation/CoreBluetooth/Reference/CBCentralManagerDelegate_Protocol/index.html#//apple_ref/doc/constant_group/Advertisement_Data_Retrieval_Keys)).

Why don't you get started on the Linux side, see [lib/hci-socket/gap.js](https://github.com/sandeepmistry/noble/blob/master/lib/hci-socket/gap.js) and submit a pull request. I can handle the OS X side later if you don't have access to a Mac.

bluetooth-mdw, 2016-07-28:
OK...platform differences. Makes sense.

Not surprisingly perhaps I did go ahead and add what I needed to gap.js yesterday! So yes, I'll submit a PR.

Cheers

Martin

sandeepmistry, 2016-09-05:
I'm closing this now the #438 has been merged.

Unfortunately, OS X does not seem to report `CBAdvertisementDataSolicitedServiceUUIDsKey` so this is a Linux only field for now.

closed 2016-09-05 by sandeepmistry