noble/noble #437: Unable to read services/characteristics on RPi3 with A&D_UA-651BLE device

tobyjaffey, 2016-07-27:
Using Raspberry Pi 3 with Noble, I am unable to list services and characteristics from the `A&D_UA-651BLE_5F087D`.

Can anyone offer any things to try?

My Raspberry Pi bluez version is 5.23-2+rpi2

	pi@raspberrypi:~ $ dpkg --status bluez | grep '^Version:'
	Version: 5.23-2+rpi2

Noble on OSX does work for the same device, though it must be paired first. Connecting to the device (using Noble) triggers the pairing dialog in OSX.

Once paired, on OSX, I run `node node_modules/noble/examples/peripheral-explorer.js 5C:31:3E:5F:08:7D`

	peripheral with ID 9bf5e048663640f492f83072c5c6fd61 found
  		Local Name        = A&D_UA-651BLE_5F087D
  		Service Data      =
  		Service UUIDs     = 1810

		services and characteristics:
		1810 (Blood Pressure)
  		2a35 (Blood Pressure Measurement)
    	...

On the Pi3, I first pair the device using:

	$ bluetoothctl
		scan on
		...
		pair 5C:31:3E:5F:08:7D
		[CHG] Device 5C:31:3E:5F:08:7D Paired: yes
		Pairing successful

I then run `sudo node node_modules/noble/examples/peripheral-explorer.js 5c:31:3e:5f:08:7d`

The full output I get is (note no listed services/characteristics):

	peripheral with ID 5c313e5f087d found
  		Local Name        = A&D_UA-651BLE_5F087D
  		Service Data      =
  		Service UUIDs     = 1810

	services and characteristics:

This is the output from `hcidump -t -x` while `peripheral-explorer.js` was running.

    HCI sniffer - Bluetooth packet analyzer ver 5.23
    device: hci0 snap_len: 1500 filter: 0xffffffff
    2016-07-27 15:37:49.123064 < HCI Command: Set Event Mask (0x03|0x0001) plen 8
        Mask: 0xfffffbff07f8bf3d
    2016-07-27 15:37:49.123548 > HCI Event: Command Complete (0x0e) plen 4
        Set Event Mask (0x03|0x0001) ncmd 1
        status 0x00
    2016-07-27 15:37:49.124246 < HCI Command: LE Set Event Mask (0x08|0x0001) plen 8
        mask 0x1f00000000000000 (Reserved)
    2016-07-27 15:37:49.124709 > HCI Event: Command Complete (0x0e) plen 4
        LE Set Event Mask (0x08|0x0001) ncmd 1
        status 0x00
    2016-07-27 15:37:49.124737 < HCI Command: Read Local Version Information (0x04|0x0001) plen 0
    2016-07-27 15:37:49.125179 > HCI Event: Command Complete (0x0e) plen 12
        Read Local Version Information (0x04|0x0001) ncmd 1
        status 0x00
        HCI Version: 4.1 (0x7) HCI Revision: 0xb6
        LMP Version: 4.1 (0x7) LMP Subversion: 0x2209
        Manufacturer: Broadcom Corporation (15)
    2016-07-27 15:37:49.125336 < HCI Command: Write LE Host Supported (0x03|0x006d) plen 2
      01 00 
    2016-07-27 15:37:49.125707 > HCI Event: Command Complete (0x0e) plen 4
        Write LE Host Supported (0x03|0x006d) ncmd 1
        00 
    2016-07-27 15:37:49.125804 < HCI Command: Read LE Host Supported (0x03|0x006c) plen 0
[... 100 lines cut ...]
        Disconnect (0x01|0x0006) status 0x00 ncmd 1
    2016-07-27 15:37:59.837697 > HCI Event: Number of Completed Packets (0x13) plen 5
        handle 64 packets 2
    2016-07-27 15:37:59.837820 > HCI Event: Disconn Complete (0x05) plen 4
        status 0x00 handle 64 reason 0x16
        Reason: Connection Terminated by Local Host
    2016-07-27 15:37:59.921445 < HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
        type 0x00 (passive)
        interval 60.000ms window 30.000ms
        own address: 0x00 (Public) policy: All
    2016-07-27 15:37:59.922744 > HCI Event: Command Complete (0x0e) plen 4
        LE Set Scan Parameters (0x08|0x000b) ncmd 1
        status 0x00
    2016-07-27 15:37:59.922800 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
        value 0x01 (scanning enabled)
        filter duplicates 0x01 (enabled)
    2016-07-27 15:37:59.925027 > HCI Event: Command Complete (0x0e) plen 4
        LE Set Scan Enable (0x08|0x000c) ncmd 1
        status 0x00
    2016-07-27 15:37:59.925082 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
        value 0x00 (scanning disabled)
        filter duplicates 0x01 (enabled)
    2016-07-27 15:37:59.927549 > HCI Event: Command Complete (0x0e) plen 4
        LE Set Scan Enable (0x08|0x000c) ncmd 1
        status 0x00

sandeepmistry, 2016-07-27:
Hi @tobyjaffey,

At this time pairing with `bluetoothctl` is not supported, as noble does not use the Linux kernels SMP stack, and has it's own built-in SMP stack that only supports "Just Works" pairing.

I would suggest unpairing using `bluetoothctl` and using noble to access a secure characteristic on the device, it will inmate the pairing process.

tobyjaffey, 2016-07-28:
@sandeepmistry,

Ok, that makes sense.

However, with this device, I get no response to `discoverServices()` (presumably until I have paired). How can I access a secure characteristic without the object that comes back from `discoverServices()` and `discoverCharacteristics()`?

I know the uuid's. Is it possible to build the object for `characteristic.read()` without having scanned?

Thanks.

ringtailsoftware, 2016-08-04:
I have tracked this down further. In my hcidump output, the slave device sends to noble "SMP: Security Request (0x0b)".

This is not handled in smp.js. According to https://books.google.co.uk/books?id=-LMq0NhoEQgC&pg=PA223&lpg=PA223&dq=smp+security+request+0x0b&source=bl&ots=AYsVtiSATz&sig=6CB0Oh1r_kZT42Na3f4mrXfOmZM&hl=en&sa=X&ved=0ahUKEwjT_fGw06jOAhVIJsAKHWw_C044ChDoAQgkMAE#v=onepage&q=smp%20security%20request%200x0b&f=false this should trigger the master to send a pairing request.