noble/noble #439: Best practices for getting noble to poll for notifications quickly and consistently?

mbifulco, 2016-08-01:
Hi there - 
I'm running noble on a Raspberry pi 3 device, using an external antenna - trying to read heart rate from several BLE heart rate monitors at a time. 

I've basically taken the intro code from this repo and changed it to only look for the Heart Rate Service (`180d`) - which has done a good job of ensuring that my Pi only connect to devices I'm interested in... but past that I haven't changed much.

I've noticed that noble doesn't seem to consistently connect to all of the devices I've got in the room with me.  Even with an CSR antenna, often times it only seems to connect to 2-4 devices, rather than all 5 that are possible.  Is there something I can do to see what's happening here?

In addition to that, once the devices are found, paired, and connected to, often times the notifications I subscribe to for the heart rate service don't happen very quickly (initially).  Sometimes I'm waiting 30s-1min before a single heart rate comes back from a given device.  Is that a known issue?

mbifulco, 2016-08-01:
I should add that testing on a Macbook Pro yields different results - I hit the 6 device limit right away (using the internal bluetooth radio), and get notifications pretty quickly thereafter, every time.

sandeepmistry, 2016-09-07:
Hi @mbifulco,

>  Is there something I can do to see what's happening here?

Let's start with an HCI dump capture: `sudo hcidump -t -x` - this will shed some light as to what's going on at the adapter level.

sandeepmistry, 2016-09-17:
Closing this due to lack of activity for now.

closed 2016-09-17 by sandeepmistry