noble/noble #440: "userDescriptionDescriptor.readValue is not a function" error fix?

sharkwheels, 2016-08-02:
I'm getting a userDescriptionDescriptor.readValue is not a function error on a raspberry pi, when checking out some light blue beans, it doesn't happen on OSX. I'm unclear as to what the error is and how to fix it, wondering if someone can help shed some light on this?

TypeError: userDescriptionDescriptor.readValue is not a function
    at /home/pi/noble_test/explorer.js:107:55
    at /home/pi/noble_test/node_modules/async/dist/async.js:2736:29
    at /home/pi/noble_test/explorer.js:103:34
    at wrappedIteratee (/home/pi/noble_test/node_modules/async/dist/async.js:2733:17)
    at eachOfArrayLike (/home/pi/noble_test/node_modules/async/dist/async.js:1002:13)
    at eachOf (/home/pi/noble_test/node_modules/async/dist/async.js:1052:9)
    at Object.<anonymous> (/home/pi/noble_test/node_modules/async/dist/async.js:2753:17)
    at /home/pi/noble_test/explorer.js:100:29
    at Characteristic.<anonymous> (/home/pi/noble_test/node_modules/noble/lib/characteristic.js:115:7)
    at Characteristic.g (events.js:286:16)

sandeepmistry, 2016-08-22:
@sharkwheels could you please share:

 * a simple Node.js app to reproduce the issue
 * debug logs: `node DEBUG=* <file>.js`
 * HCI dump capture: `sudo hcidump -t -x`

sharkwheels, 2016-08-23:
@sandeepmistry 

The node app is a modified explorer.js example: http://pastebin.com/xJZ0rDcC

Debug logs: 
< Debugger listening on [::]:5858
connecting to 127.0.0.1:5858 ... ok
break in explorer.js:1
> 1 var async = require('async');
  2 var noble = require('noble');
  3 var peripheralIdOrAddress = "d03972c9d02e";

sudo hcidump -t -x
HCI sniffer - Bluetooth packet analyzer ver 5.41
device: hci0 snap_len: 1500 filter: 0xffffffff

sandeepmistry, 2016-08-24:
I don't see any debug output. I had an incorrect command `sudo DEBUG=* node <file>.js`.

Also, make sure to run `hcidump` in a different terminal window at the same time as node is running.  There will be way more output.

sharkwheels, 2016-08-24:
Ah ok. Sorry. re: multiple terminal sessions. 

Here's the debug output: http://pastebin.com/P1hHn9nd
Here's the hcidump: http://pastebin.com/JsvSmWXa

sandeepmistry, 2016-09-05:
@sharkwheels thanks.

It looks like a disconnect is set out during the read:

```
2016-08-24 14:43:29.020871 > ACL data: handle 64 flags 0x02 dlen 10
    ATT: Find Information resp (0x05)
      format: uuid-16
        handle 0x0005, uuid 0x2901 (GATT(desc) User Description)
2016-08-24 14:43:29.029498 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2016-08-24 14:43:29.029909 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2016-08-24 14:43:29.030759 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 64 reason 0x13
    Reason: Remote User Terminated Connection
2016-08-24 14:43:29.031170 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2016-08-24 14:43:29.036466 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 1
2016-08-24 14:43:29.036619 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x16
    Reason: Connection Terminated by Local Host
```

So maybe the kernel is interfering? What kernel are you using (`uname -a`)?

jasonc1025, 2016-09-06:
Hi @sandeepmistry, @sharkwheels, and others,
I'm experiencing the same symptoms, running 'sudo DEBUG=* node peripheral-explorer.js FC:12:B8:0F:1C:60' - on a Raspberry Pi3 under:

`Linux raspberrypi 4.4.19-v7+ #906 SMP Tue Aug 23 15:53:06 BST 2016 armv7l GNU/Linux`

It ran well on my SDcard-1, but on my SDcard-2 - I experienced a similar symptom as noted below (in red, sorry if I captured too much info... just being convservative):

[image]

On the left is the GOOD btmon log (SDcard-1) and on the right is the BAD btmon log (SDcard-2).  The key difference is that for SDcard-2, I've been experiementing with installing upgrades for 1) WaveShare 3.2in touchsreen, 2) Webix, and 3) Meteor:

[image]

Personally, I'm excited learning about Node/Noble... since I'm developing a Robotics Educational Kit for kids.  We're trying to use a RaspberryPi3 w/ a touch screen as a Bluetooth Remote-Controller (thus Noble) to control an Ardunio-based (Bluefruit nRF8001) bot.  Feel free to check out our website: [www.EandE.world](url).

Thanks @sandeepmistry and others for any help.  : )

sandeepmistry, 2016-09-06:
@sharkwheels could you please provide text logs instead of screenshots next time.

Also, what if you log the value of ` userDescriptionDescriptor` just to get more details.

jasonc1025, 2016-09-08:
Thanks for the request for the text logs @sandeepmistry, I'll try to extract them on my next submission.  The screenshots do appear unreadable, yet actually I've been able to zoom in at 250% (via my Chrome browser) to actually see the text more clearly.  As per your request, the 'yellow' box above is the 'userDescriptionDescriptor', and it's interesting that the 'Bad' right-column does have an interesting 'HCI Command Disconnect' as well as a few other 'suspect' commands, all identified in 'red'.  Thanks for your time.  : )

sandeepmistry, 2016-09-11:
@jasonc1025 it's readable but not searchable via a browser or copying+pasting in a text editor.

I'm interested in what `console.log(userDescriptionDescriptor);` prints out just before `userDescriptionDescriptor.readValue(...)`.

jasonc1025, 2016-09-13:
Hi @sandeepmistry, thanks for explaining the advantage behind text logs. : ) 

Well this should be good news... when I booted up the RPi again to hopefully obtain new text logs, the previous symptoms -- related to userDescriptionDescriptor failures - never appeared again.  Apparently, the symptoms mysteriously resolved itself.  (As a background note, this all happened on the NOOB OS.)  I'm able to make the appropriate Bluetooth read and write operations - without any glitches.  Yay!  : )

sandeepmistry, 2016-09-26:
I'm closing this due to lack of activity, please re-open with request info from https://github.com/sandeepmistry/noble/issues/440#issuecomment-244796914 if you are still interested.

closed 2016-09-26 by sandeepmistry

poggenpower, 2017-05-31:
Run into the same error, added a `console.log('  userDescriptionDescriptor     = ' + userDescriptionDescriptor);` in peripheral-explorer.js right before the error line, "userDescriptionDescriptor" seems to contain simply "true".

noble/examples/peripheral-explorer.js output:
[noble.txt](https://github.com/sandeepmistry/noble/files/1041964/noble.txt)

hcidump -t -x:
[hcidump.txt](https://github.com/sandeepmistry/noble/files/1041966/hcidump.txt)

Vanuan, 2017-10-31:
```

  hci onSocketData: 0222200a0006000400050114000129 +29ms
  hci 	event type = 2 +0ms
  hci 		cid = 4 +0ms
  hci 		handle = 34 +0ms
  hci 		data = 050114000129 +0ms
  hci set scan enabled - writing: 010c20020001 +2ms
  hci disconnect - writing: 01060403220013 +2ms
/test.js:102
                            userDescriptionDescriptor.readValue(function(error, data) {
                                                      ^

TypeError: userDescriptionDescriptor.readValue is not a function

```

Vanuan, 2017-10-31:
Apparently `userDescriptionDescriptor` has a value of `true`

Vanuan, 2017-10-31:
It looks like true value is from here:
```
async.detect(['file1','file2','file3'], function(filePath, callback) {
    fs.access(filePath, function(err) {
        callback(null, !err)
    });
}, function(err, result) {
    // result now equals the first file in the list that exists
});
```

Hm... which version of async library does it require?

Vanuan, 2017-10-31:
Installing `async@0.2.9` did the trick