noble/noble #441: pizza example working?

dbetz, 2016-08-03:
I'm trying to run the pizza example on a pair of Beaglebone Green Wireless modules, one running the noble central and the other running the bleno peripheral. The central is able to find the peripheral and connect to it but discoverServices returns an empty array of services. Any idea why that might happen?

sandeepmistry, 2016-08-22:
@dbetz could you please share:

 * debug logs: `node DEBUG=* <file>.js`
 * HCI dump capture: `sudo hcidump -t -x`

Maybe `bluetoothd` is still running on the bleno side?

dbetz, 2016-08-23:
Yes, bluetoothd is running on both the noble and the bleno side. Is that a problem?

dbetz, 2016-08-23:
I just noticed this in the output of hcidump on the bleno side. Could this be causing my problem?

2016-08-23 15:27:33.156270 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x0c
    Error: Command Disallowed

sandeepmistry, 2016-08-24:
> Yes, bluetoothd is running on both the noble and the bleno side.

See https://github.com/sandeepmistry/bleno#linux for bleno side.

> I just noticed this in the output of hcidump on the bleno side. Could this be causing my problem?

No, this is just to clear advertisement state on startup.

Could you also share the full HCI dumps?

dbetz, 2016-08-24:
Here is the debug output as well as the hcidump output for the noble side. The hcidump for the bleno side was basically empty. Is that because I shut down bluetoothd?

[bleno.txt](https://github.com/sandeepmistry/noble/files/434845/bleno.txt)

[noble.txt](https://github.com/sandeepmistry/noble/files/434846/noble.txt)

sandeepmistry, 2016-08-28:
@dbetz 

> Is that because I shut down bluetoothd?

Yes, you'll also need to run `sudo hciconfig hci0` to power up the adapter after closing `bluetoothd`.

From the logs:

```
  bleno platform linux +53ms
  bleno stateChange poweredOff +26ms
```

dbetz, 2016-08-29:
Before the run that produced the logs I posted, I did the following commands from a root shell:

systemctl stop bluetooth
hciconfig hci0 up

I think I got those two commands from the bleno readme. Is that not sufficient to guarantee that bluetooth is powered up?

Thanks,
David

sandeepmistry, 2016-09-05:
@dbetz 

> I think I got those two commands from the bleno readme. Is that not sufficient to guarantee that bluetooth is powered up?

If your system is using `systemd` yes. You can check if `bluetoothd` is running using something like: `ps -eaf | grep bluetoothd`

dbetz, 2016-09-05:
I did that and verified that bluetoothd was not running prior to starting peripheral.js.

sandeepmistry, 2016-09-05:
@dbetz could you redo the HCI dump capture, since the previous ones you shared had the powered off state. Thanks.

sandeepmistry, 2016-09-17:
Closing this due to lack of activity for now.

closed 2016-09-17 by sandeepmistry