noble/noble #443: lastet is not reading TI (CC2650 SensorTag) devices anymore

nanohex, 2016-08-06:
getting error below.

nodejs peripheral-explorer.js B0:B4:48:B9:8E:83 
peripheral with ID b0b448b98e83 found
  Local Name        = CC2650 SensorTag
  Manufacturer Data = 0d00030000
  Service Data      = 
  Service UUIDs     = aa80

services and characteristics:
noble: unknown peripheral b0b448b98e83 connected!

nanohex, 2016-08-11:
This worked all before until most recently getting the latest to work under node-red

noble, noble-device, sensortag, node-red-node-sensortag

11 Aug 05:12:16 - [info] [sensorTag:da38c05e.25c74] found sensor tag: b0b448b98e83
11 Aug 05:12:16 - [info] [sensorTag:752fad59.8ad054] connected to sensor tag: b0b448b98e83
noble warning: unknown peripheral b0b448b98e83
11 Aug 05:12:16 - [info] [sensorTag:da38c05e.25c74] connected to sensor tag: b0b448b98e83
noble warning: unknown peripheral b0b448b98e83
noble: unknown peripheral null connected!
noble: unknown peripheral b0b448b98e83 connected!
11 Aug 05:12:17 - [info] [sensorTag:752fad59.8ad054] disconnected 
11 Aug 05:12:17 - [info] [sensorTag:da38c05e.25c74] disconnected 
noble: unknown peripheral b0b448b98e83 disconnected!
noble: unknown peripheral b0b448b98e83 connected!
noble: unknown peripheral b0b448b98e83 disconnected!
noble: unknown peripheral b0b448b98e83 connected!

running hcidump -t -x

HCI sniffer - Bluetooth packet analyzer ver 5.39
device: hci0 snap_len: 1500 filter: 0xffffffff
2016-08-11 05:12:01.281772 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 44 reason 0x13
    Reason: Remote User Terminated Connection
2016-08-11 05:12:01.284094 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2016-08-11 05:12:01.284263 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 44 reason 0x13
    Reason: Remote User Terminated Connection
2016-08-11 05:12:01.286094 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 0
2016-08-11 05:12:01.289103 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 44 reason 0x16
    Reason: Connection Terminated by Local Host
2016-08-11 05:12:01.291093 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x02 handle 44 reason 0x02
    Error: Unknown Connection Identifier
2016-08-11 05:12:01.292083 > HCI Event: Command Status (0x0f) plen 4
    Error: Unknown Connection Identifier
2016-08-11 05:12:01.292083 > HCI Event: Command Status (0x0f) plen 4
    Unknown (0x00|0x0000) status 0x00 ncmd 1
2016-08-11 05:12:16.326977 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen$
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
2016-08-11 05:12:16.329826 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-08-11 05:12:16.329970 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen$
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
2016-08-11 05:12:16.331841 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2016-08-11 05:12:16.405819 > HCI Event: LE Meta Event (0x3e) plen 26
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr B0:B4:48:B9:8E:83 (Public)     
      Flags: 0x05
      Shortened service classes: 0xaa80
      Unknown type 0xff with 5 bytes data
      RSSI: -34
2016-08-11 05:12:16.408841 > HCI Event: LE Meta Event (0x3e) plen 39
    LE Advertising Report
      SCAN_RSP - Scan Response (4)
      bdaddr B0:B4:48:B9:8E:83 (Public)
      Complete local name: 'CC2650 SensorTag'
[... 77 lines cut ...]
      Shortened service classes: 0xaa80
      Unknown type 0xff with 5 bytes data
      RSSI: -34
2016-08-11 05:12:16.692932 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen$
    value 0x00 (scanning disabled)
    filter duplicates 0x00 (disabled)
    value 0x00 (scanning disabled)
    filter duplicates 0x00 (disabled)       
2016-08-11 05:12:16.704820 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2016-08-11 05:12:16.704914 < HCI Command: LE Create Connection (0x08|0x000d) pl$
    bdaddr B0:B4:48:B9:8E:83 type 0
    interval 96 window 48 initiator_filter 0
    own_bdaddr_type 0 min_interval 40 max_interval 56
    latency 0 supervision_to 42 min_ce 0 max_ce 0
2016-08-11 05:12:16.709810 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
2016-08-11 05:12:16.912805 > HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 45, role master 
      bdaddr B0:B4:48:B9:8E:83 (Public)
2016-08-11 05:12:16.913188 < HCI Command: LE Read Remote Used Features (0x08|0x$
  2D 00
2016-08-11 05:12:16.915833 > HCI Event: Command Status (0x0f) plen 4

sandeepmistry, 2016-08-21:
@nanohex which kernel version are you using?

```
2016-08-11 05:12:16.600847 > HCI Event: Disconn Complete (0x05) plen 4
status 0x00 handle 44 reason 0x3e
016-08-11 05:12:16.600847 > HCI Event: Disconn Complete (0x05) plen 4
status 0x00 handle 44 reason 0x3e
Reason: Connection Failed to be Established
```

Looks really strange given:

```
2016-08-11 05:12:16.436166 < HCI Command: LE Create Connection (0x08|0x000d) pl$
bdaddr B0:B4:48:B9:8E:83 type 0
interval 96 window 48 initiator_filter 0
own_bdaddr_type 0 min_interval 6 max_interval 12

latency 0 supervision_to 200 min_ce 4 max_ce 6
2016-08-11 05:12:16.440840 > HCI Event: Command Status (0x0f) plen 4
LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
2016-08-11 05:12:16.536816 > HCI Event: LE Meta Event (0x3e) plen 19

LE Connection Complete
status 0x00 handle 44, role master
bdaddr B0:B4:48:B9:8E:83 (Public)
```

above.

nanohex, 2016-08-22:
uname -a >>
Linux nanopineo 3.4.112-sun8i #12 SMP PREEMPT Tue Aug 16 16:27:48 CEST 2016 armv7l armv7l armv7l GNU/Linux

cat /proc/version >>
Linux version 3.4.112-sun8i (root@armbian) (gcc version 5.3.1 20160413 (Ubuntu/Linaro 5.3.1-14ubuntu2) ) #12 SMP PREEMPT Tue Aug 16 16:27:48 CEST 2016

hope this helps.

sandeepmistry, 2016-08-22:
@nanohex thanks.

Do you have time to try downgrading the kernel?

nanohex, 2016-08-22:
sure, ok downgrading from 5.17 to 5.10 and lets see will update.

nanohex, 2016-08-22:
still getting same error

nanohex, 2016-08-22:
22 Aug 00:58:33 - [warn] ------------------------------------------------------
22 Aug 00:58:33 - [info] Settings file  : /root/.node-red/settings.js
22 Aug 00:58:33 - [info] User directory : /root/.node-red
22 Aug 00:58:33 - [info] Flows file     : /root/.node-red/flows_myiotserver.com.json
22 Aug 00:58:33 - [info] Server now running at http://127.0.0.1:1880/
22 Aug 00:58:33 - [info] Starting flows
22 Aug 00:58:33 - [info] Started flows
22 Aug 00:59:09 - [info] [sensorTag:37e9e3d4.c8161c] found sensor tag: b0b448b98e83
22 Aug 00:59:10 - [info] [sensorTag:37e9e3d4.c8161c] connected to sensor tag: b0b448b98e83
noble: unknown peripheral b0b448b98e83 connected!

So went all the way back to March, it was working in June and stopped in July.

name -r
Linux  3.4.110-sun8i #18 SMP PREEMPT Tue Mar 8 20:03:32 CET 2016 armv7l armv7l armv7l GNU/Linux

nanohex, 2016-08-22:
22 Aug 00:58:33 - [info] Server now running at http://127.0.0.1:1880/
22 Aug 00:58:33 - [info] Starting flows
22 Aug 00:58:33 - [info] Started flows
22 Aug 00:59:09 - [info] [sensorTag:37e9e3d4.c8161c] found sensor tag: b0b448b98e83
22 Aug 00:59:10 - [info] [sensorTag:37e9e3d4.c8161c] connected to sensor tag: b0b448b98e83
noble: unknown peripheral b0b448b98e83 connected!

So went all the way back to March, it was working in June and stopped in July.

name -r
Linux 3.4.110-sun8i #18 SMP PREEMPT Tue Mar 8 20:03:32 CET 2016 armv7l armv7l armv7l GNU/Linux

sandeepmistry, 2016-08-24:
> So went all the way back to March, it was working in June and stopped in July.

Do you have more details on which versions of things (npm packages, kernel etc)?

sandeepmistry, 2016-09-11:
@nanohex please try with the master version of noble:

```
cd node_modules/noble-device/node_modules
rm -rf noble
npm install sandeepmistry/noble
```

sandeepmistry, 2016-09-17:
Closing this due to lack of activity for now.

closed 2016-09-17 by sandeepmistry