noble/noble #445: Change security level

reallyroy, 2016-08-11:
I have a clarification rather than an issue.

How do I set or change the bluetooth security level in the Noble library? I looking to set up secure encrypted data transfer between my peripheral and the central module (noble) but I'm unable to find out if or how it is possible from the readme or code.

Would you be able to clarfiy or point me to the place where I can get this clarified?

Alfred1992, 2016-08-12:
I would like to know too

sandeepmistry, 2016-08-21:
@reallyroy @Alfred1992 

> How do I set or change the bluetooth security level in the Noble library?

It's all transparent following the Apple iOS/OS X model. When a peripheral device responds with `ATT_ECODE_AUTHENTICATION`, `ATT_ECODE_AUTHORIZATION` or `ATT_ECODE_INSUFF_ENC` for an GATT/ATT operation the pairing process will kick off.

At this time only "Just Works" pairing is supported, and pairing keys are not persisted.

closed 2016-08-21 by sandeepmistry

reallyroy, 2016-08-29:
Thanks Sandeep. A few questions though, is this also the case with the Bleno? If we need to use BLE 4.2 ECDH features, what would we have to do?

PanzerHabba, 2016-10-19:
@reallyroy 

Have you found any workarounds for this ?
I would also like to change ble securiy level.