noble/noble #446: nwjs crash when require('noble')

timfish, 2016-08-14:
```
platform: OSX v10.11.6
node: 6.3.1
npm: 3.10.3
xcode: 7.3.1
nwjs: 0.16.1
```

Sorry for yet another issue related to nwjs/electron!

I'm using a new-ish version of nwjs which doesn't require `nw-gyp`. `npm install noble` has no errors, but when I `require('noble')` in JavaScript I get a crash with some errors from the dynamic linker. 

```
Tims-MacBook:...$ npm install noble --save

> xpc-connection@0.1.4 install /Users/.../node_modules/xpc-connection
> node-gyp rebuild

  CXX(target) Release/obj.target/binding/src/XpcConnection.o
  SOLINK_MODULE(target) Release/binding.node
aurelia-skeleton-navigation@1.0.0-beta.1.2.1 /Users/...
└─┬ noble@1.6.0 
  ├── bplist-parser@0.0.6 
  └── xpc-connection@0.1.4 

npm WARN optional Skipping failed optional dependency /noble/bluetooth-hci-socket:
npm WARN notsup Not compatible with your operating system or architecture: bluetooth-hci-socket@0.4.4

Tims-MacBook:...$ nw .
Using nw.js v0.16.1-sdk
dyld: lazy symbol binding failed: Symbol not found: __ZN2v816FunctionTemplate3NewEPNS_7IsolateEPFvRKNS_20FunctionCallbackInfoINS_5ValueEEEENS_5LocalIS4_EENSA_INS_9SignatureEEEi
  Referenced from: /Users/.../node_modules/xpc-connection/build/Release/binding.node
  Expected in: flat namespace

dyld: Symbol not found: __ZN2v816FunctionTemplate3NewEPNS_7IsolateEPFvRKNS_20FunctionCallbackInfoINS_5ValueEEEENS_5LocalIS4_EENSA_INS_9SignatureEEEi
  Referenced from: /Users/.../node_modules/xpc-connection/build/Release/binding.node
  Expected in: flat namespace

====================
bye!
```

timfish, 2016-08-14:
Further testing shows that this also occurs with `require('xpc-connection')` so maybe the bug should be filed there instead...

sandeepmistry, 2016-08-21:
Hi @timfish, yes please move this issue to that repo: https://github.com/sandeepmistry/node-xpc-connection.

Also, please try it out with `nw-gyp` if possible.

timfish, 2016-08-22:
New issue for `node-xpc-connection` https://github.com/sandeepmistry/node-xpc-connection/issues/15

closed 2016-08-22 by timfish