noble/noble #447: Missing information in peripheral advertisement discovery data

andreydung, 2016-08-15:
I can run examples/advertisement-discovery.js, and it shows a list of devices with full information.

```
peripheral discovered (518e5806703b4173a12030c9e369f10f with address <ef:9d:c8:3d:ae:eb, unknown>, connectable true, RSSI -42:
       	hello my local name is:
       		Andrey
       	can I interest you in any of the following advertised services:
       		[]
       	here is my manufacturer data:
       		"00007377616c6c6f77"
       	my TX power level is:
       		-4
.... and some other peripherals
```

However, if I use UUID in startScanning to only pick wanted devices:

`    noble.startScanning(['7468656861626974736c61626e616269'], false);
`

The peripheral objects returned misses TX powerlevel, manufacturere data:

```
peripheral discovered (518e5806703b4173a12030c9e369f10f with address <ef:9d:c8:3d:ae:eb, unknown>, connectable true, RSSI -42:
       	hello my local name is:
       		Andrey
       	can I interest you in any of the following advertised services:
       		[]
       	here is my manufacturer data:
       		"00007377616c6c6f77"
       	my TX power level is:
       		-4
```

How can this be fixed?

sandeepmistry, 2016-08-21:
@andreydung, could you please share the debug logs? `DEBUG=* node <file>.js`

> I'm wondering, in which file is this implemented if the code is running on OSx El Capitan?

`lib/mac/yosemite.js`

sandeepmistry, 2016-09-11:
I'm closing this due to lack of activity, please re-open with requested info. if you are still interested.

closed 2016-09-11 by sandeepmistry