noble/noble #449: Unknown peripheral device failed to retrieve any information

alokranjan007, 2016-08-19:
Hello, 
I am trying to connect my BLE device angel sensor) using the index.js . However, It is connected it is throwing an error just after a connection. It is saying after the command like this `sudo node index.js 
Searching for Angel Sensor
connected to Angel Sensor 04453
UUID: 000780030213
Address: 00:07:80:03:02:13
noble warning: unknown peripheral 000780030213
` where uuid is my BLE sensor Mac Id. 
Would you please help me to resolve this error? What I am doing wrong :( 

Thank you in advance

nanohex, 2016-08-21:
I am getting the same messages https://github.com/sandeepmistry/noble/issues/443

sandeepmistry, 2016-08-22:
Hi @alokranjan007,

Could you please provide the following?

 * simple Node.js app to reproduce the issue
 * debug logs: `sudo DEBUG=* node <file>.js`
 * HCI dump: `sudo hcidump -t -x`
 * Kernel version

alokranjan007, 2016-08-22:
Hi sandeep,

Thank you for your reply.  I apologise for being late to your queries.

Please excuse me I am out of office and will be unaccessible for next 10 days.  It's a kind of emergency and can't be ignored.

So once I joined back I would provide you all details.

Thank you for your kindness and active response.

Thank you,
Alok

---- Sandeep Mistry wrote ----

Hi @alokranjan007<https://github.com/alokranjan007>,

Could you please provide the following?

  *   simple Node.js app to reproduce the issue
  *   debug logs: sudo DEBUG=* node <file>.js
  *   HCI dump: sudo hcidump -t -x
  *   Kernel version

sandeepmistry, 2016-09-17:
Closing this due to lack of activity for now.

closed 2016-09-17 by sandeepmistry