noble/noble #450: Not finding all devices

tellandev, 2016-08-23:
Hi

I am using Noble on a Raspberry Pi 3 and it is not managing to find my Samsung S7, a Jaguar car kit or an  iPhone 5 when doing a scan. It is however finding some of the BLE sensors I have. If I use the scan build into Jessie on the Pi it finds all the phones and the BLE devices we have. Are they getting filtered out or do I need to do a different kind of scan to find phones and car kits?

sandeepmistry, 2016-08-24:
Hi @tellandev.

Please try: https://github.com/sandeepmistry/noble#reporting-all-hci-events

tellandev, 2016-08-24:
Hi

Thanks for that but I think I have a different issue. If I run your enter
exit example it never picks up car kits or cell phones. Only beacons and
other BLE devices. For my use case, I just need to detect car kits and
phones but not connect to them.

Hope that makes sense

>
> Hi @tellandev.
>
> Please try:
https://github.com/sandeepmistry/noble#reporting-all-hci-events

tellandev, 2016-08-24:
Hi

I viewed this on a phone and it did not jump to the bookmark as you intended and displayed bits fr pulse. Just viewed your link on a desktop and it looks good. I will try it in the morning.

Thanks

tellandev, 2016-08-30:
Hi @sandeepmistry 

I have tried the reporting all hci events and I still do not get the phones or car kits near by. Any ideas?

sandeepmistry, 2016-09-05:
@tellandev what command are you using to scan for devices outside of noble?

noble will only report BLE advertisements. If your "phones or car kits near by." are Bluetooth Classic this will not work. For iPhone you can use an app like LightBluetooth to make the phone advertise using BLE.

tellandev, 2016-09-06:
Hi @sandeepmistry 

I think that is the problem then. The devices I need to detect are more likely Bluetooth Classic but I need to be able to detect both Bluetooth Classic and BLE devices for the project I am working on. Would it be possible for you to point me in the right direction to modify noble to work with Classic also or is that simply not possible?

sandeepmistry, 2016-09-06:
> Would it be possible for you to point me in the right direction to modify noble to work with Classic also or is that simply not possible?

You can modify [lib/hci-socket/hci.js](https://github.com/sandeepmistry/noble/blob/master/lib/hci-socket/hci.js) to add an HCI command to start a normal scan and listen for the relevant event(s) for non-LE devices.

I'm going to close this as noble is for BLE devices only.

closed 2016-09-06 by sandeepmistry

tellandev, 2016-09-07:
Thanks Sandeep. I will make the changes on a local copy.

c-o-m-m-a-n-d-e-r, 2016-12-19:
can you tell me how to do this?

i try to locate iphones in my house with this but without luck... 

thanks

jaewonha, 2019-08-16:
@sandeepmistry @tellandev would you share what changes did you made at the file?